bluekitchen/btstack #2: Baud rate problem

chunhua523, 2015-04-15:
Hi,

I'm using MSP430F5438A + CC2564B to send ADC data to an android phone.
It's successful but has a slow data rate, less than one data per sec.
I've referred to spp_and_le_counter.c, and set timers to send notifications.
Even though I set the timeout at 1ms, it's useless.

My target data rate is 1.5k per sec. How could I resolve this problem?

This is my demo vedio: https://youtu.be/vOHGzDM1oJg

Thanks!

wenjiegong, 2015-04-16:
you can not set baud rate on RFCOMM which will transfer the data as fast as it can. The article 'How Fast is Bluetooth?' is useful to understand the issue. You may change the MTU on L2CAP packets. If a system has a smaller L2CAP MTU, the size of RFCOMM data will also be restricted.

mringwal, 2015-04-16:
I've posted here: https://groups.google.com/d/msg/btstack-dev/BuBDK915-fI/gLT49Vyi2EYJ

closed 2015-04-16 by mringwal