bluekitchen/btstack #6: Testing BTstack on windows 7 using MSYS

izynet, 2015-07-01:
Hi, I've been through the Quick Start guide and I am stuck at the instruction "compile and install libusb".

gcc is complaining about missing headers like termios.h when I try to compile libusb from the platform/libusb folder.

Should I go for the precompiled libs from the libusb project or am I missing something ?

I am ready to draft a how-to for windows users if I can get BTstack working.

Thanks in advance for the reply.

mringwal, 2015-07-01:
You need to have libusb installed in /usr/local/lib before running make in platform/libusb.

new url for libusb is: libusb.info

you can either compile libusb yourself or try to use the pre compiled windows binaries. I think I've compiled them before (as the point is "compile & install").

closed 2015-07-01 by mringwal

notmikeb, 2017-09-14:
when missing unzip in msys shell, using "pacman -S unzip" to install unzip tool
use port/windows-winusb to build examples
Add "gap_inquiry" inside Makefile and 'make' will has a gap_inquiry.exe