bluekitchen/btstack #8: BTStack connection issue

swfredrik, 2015-07-13:
Having some trouble connecting to the stack. The pairing process with the phone is ok, but Im not able to do a connect. Is there anything obvious that you can see from this log during the connection process? 

- HCI_EVENT_READ_REMOTE_SUPPORTED_FEATURES_COMPLETE, bonding flags 6
LOG -- HCI_EVENT_LINK_KEY_REQUEST
LOG -- responding to link key request
LOG -- IO Capability Request received, stack bondable 1, io cap 1
LOG -- gap_mitm_protection_required_for_security_level 0
LOG -- hci_emit_security_level 3 for handle 1
LOG -- l2cap - security level update
LOG -- gap_request_security_level 0, current level 3
LOG -- hci_emit_security_level 3 for handle 1
LOG -- l2cap - security level update
LOG -- l2cap - state 5
LOG -- L2CAP_EVENT_INCOMING_CONNECTION addr BC:F5:AC:71:69:A3 handle 0x1 psm 0x1 local_cid 0x40 remote_cid 0x49
LOG -- L2CAP_ACCEPT_CONNECTION local_cid 0x40
LOG -- l2cap_stop_rtx for local cid 0x40
LOG -- l2cap_start_rtx for local cid 0x40
LOG -- gap_request_security_level 2, current level 3
LOG -- hci_emit_security_level 3 for handle 1
LOG -- l2cap - security level update
LOG -- l2cap - state 5
LOG -- L2CAP_EVENT_INCOMING_CONNECTION addr BC:F5:AC:71:69:A3 handle 0x1 psm 0x3 local_cid 0x41 remote_cid 0x47
LOG -- rfcomm_max_frame_size_for_l2cap_mtu:  1017 -> 1012
LOG -- L2CAP_EVENT_INCOMING_CONNECTION (l2cap_cid 0x41) for PSM_RFCOMM => accept
LOG -- L2CAP_ACCEPT_CONNECTION local_cid 0x41
LOG -- l2cap - state 9
LOG -- l2cap_stop_rtx for local cid 0x41
LOG -- l2cap_start_rtx for local cid 0x41
LOG -- L2CAP signaling handler code 4, state 9
LOG -- l2cap cid 64, hint 0, type 1
LOG -- L2CAP signaling handler code 5, state 9
LOG -- l2cap_stop_rtx for local cid 0x40
LOG -- L2CAP signaling handler code 4, state 9
LOG -- l2cap cid 65, hint 0, type 1
LOG -- L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr BC:F5:AC:71:69:A3 handle 0x1 psm 0x1 local_cid 0x40 remote_cid 0x49 local_mtu 1017, remote_mtu 672, flush_timeout 0
LOG -- L2CAP_EVENT_CREDITS local_cid 0x40 credits 1
LOG -- L2CAP signaling handler code 5, state 9
LOG -- l2cap_stop_rtx for local cid 0x41
LOG -- L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr BC:F5:AC:71:69:A3 handle 0x1 psm 0x3 local_cid 0x41 remote_cid 0x47 local_mtu 1017, remote_mtu 1691, flush_timeout 0
LOG -- L2CAP_EVENT_CHANNEL_OPENED for PSM_RFCOMM, status 0
LOG -- rfcomm_max_frame_size_for_l2cap_mtu:  1017 -> 1012
LOG -- L2CAP_EVENT_CREDITS local_cid 0x41 credits 1
LOG -- Received SABM #0
LOG -- Sending UA #0
LOG -- Multiplexer up and running
LOG -- L2CAP_EVENT_CREDITS local_cid 0x41 credits 1
LOG -- Received UIH Parameter Negotiation Command for #2, credits 7
LOG -- rfcomm_channel_create for service 8006998, channel 1 --- list of channels:
LOG -- -> Inform app
LOG -- RFCOMM_EVENT_INCOMING_CONNECTION addr BC:F5:AC:71:69:A3 channel #1 cid 0x01
LOG -- rfcomm_channel_ready_for_incoming_dlc_setup state var 0002
LOG -- L2CAP_EVENT_CREDITS local_cid 0x40 credits 1
LOG -- L2CAP signaling handler code 6, state 10
LOG -- L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x40
LOG -- l2cap_stop_rtx for local cid 0x40
LOG -- rfcomm_channel_ready_for_incoming_dlc_setup state var 0002
LOG -- gap_request_security_level 0, current level 3
LOG -- hci_emit_security_level 3 for handle 1
LOG -- l2cap - security level update
LOG -- l2cap - state 5
LOG -- L2CAP_EVENT_INCOMING_CONNECTION addr BC:F5:AC:71:69:A3 handle 0x1 psm 0x1 local_cid 0x42 remote_cid 0x48
LOG -- L2CAP_ACCEPT_CONNECTION local_cid 0x42
LOG -- l2cap - state 10
LOG -- rfcomm_channel_ready_for_incoming_dlc_setup state var 0002
LOG -- l2cap_stop_rtx for local cid 0x42
LOG -- l2cap_start_rtx for local cid 0x42
LOG -- L2CAP signaling handler code 4, state 9
LOG -- l2cap cid 66, hint 0, type 1
LOG -- rfcomm_channel_ready_for_incoming_dlc_setup state var 0002
LOG -- L2CAP signaling handler code 5, state 9
LOG -- l2cap_stop_rtx for local cid 0x42
LOG -- L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr BC:F5:AC:71:69:A3 handle 0x1 psm 0x1 local_cid 0x42 remote_cid 0x48 local_mtu 1017, remote_mtu 672, flush_timeout 0
LOG -- L2CAP_EVENT_CREDITS local_cid 0x42 credits 1
LOG -- L2CAP_EVENT_CREDITS local_cid 0x42 credits 1
LOG -- rfcomm_channel_ready_for_incoming_dlc_setup state var 0002
LOG -- L2CAP signaling handler code 6, state 10
LOG -- L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x42
LOG -- l2cap_stop_rtx for local cid 0x42
LOG -- rfcomm_channel_ready_for_incoming_dlc_setup state var 0002
LOG -- Received DISC #0, (ougoing = 0)
LOG -- TODO: no connections established - delete link key prophylactically
LOG -- Sending UA #0
LOG -- Closing down multiplexer
LOG -- L2CAP_EVENT_CREDITS local_cid 0x41 credits 1
LOG -- RFCOMM_EVENT_OPEN_CHANNEL_COMPLETE status 0x70 addr BC:F5:AC:71:69:A3 handle 0x1 channel #1 cid 0x01 mtu 100
LOG -- L2CAP_DISCONNECT local_cid 0x41 reason 0x13
LOG -- L2CAP signaling handler code 6, state 11
LOG -- L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x41
LOG -- L2CAP_EVENT_CHANNEL_CLOSED cid 0x41, mult 0
LOG -- l2cap_stop_rtx for local cid 0x41

mringwal, 2015-07-13:
Looks like the other side shuts it down. Could you please post the full packet log, see http://bluekitchen-gmbh.com/btstack/how_to/#bluetooth-hci-packet-logs

swfredrik, 2015-07-15:
Added the HCI dump log, I've only got 5k for debug logging but here is the final part of the log including the hci dump. 

0 14 00 40 00 06 00 00 00 0F 35 03 19 12 00 02 90 35 05 0A 00 00 FF FF 00 
ACL => 01 00 0F 00 0B 00 47 00 07 00 00 00 06 00 03 36 00 00 00 
LOG -- L2CAP_EVENT_CREDITS local_cid 0x40 credits 1
EVT <= 74 03 40 00 01 
EVT <= 6C 00 
EVT <= 13 05 01 01 00 01 00 
ACL <= 01 20 08 00 04 00 41 00 03 3F 01 1C 
LOG -- Received SABM #0
LOG -- Sending UA #0
ACL => 01 00 08 00 04 00 46 00 03 73 01 D7 
LOG -- L2CAP_EVENT_CREDITS local_cid 0x41 credits 1
EVT <= 74 03 41 00 01 
LOG -- Multiplexer up and running
EVT <= 6C 00 
EVT <= 13 05 01 01 00 01 00 
ACL <= 01 20 0C 00 08 00 01 00 06 07 04 00 40 00 47 00 
LOG -- L2CAP signaling handler code 6, state 10
ACL => 01 00 0C 00 08 00 01 00 07 07 04 00 40 00 47 00 
LOG -- L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x40
EVT <= 71 02 40 00 
LOG -- l2cap_stop_rtx for local cid 0x40
EVT <= 6C 00 
EVT <= 13 05 01 01 00 01 00 
ACL <= 01 20 12 00 0E 00 41 00 03 EF 15 83 11 02 F0 00 00 DE 03 00 07 70 
LOG -- Received UIH Parameter Negotiation Command for #2, credits 7
LOG -- rfcomm_channel_create for service 8006998, channel 1 --- list of channels:
LOG -- -> Inform app
LOG -- RFCOMM_EVENT_INCOMING_CONNECTION addr BC:F5:AC:71:69:A3 channel #1 cid 0x01
EVT <= 82 09 A3 69 71 AC F5 BC 01 01 00 
LOG -- rfcomm_channel_ready_for_incoming_dlc_setup state var 0002
EVT <= 13 05 01 01 00 01 00 
ACL <= 01 20 0C 00 08 00 01 00 02 08 04 00 01 00 48 00 
LOG -- gap_request_security_level 0, current level 3
LOG -- hci_emit_security_level 3 for handle 1
EVT <= E0 03 01 00 03 
LOG -- l2cap - security level update
LOG -- l2cap - state 5
LOG -- L2CAP_EVENT_INCOMING_CONNECTION addr BC:F5:AC:71:69:A3 handle 0x1 psm 0x1 local_cid 0x42 remote_cid 0x48
EVT <= 72 0E A3 69 71 AC F5 BC 01 00 01 00 42 00 48 00 
LOG -- L2CAP_ACCEPT_CONNECTION local_cid 0x42
ACL => 01 00 10 00 0C 00 01 00 03 08 08 00 42 00 48 00 00 00 00 00 
LOG -- l2cap - state 10
EVT <= 6C 00 
LOG -- rfcomm_channel_ready_for_incoming_dlc_setup state var 0002
ACL => 01 00 10 00 0C 00 01 00 04 03 08 00 48 00 00 00 01 02 F9 03 
LOG -- l2cap_stop_rtx for local cid 0x42
LOG -- l2cap_start_rtx for local cid 0x42
EVT <= 6C 00 
LOG -- rfcomm_channel_ready_for_incoming_dlc_setup state var 0002
ACL <= 01 20 10 00 0C 00 01 00 04 09 08 00 42 00 00 00 01 02 A0 02 
LOG -- L2CAP signaling handler code 4, state 9
LOG -- l2cap cid 66, hint 0, type 1
ACL => 01 00 12 00 0E 00 01 00 05 09 0A 00 48 00 00 00 00 00 01 02 A0 02 
EVT <= 6C 00 
LOG -- rfcomm_channel_ready_for_incoming_dlc_setup state var 0002
EVT <= 13 05 01 01 00 02 00 
ACL <= 01 20 0E 00 0A 00 01 00 05 03 06 00 42 00 00 00 00 00 
LOG -- L2CAP signaling handler code 5, state 9
LOG -- l2cap_stop_rtx for local cid 0x42
LOG -- L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr BC:F5:AC:71:69:A3 handle 0x1 psm 0x1 local_cid 0x42 remote_cid 0x48 local_mtu 1017, remote_mtu 672, flush_timeout 0
EVT <= 70 15 00 A3 69 71 AC F5 BC 01 00 01 00 42 00 48 00 F9 03 A0 02 00 00 
LOG -- L2CAP_EVENT_CREDITS local_cid 0x42 credits 1
EVT <= 74 03 42 00 01 
EVT <= 13 05 01 01 00 01 00 
ACL <= 01 20 18 00 14 00 42 00 06 00 00 00 0F 35 03 19 01 00 02 90 35 05 0A 00 00 FF FF 00 
ACL => 01 00 70 00 6C 00 48 00 07 00 00 00 67 00 64 36 00 61 36 00 5E 09 00 00 0A 00 01 00 01 09 00 01 36 00 03 19 11 01 09 00 04 36 00 0E 36 00 03 19 01 00 36 00 05 19 00 03 08 01 09 00 05 36 00 03 19 10 02 09 00 06 36 00 09 09 65 6E 09 00 6A 09 01 00 09 00 09 36 00 09 36 00 06 19 11 01 09 01 02 09 01 00 25 0D 53 50 50 20 49 6E 74 65 72 66 61 63 65 00 
LOG -- L2CAP_EVENT_CREDITS local_cid 0x42 credits 1
EVT <= 74 03 42 00 01 
EVT <= 6C 00 
LOG -- rfcomm_channel_ready_for_incoming_dlc_setup state var 0002
EVT <= 13 05 01 01 00 01 00 
ACL <= 01 20 0C 00 08 00 01 00 06 0A 04 00 42 00 48 00 
LOG -- L2CAP signaling handler code 6, state 10
ACL => 01 00 0C 00 08 00 01 00 07 0A 04 00 42 00 48 00 
LOG -- L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x42
EVT <= 71 02 42 00 
LOG -- l2cap_stop_rtx for local cid 0x42
EVT <= 6C 00 
LOG -- rfcomm_channel_ready_for_incoming_dlc_setup state var 0002
EVT <= 13 05 01 01 00 01 00 
EVT <= 01 01 00 
ACL <= 01 20 08 00 04 00 41 00 03 53 01 FD 
LOG -- Received DISC #0, (ougoing = 0)
LOG -- TODO: no connections established - delete link key prophylactically
LOG -- Sending UA #0
LOG -- Closing down multiplexer
ACL => 01 00 08 00 04 00 46 00 03 73 01 D7 
LOG -- L2CAP_EVENT_CREDITS local_cid 0x41 credits 1
EVT <= 74 03 41 00 01 
LOG -- RFCOMM_EVENT_OPEN_CHANNEL_COMPLETE status 0x70 addr BC:F5:AC:71:69:A3 handle 0x1 channel #1 cid 0x01 mtu 100
EVT <= 80 0E 70 A3 69 71 AC F5 BC 01 00 01 01 00 64 00 
LOG -- L2CAP_DISCONNECT local_cid 0x41 reason 0x13
EVT <= 6C 00 
ACL => 01 00 0C 00 08 00 01 00 06 04 04 00 46 00 41 00 
EVT <= 6C 00 
EVT <= 13 05 01 01 00 02 00 
ACL <= 01 20 0C 00 08 00 01 00 06 0B 04 00 41 00 46 00 
LOG -- L2CAP signaling handler code 6, state 11
ACL => 01 00 0C 00 08 00 01 00 07 0B 04 00 41 00 46 00 
LOG -- L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x41
EVT <= 71 02 41 00 
LOG -- L2CAP_EVENT_CHANNEL_CLOSED cid 0x41, mult 0
LOG -- l2cap_stop_rtx for local cid 0x41
ACL <= 01 20 0C 00 08 00 01 00 07 04 04 00 46 00 41 00 
EVT <= 6C 00 
EVT <= 13 05 01 01 00 01 00

mringwal, 2015-07-16:
Thanks. Without the packets for HCI and L2CAP connection setup, Apple's PacketLogger won't properly decode the higher layer messages. I'll respond on the mailing list.

closed 2015-07-16 by mringwal

flograsso, 2017-12-06:
How can i deactivate the "EVT , LOG, ACL and CMD" messages on the Serial port?