bluekitchen/btstack #9: iOS build error

shattars3d, 2015-07-13:
[image]

having compiling issues/errors.. what is the work around this?...

mringwal, 2015-07-13:
Can you run the file command in a terminal against libBTstack.dylib?
Did you copy it from your iOS device?

I get:

    $ file libBTstack.dylib
    libBTstack.dylib: Mach-O universal binary with 2 architectures
    libBTstack.dylib (for architecture armv6):	Mach-O dynamically linked shared library arm
    libBTstack.dylib (for architecture arm64):	Mach-O 64-bit dynamically linked shared library

shattars3d, 2015-07-13:
hey, no… i got my libBTstack.dylib from online.. from some research…  do i have to run any make commands?… because i didn't run..

Go to btstack/platforms/ios and run make

	cd btstack
	make package

If everything went right, you'll end up with a .deb package that you can install via:

	make install

mringwal, 2015-07-14:
Please try with this version: https://dl.dropboxusercontent.com/u/5138628/libBTstack.dylib

shattars3d, 2015-07-14:
hey thanks, i have tried your libBTstack.dylib. the Xcode built succeeded but i got this error not when the app is present on the iPad...

[image]

mringwal, 2015-07-14:
Oh. I didn't know that there are still Xcode projects. I gave up on using Xcode for JB development when theos came out and made everything understandable instead of hoping that Xcode wouldn't screw up.

So... I've deleted these projects but updated the Makefile for the WiiMoteOpenGL demo. Please see the updated [README](https://github.com/bluekitchen/btstack/tree/master/platforms/ios)

closed 2015-07-14 by mringwal

MattSich, 2016-05-06:
So there's not way to work with this in xCode? @mringwal

mringwal, 2016-05-06:
Well, if you like you could setup an Xcode project and compile with it. It's just that you have to maintain the project then. It's also possible to use the existing Makefiles with Xcode as far as I know, it just doesn't work whenever I try - and I'm fine with Makefiles and Sublime Text.