bluekitchen/btstack #11: Btstack not compatible with ios 9.0.2

else-one, 2015-11-26:
Hi，mringwal
  I am developing an IOS application based on btstack, but it does not seem to work on Ios 9.0.2.
  I compile the CocoaTouch sample, it works on ios 8.4  but also not on ios 9.0.2.

[image]

The bt_open() method just return -1,so is it not  compatible with ios 9.0.2 ? Or there are some new changes?

Waiting for your reply, thank you!

mringwal, 2015-11-26:
BTstack is compatible with iOS 9.0.2. BTstack GPS works without changes.

The BTstack library communicates with BTDaemon via /tmp/btstack. I think that you cannot access this from sandboxed apps (like from an app tweak) anymore. Where is your code running?

If this is the problem, I'd recommend to create a standalone daemon and communicate with this daemon via CFMessages & RocketBootstrap.

else-one, 2015-11-26:
OK,I think this is the problem.
My application is really a sandbox app, I will try this to validate.

mringwal, 2015-11-27:
If it's your app, and not a tweak, you could just build it with THEOS, like the provided Wii Remote demo app. If compiled and installed the JB-way, it will not be restricted. (It should be possible to break out of your own sandbox on a JB but I'm not sure about how to. It might be possible to just add some extra entitlements)

closed 2015-11-27 by mringwal