bluekitchen/btstack #16: SDP Bug

jcm102120, 2016-09-15:
I'm seeing some weird behavior with sdp services that I am not sure what the issue is.  If I register multiple services then do a sdptool browse from another machine I am seeing the same service printed twice rather than two different services.  For example:

sdptool browse:
Service RecHandle: 0x10001
Service Class ID List:
  "" (0x1800)
Protocol Descriptor List:
  "L2CAP" (0x0100)
    PSM: 31
  "ATT" (0x0007)
    uint16: 0x14
    uint16: 0x1e

Service RecHandle: 0x10001
Service Class ID List:
  "" (0x1800)
Protocol Descriptor List:
  "L2CAP" (0x0100)
    PSM: 31
  "ATT" (0x0007)
    uint16: 0x14
    uint16: 0x1e

The second thing i registered appears twice but the first registered service is missing.
Can anyone confirm this issue?

mringwal, 2016-09-16:
Thanks for reporting. 

Quick check: did you use different record handles for the 2 services?

jcm102120, 2016-09-16:
For that example I had used 0x10000 and 0x10001.  It seems that the last service to be registered is displayed multiple times no matter what I try as the record handles.

jcm102120, 2016-09-16:
This issue can be closed, this is expected behavior for the way I was building the record.  The new version of btstack does not copy the record when adding to the linked list.  I was using one buffer to create and register each record.  Using a new buffer for each record fixes the problem.

closed 2016-09-16 by jcm102120

mringwal, 2016-09-17:
thanks for debugging this yourself.

In the newer version, we have replaced the two different implementations (one was using malloc, the other did not allow to place the record in Flash) with a unified one that requires the client to provide the storage, but the record could be placed in Flash to save the 100-200 bytes of the SDP record on very small systems

I just checked, the documentation is minima (@param record is not copied!). Any suggestion what to change in the docs on that would have prevented you from running into this issue?

jcm102120, 2016-09-20:
If it were mentioned in section  "10.4.1. Create  and  announce  SDP  records" in the doc, it might help someone else in the future.  Thanks for the work on this and for the help!

mringwal, 2016-09-21:
Good idea. I've added something on the develop branch: http://bluekitchen-gmbh.com/btstack/develop/protocols/#sdp-service-discovery-protocol