bluekitchen/btstack #18: After jailbreaking iOS 10 the btsack need to update.

iOS4all, 2016-12-30:
Hi there as I said in the title. Btsack needs to update to jailbroken iOS 10 please. In order to use controllers4all

mringwal, 2017-01-01:
What's the problem on iOS 10?

I don't have a device that's supported by the Yalu JB.

iOS4all, 2017-01-01:
@mringwal thank you for replying. The btsack stop working on jailbroken iOS 10.1.1

I hope if you have an idea it could help. as you said you didn't have a device on jailbroken iOS 10. I can be your tester if you want.
Thanks

iOS4all, 2017-01-02:
I can't connect with any games on iOS 10 with (controllers4all) it always said:
Cannot connect to btsack,please reinstall btsack/reboot your device

This tweak need to be compatible with iOS 10.

ddagunts, 2017-01-04:
@mringwal I took a look at this.  Appears there were two issues one of which is already fixed.  One issue was that sandboxed apps couldn't talk to BTstack without RocketBootstrap being updated (or apps being unsandboxed or perhaps switching to TCP for BTdaemon would work as well).  RocketBootstrap was just updated and now works on iOS 10.  

The other issue is that the jailbreak doesn't load the launch daemons for btstack or controllers for all.  If I manually load them everything works and I can control games.  So basically BTstack works fine, @iOS4all just wait for a more solid jailbreak.

mringwal, 2017-01-04:
Thanks for the actual information. I'm not (very) involved with JB at the moment, but have an iPhone 6 Plus with 10.0.1 ready once there is a JB for a few tests. (I also updated my main iPhone SE to 10.2 as there were rumours about another exploit...)

iOS4all, 2017-01-04:
@ddagunts yeah it seems btsack is not broken at all. Because when I updated RocketBootstrap the contollers 4 all is start working great.
@mringwal thank you so much for this amazing thing (btsack)

simplyargyll, 2017-01-27:
How did you get it to work? Yalu 10.2 user here

german8835, 2017-01-31:
I need some help. I am on Yalu 10.2 jailbreak, and I had BTStack installed. After I deleted it, my Bluetooth is turning on only from settings app, a haven't  had any tweaks installed except the whole controller for all tweak package. I also rebooted my device twice, and tried to install BTstack again, and delete again. Thanks for your help.

mringwal, 2017-02-01:
@german8835 BTstack does not modify any other file on your device. If you de-install it and do a reboot, everything should be as before. If you have problems, they are not caused by BTstack. :(

iOS4all, 2017-02-01:
@mringwal you're right. Because I have a jailbroken iPad Air 2 on 10.2 and never notice any problem of btstack at all because substrate is like before unlike my jailbroken iPhone 7 on 10.1.1 because of need to do substrate alone so I have some issues there. So I believe that there's no problem with btstack or C4all but this is a jailbreak issue.

hund567, 2017-02-04:
@iOS4all @mringwal help me plz. i get a ipad pro 10.2 with yalu jailbreak. btstack and btc mouse installed. but ipad cannot recognise the bluetooth mouse,there's no cursor on the screen. and i installed the latest rocketbootstrap, what's the problem? thank u so much!!

hund567, 2017-02-04:
@iOS4all u got your device found by the device？

hund567, 2017-02-04:
@iOS4all sorry ，wrong typing. u got your device found by the ipad？

mringwal, 2017-02-06:
BTC Mouse & Trackpad - which is not related to BTstack - does not work on iOS 10. I don't know what's wrong currently. I also consider support for Bluetooth 4.0 mouse devices and might create a new BTC Mouse+ product. Will let you know once I made progress.

hund567, 2017-02-06:
thanks ! and if u want some guy test the tweak, plz count me in. love btc mouse so much！IT really provide great convenience! thanks again!

kgfromthemv, 2017-03-02:
to readdress what @iOS4all said, I think BTstack needs updates for iOS 10. I have two devices jailbroken on iOS 10: a 9.7" iPad Pro on 10.2 and an iPhone 7 on 10.1.1. I've tried using BTstack and CfA extensively on both. I've tried using older versions of BTstack, as some recommended that to get it to work with Controllers for All. I've tried loading/launching the BTstack daemons manually and still can't get it to work. 

RocketBoostrap has been updated for some time now, as @ddagunts suggested previously could be the cause of the issue, but I still haven't been able to get BTstack to work properly with CfA. I know you're not the dev of CfA, but I think the problem lies within BTstack. 

I would love to help you troubleshoot any issues and am available at almost any time to do any sort of troubleshooting you might need. 

Thanks for all the hard work you've put into this over the years. This functionality has always been the main source of motivation for me to jailbreak and I would love to see it working properly on the iOS 10 JB.

mringwal, 2017-03-03:
Hi Kyle

What did you try with BTstack itself? What did not work?

I've tried my BTstack GPS and it properly reports nearby devices (did not have a GPS device ready).
You can also run e.g. the 'inquiry' command on a shell, which requires that you disable Apple Bluetooth manually first. That works, too.

So everything seems to be fine from my point of view. 

As you might understand, I'm not inclined to debug other people's commercial products. If someone, e.g. CfA's author, files a bug report about incorrect behaviour of BTstack, I'll surely look after it (fixes would be appreciated as well).

Cheers
 Matthias

kgfromthemv, 2017-03-04:
I will outline the exact process I went through, and the results I had. I'm assuming you have a basic understanding of what Controllers for All does? That is my main use-case for BTstack. 

I've also included the hci_dump.pklg file which can be found [here](https://drive.google.com/file/d/0B1b1HtMNAlceX1lkMXlLa1JQZ3c/view?usp=sharing). I had logging toggled on for BTstack during the entire process:

I used Cydia to install Controllers for All, which automatically queue'd the latest version of BTstack, 1.1-1. After the install, Cydia prompted me to Restart Springboard. I know from previous experience that it is best to reboot the device after installing something like BTstack, so that is what I did. I rebooted and re-jailbroke my device. Then, I went to the Controllers for All settings pane and attempted to "Pair" my controller. As soon as I opened the settings pane for CfA, I got a notification saying something like "Cannot connect to BTstack. Please reinstall BTstack and reboot your device and try again". I figured that it is possible the BTstack LaunchDaemon didn't load, so I went to Mobile Terminal and ran `launchctl launch /Library/LaunchDaemons/ch.ringwald.BTstack.plist`, then `launchctl start /Library/LaunchDaemons/ch.ringwald.BTstack.plist`. I attempted to pair my controller again, but got the same notification saying "cannot connect to BTstack". 

I had read on other forums and reddit posts that an earlier version of BTstack was working for people. I had already saved a .deb of version 0.8-3 of BTstack, so I uninstalled CfA, and v. 1.1-1 of BTstack, then installed v 0.8-3, then reinstalled CfA. Rebooted device, rejailbroke and attempted to pair controller. This version of BTstack allows manual toggling of either the BTstack, the stock iOS bluetooth, or "Off". I toggled BTstack on, then attempted to pair my controller. I was able to successfully "pair" the controller and "connect" the controller. (If you're not familair with the difference, "pairing" is linking the controller to the device, "connecting" is actually switching the controller on and preparing to use it in a game.) 

Now that the controller was connected, I attempted to launch a game that normally works with CfA. It is a game that normally allows for MFi controllers, and I have used CfA successfully on previous jailbreaks with this game. Once I launched the game, I again received the "Cannot connect to BTstack, please reinstall BTstack and reboot your device" notification. No input was detected in the game and the controller shut off. At this point I decided to save the log file and come here to explain everything. 

One thing that crossed my mind is that, the last time you updated BTstack, all of the existing jailbreaks were "untethered", meaning that your phone was still jailbroken upon reboot, meaning also that the jailbreak process happened during the boot process. With the existing jailbreak, the jailbreak process happens _after_ the boot process. I thought that might be a cause of the problems people are experiencing. I search reddit at least 2 times a week for news of people having success with CfA and BTstack on the iOS 10 jailbreak, but I never find any. I figured I would at least attempt to help make some progress on the fix, if I'm able. Hopefully all of this helps.

mringwal, 2017-03-30:
Clients communicate with the BTstack Daemon (BTDaemon) via the unix domain socket /tmp/btstack.
A tweak inside an App Store app will be Sand-boxed and cannot access /tmp/btstack. This has been the case for a few iOS versions and that's probably why you get "cannot connect to BTstack". 

I'll update the topic.