bluekitchen/btstack #19: There is not any "force button" on BTStack 1.1-1

karless, 2017-01-07:
I just jnstalled BTStack in order to use controllers for all but when i want to connect a bluetooth device like my apple watch ,i should reboot my jailbroken device(iphone 6s ios 9.3.3).i read that i have to press "force off" button without rebooting my device.but there is not any button https://imgur.com/a/p3znb

mringwal, 2017-01-07:
The "force off" button only shows up on the right when BTstack is active. (AirDrop doesn't work right now, so I cannot add a screenshot)

karless, 2017-01-07:
Ok i just found it!i have another problem,i should reboot my device in order ios's bluetooth get back active.Even if i force off BTstack i cant connect any device via bluetooth.for exampe if i want to connect my apple watch i should reboot my device.any way to fix it?

mringwal, 2017-01-09:
Sorry, no idea.

Over time, Apple added more and more Bluetooth features such that using BTstack becomes a less and less optimal solution. It's possible to extend iOS Bluetooth like my BTC Mouse & Trackpad tweak, but I still didn't get around to update BTstack GPS, let alone others like WeBe++ or Blutrol.

closed 2017-01-09 by mringwal