bluekitchen/btstack #21: Do you have any plan for mbed support?

ismailhkose, 2017-02-06:


mringwal, 2017-02-06:
Not currently. I'm not too familiar with mbed: I only know that it has the Mass Storage firmware update and a web IDE (while I personally prefer a command line and Makefile).

mringwal, 2017-03-30:
What parts of mbed are you interested in? The Real-Time OS? The device drivers? 

What hardware (board + Bluetooth chipset) would you be interested in?

kennel2014, 2017-04-07:
http://bluekitchen-gmbh.com/ ，says has supported MFI，but i cannot find iap or mfi code from the master branch yet.

mringwal, 2017-04-07:
@kennel2014 The MFi/iAP code is only available to MFi Licensees. If your company is in the MFi program, please contact us at contact@bluekitchen-gmbh.com.

ismailhkose, 2017-04-25:
I would prefer to use this stack at mbed for BT classic and BLE applications. GATT and SPP would be good for initial application. I am familiar with mbed, but i dont know if i have time to help you for conversion. Do you accept external contribution to the project?

mringwal, 2017-04-26:
@ismailhkose I'm not too familiar with mbed: it's everything and nothing. There's a website, an online-compiler/IDE, a build system, board support packages for various boards, a way to upload firmware images via USB Mass Storage, and an RTOS - among other things. Could you explain a bit more?

For a BTstack port, we would need at least on dev board supported by mbed and a Bluetooth chipset + working UART drivers. I'm not sure if there is such a hardware setup (board + Bluetooth "shield") that does not require soldering/jumper wires. I currently play with the SMT32 F4 Discovery Board. For this, TI provides an adapter board for their CC256x line of Bluetooth chipsets.

Another option could be to use an USB Bluetooth dongle with devices that support USB Host. That could make sense. 

Yes, we would accept contributions, however, we would require to keep the full copyright (similar to e.g. Facebook) as it would get too complicated for our customers otherwise.

ismailhkose, 2017-05-10:
mbed is a platform and an operating system that is designed to support mostly IoT for 32-bit ARM microcontrollers. This platform is developed and supported by ARM,

You dont have to use online compiler. There is also offline compiler option, which is called mbed CLI
mbed supports C++ and increase and ease rapid development with OOP approach. 

There are many available supported boards. 

Also, Maxim Integrated has mbed board that uses PAN1326B Bluetooth IC. But, there isn't any BT/BLE stack available yet. You can see the details in the following page.

https://developer.mbed.org/platforms/MAX32630FTHR/

mringwal, 2017-05-11:
Thanks for the overview. In the meantime, I did work on a custom port of BTstack to an mbed board, but the lower layer stuff was already in place. Also without an RTOS.

Anyway, thanks especially for the hint about the MAX32630FTHR. I saw it in a Digikey mailing, but couldn't figure what Bluetooth module it uses - shame on me, while it's not on the web page, the data sheet has all the details

The MAX32630FTHR would be a good start for an mbed port. I'm gonna get one and try to make port & post like this one:  http://bluekitchen-gmbh.com/btstack-port-for-stm32-f4-discovery-board-with-cc256x/

andrewwade, 2017-12-04:
Hello, I would also love to have this ported to mbed. The good thing is if you use the mbed lower level implementations of things like timers, uart, and flash, BTStack will become easily portable to any mbed supported MCU. What is the status on this?

mringwal, 2017-12-05:
Hi @andrewwade. @ismailhkose did a port of BTstack for the Maxim FTHR board and we've added it to the supported ports after a bit of cleanup. It did use the Maxim Peripheral drivers directly and would be a good platform to switch to mbed. 

What mbed platform would you like to use BTstack on? Do you need/want multithreading?

andrewwade, 2017-12-07:
I primarily am interested in porting it to the kinetis MCU’s. Our product uses a k22f and the Toshiba tc35661 Bluetooth chip. We do use Mbed RTOS as well.

mringwal, 2017-12-07:
@andrewwade thanks for the overview. BTstack isn't thread-safe, but runs well on a dedicated thread e.g. on FreeRTOS or mbed RTOS. In this case, all Bluetooth communication happens on the "Bluetooth" thread and queues could be used to communication with other threads.

If this approach fits your environment, BTstack should be a good fit. If you like, please contact us at contact@bluekitchen-gmbh.com to discuss how we can help you with your port.