bluekitchen/btstack #24: Use UART instead of SPI

agdl, 2017-02-15:
I just have a quick question. Is it possible to use the UART instead of the SPI in the Arduino porting?

mringwal, 2017-02-15:
In general, there's no problem with using UART (H4) with the Arduino port if the RTS line is working. 

The current port was for a EM9301 that has SPI and UART H4 but does not have RTS/CTS. I've used SPI because of this. 

I think the UART driver of Arduino with AVR uses a ring buffer and also drives RTS before the buffer overruns.

agdl, 2017-02-16:
Ok nice! Can you please tell me where have I to look for it please? I mean where I can find a guide to tweak the code to use the UART

mringwal, 2017-02-20:
Hi. Do you want to use a regular H4 module or the EM9301 via H4?

There's no guide exactly for this. You'd need to change the file port/arduino/bsp_arduino_em9301.cpp. There, instead of SPI you should use the Serial module. If you can power cycle the module (e.g. if it has a RESET/SHUTDOWN line), you can define HAVE_SHUTDOWN>. Feel free to ask more on BTstack Dev List. http://groups.google.com/group/btstack-dev

closed 2017-02-20 by mringwal