bluekitchen/btstack #27: BTC Mouse & Trackpad IOS 10

goliath88, 2017-03-29:
Hello. I tryed to use BTC mouse on my Ipad air with ios 10 on it, but it seems my bluetooth can't find the mouse. 
On ios 9 there where no problems. 
Will there be an update for this great app?

mringwal, 2017-03-30:
BTC Mouse & Trackpad is published by BlueKitchen, but it is not directly related to BTstack - the only common part is that it uses Bluetooth.

There's a change in iOS10 that stops me from intercepting Bluetooth packets. I don't know yet if/when I manage to fix this. Please follow me on Twitter @mringwal for updates on BTC Mouse & Trackpad

wadisnake, 2017-04-15:
Hello, I know I shouldn't ask and I am sorry for that, but have you made any progress ? :) big community is waiting for you to make IPad better :)

mringwal, 2017-04-16:
no progress yet. sorry.

wadisnake, 2017-04-16:
Thank you so much for the update. Looking forward to it :) 
Good end of weekend !

jannick1900, 2017-05-28:
Any progress ? Hope for a respond

VideoGeek, 2017-06-08:
Same here I'm reaching out,  I would pay $30+ plus for this....Apple need to sort themselves out, I have no idea what goes in to making this work but please get it working on iOS 10 👍

bbonmars, 2017-06-14:
BTC Mouse & Trackpad is working for me on IOS 10.2. When I first jailbroke my iPad Air it did not work. However Cydia pushed out substrate, I installed all extensions, then I re-installed BTC and it now works.

I'm using the following: IOS 10.2, Yalu 10.2, iPad Air, Logitech FabricSkin Keyboard Folio i5,  Logitech Ultrathin Touch Mouse.

Mouse performance: scrolling works (Safari, Settings), mouse pointer reaches all areas of the screen, app icons open with one right click, apps close with left click, at lock screen mouse can click on passcode numbers and unlock iPad. 

However there can be Cydia apps that interfere. 

I recommend: only installing BTC (then after it works install Cydia apps one at a time), make sure Cydia substrate and extensions are up to date, re-install BTC Mouse & Trackpad.

snowswords, 2018-01-14:
how about ios11?

mringwal, 2018-01-19:
No progress here, sorry. BTstack GPS probably will still work in iOS 11, while BTC Mouse & Trackpad won't work until it's fixed for iOS 10 (and later).

redsuns78, 2018-02-03:
Is there any chance that you can update for iOS 10. For a long time, so many people who bought your great tweak have been waiting for the update. If not, please tell us you can't so that we don't expect no longer..

sojubada, 2018-02-25:
Hi~ mringwal, i hear you can’t catach the BT signal on iPhone.

In my think, Apple TV web browser(Jailbreaken device) have mouse courser and work with bluetooth controller.

So if you are considering that things, you can make the solution. 

Many people want to use apple tablet like laptop, So mouse is the most important factor. 

I’m not English well, so sorry ;)

Fusi0nX, 2018-03-16:
@mringwal Hi, wondering if there has been any progress on BTC Mouse & Trackpad for ios 11?

mringwal, 2018-03-28:
@sojubada that's a great idea. A while ago, I've made a proof-of-concept of a web browser controlled by a Bluetooth LE mouse that could be in the App Store. Didn't get to work on it more.

mringwal, 2018-03-28:
I'm fully busy developing and supporting BTstack and didn't try the iOS 11 JB yet for BTC. Sorry.

miguelromero2000, 2018-03-31:
@bbonmars @mringwal I have finally bought  Logitech Ultrathin Touch Mouse T631 in Amazon and it works a lot better than Magi Mouse 2. Accuracy is not perfect but much better now and it should do the job.
Thank you a lot @mringwal for your effort and all your work :)

redsuns78, 2018-03-31:
@miguelromero2000 Thank you for sharing. Do you think Ultrathin Touch Mouse T631 is also discoverable on iOS 10.2? As you can see in a reddit post below, It is known that it is not visible when trying to pair. thanks again 

https://www.reddit.com/r/jailbreak/comments/5sybpc/discussion_btc_mouse_and_trackpad_works_on_1011/

miguelromero2000, 2018-04-16:
Sorry I do not really know since I don't have iOS 10. The only thing you may try is to go to a shop where they have some Logitech mouse with Bluetooth support unboxed and test them in place before buying them. Just in case you decide to buy one, I would advise you to buy the iOS compatible version of the mouse, and not the Windows version.

ptxinhua, 2018-08-08:
@bbonmars @mringwal I am using iPad air on IOS 10.1. I hope to try BTC mouse, But I couLd not download it on Cydia.  The app shows incompatibility with ios 10.1 and refused to be downloaded. Could you please tell me how to get the app?

Jonyx4, 2019-01-17:
Any update? it will be really useful to have this feature.

ducanh2002123, 2019-07-15:
BTC Mouse And Keyboard is working on my iPhones 5 (iOS 10.3.3)
My setup: A foldable two in one Keyboard+Trackpad.
The device is detected just fine and after initial connection, everything work well. 
Maybe for people's using iOS 10.3.3, a two in one keyboard would work

mringwal, 2019-07-15:
hi. interesting find. That actually makes sense. The part that is broken on iOS 10 is the code a) make mouse show up in Settings and b) enable pairing. With a keyboard/mouse combo, both work out of the box and BTC M&T then can handle the mouse reports.

bbonmars, 2019-07-15:
Hi what brand and model number is your keyboard/trackpad? I’d like to pick one up too. Thanks!

Brian Weber

ducanh2002123, 2019-07-15:
https://www.amazon.com/Foldable-Keyboard-Portable-Smartphone-Compatible/dp/B077RYCHTJ
Here's the keyboard and trackpad I used

ducanh2002123, 2019-07-15:
But there's one problem: When the iphone is rotated, the pointer doesn't
rotate with the screen, and therefore the control are rotated too. Like
when you rotate your device to the left (Landscape mode), the mouse would
still stay in portrait mode, and if you slide up on the touch pad, the
pointer will go left and vice versa

bbonmars, 2019-07-15:
Thank you for the keyboard and trackpad combo info. I had BTC working with iOS 10.3 and an apple Mighty Mouse but I upgraded to iOS 12.1 and jailbroke, but lost BTC. I’ll try this combo out.

Brian Weber

karlandoh, 2019-09-04:
Any luck anyone?

devgony, 2019-09-05:
> BTC Mouse And Keyboard is working on my iPhones 5 (iOS 10.3.3)
> My setup: A foldable two in one Keyboard+Trackpad.
> The device is detected just fine and after initial connection, everything work well.
> Maybe for people's using iOS 10.3.3, a two in one keyboard would work

how did you install the BTC Mouse tweak on ios 10.3.3?
And what is the way of jailbreak you did?(I did with h3lix)

ducanh2002123, 2019-09-05:
I just install it from Cydia, I use H3lix
Here's the list of all tweaks that got installed:
BTC Mouse & Trackpad (1.9.3)
Bluetooth Companion (also 1.9.3)
BTStack (unknown version)
hid-support
MouseSupport

I think that's it. Then I could just pair it in Bluetooth Setting as a
normal Bluetooth keyboard and have the mouse working as well. Good luck!

dodone31, 2021-09-17:
Hi! I see that @ducanh2002123 installed tbc mouse & trakpad on 10.3.3 device, is the same that i would do on my ipad air that i downgrade from ios 12.5.4 to ios 10.3.3.
after jb(H3lix) i tryed to install tbc mouse&trakpad but cydia tell me that is supported only for ios 9.3.3... I'm really sad and frustrated because this tweaks can give a new life to my old ipad.
My Informatics teacher taught that "in computer science it can be done everything" (I have two in one keyboard/trakpad bought after read this conversation) best regads and sorry for bad english!