bluekitchen/btstack #28: Multiple central and single peripheral

maximzakharov, 2017-04-09:
The manual says: 

> By default, a Bluetooth device stops sending advertisements when it gets into the Connected state. 

> 

How will it change it? And is it possible to connect several central devices to one peripheral (peripheral device with btstack)? Chipset cc2564B.

mringwal, 2017-04-10:
BTstack tries to re-enable advertisements after a connection gets established. On most chipsets, this will be responded with an error.

The CC2564B can listen for advertisements and have a few connections in master/central role. It can only enable advertisements when it's not scanning for advertisements and has no other connections. It cannot be a slave in multiple connections.

The CC2564C supports more than one connection in slave role.

closed 2017-04-10 by mringwal

maximzakharov, 2017-04-10:
OK. If I use CC2564C Service Pack, will the stack work with CC2564C?

mringwal, 2017-04-10:
Yes, it does. You can have multiple LE Peripheral connections, I'm not sure how many, but I've tried at least with two connections. The number of connections probably depends on the connection interval and the supervision timeout -- since an LE controller really must try hard to keep up more than one connection.