bluekitchen/btstack #29: le_streamer examples recursive call

Tsai1005, 2017-04-10:
I found that the le_streamer.c use ATT_EVENT_CAN_SEND_NOW to trigger another att_server_request_can_send_now_event, this will cause the recursive call, I wonder is that properly?

mringwal, 2017-04-10:
It depends on the HCI transport implementation. Most are asynchronously, so the first ATT_EVENT_CAN_SEND_NOW will start sending an ACL packet to the Bluetooth chipset and the call to att_server_request_can_send_now_event will not create a new can_send_now event since the HCI transport is still busy.

If the HCI transport is synchronous, this will cause repeat until the outgoing buffer in the Bluetooth controller is full.

Could this a problem for your application?

Tsai1005, 2017-04-10:
First about asynchronously, as you said "the call to
att_server_request_can_send_now_event will not create a new CAN_SEND_NOW
event", but how to make the le_streamer works? lacking the event driver,
the example will stop at once, did I miss something?
Second about synchronous， if I call
att_server_request_can_send_now_event，it finally depend on the HCI layer（Is
that any ACL packets ready?）if it is valid， then will cause a new
CAN_SEND_NOW, so that is so-called "synchronous" implementation? Thx

mringwal, 2017-04-10:
1. "the call to att_server_request_can_send_now_event will not create a new CAN_SEND_NOW event" right now. However, when the ACL packet is sent to the Bluetooth chipset, the requested ATT_EVENT_CAN_SEND_NOW will be generated.

2. A synchronous implementation of HCI Transport does not have a 'can send now' implementation. If the call to send_packet() is blocking/synchronous, it cannot call can_send_packet_now().. This distinction might be confusing, I didn't remember it myself. Most implementations are async and can_send_now is provided.

Tsai1005, 2017-04-11:
I understand the distinction about blocking & non-blocking implementation of HCI Transport,  I 'm clear now, I miss the l2cap_notify_channel_can_send in the l2cap layer HCI_EVENT_NUMBER_OF_COMPLETED_PA CKETS case, that works, thx a lot!

closed 2017-04-11 by Tsai1005

mringwal, 2017-04-11:
excellent. good luck!