bluekitchen/btstack #30: how to accomplish the power up sequence

Tsai1005, 2017-04-24:
I try the port/windows-h4 when I want to use chipsets（cc2564） to boot up the bt stack， how to implement the power up sequence mentioned in the spec “Dual-Mode Bluetooth_ CC2564 ModuleEvaluation Board.pdf”

mringwal, 2017-04-24:
On my CC256x test boards (an USB-2-UART + some adapter + the CC256x), I have a RESET button that I press before running any of the examples. The RESET button is connected to GND on one side and the N_SHUTDOWN pin on the other. Pressing this RESET button, does a power cycle of the module.

Tsai1005, 2017-04-25:
I figure out  the problem is not correspond to the power-up process，my environment setup like ：USB-Serial（CH340） + CC2564，and run the window-h4 examples，I wanna to verify is there any output from the USB-Serial（COM20 in my computer）and connected the COM20 to the other COM to checked，but nothing output，that means the HCI H4 doesn't work

Tsai1005, 2017-04-25:
I should not use the flow control，it run ok now，thx！

closed 2017-04-25 by Tsai1005

mringwal, 2017-04-25:
I'm glad you got it to start up. In general, it should also work with Flow Control. On a desktop system, well, you don't really need it though.