bluekitchen/btstack #31: SPP User Confirmation Failure

ismailhkose, 2017-04-25:
I am having trouble with SPP User Confirmation code
I have been using the most recent version of the BTStack (commit-id:64f92e5791eca3562fa09fa9b77bf55c6c013783).

This code is only appears after first pairing and never appears next times, unless unpair and uninstall the BT from Windows 10.

I could reproduce this issue at Windows 10 and Windows 7 using spp_counter.c  example. Once I pair with my BT device, i can get data from serial port well. When I reboot the device and try to open the serial port it try to connect. Somehow Windows 10 does not pop up BT device confirmation request. That's why connection attemption is getting failed. I dont know if this is Windows OS issue or BT stack issue. I didnt have such an issue with other BT devices.

I also tried to completely disable authentication, but the following function doesnt do anything in SPP mode. It generates and try to get confirmation.

sm_set_authentication_requirements(0)

Any idea?

Thanks

mringwal, 2017-04-26:
Hi. Could you describe your setup? On what hardware/software/OS are you running BTstack?

The SSP User Confirmation code is only shown during the pairing, to make sure that there's no man-in-the-middle attack going on. The pairing generates a link key and this link key will be used for subsequent connections. There should be no confirmation code popping up.

ismailhkose, 2017-04-26:
Hi,
I am running it at ARM Cortex M4F processor with bare metal system.

ismailhkose, 2017-04-28:
How can I disable security? The following function doesnt disable pairing.
sm_set_authentication_requirements(0);

mringwal, 2017-04-28:
If the remote side wants to pair, there's not much you can do. In theory, the SDP query should happen without encryption, but any other protocol should use encrypted connections. I think you can/could try to open an RFCOMM connection without pairing on Android via a special command. On Windows, I think there was/has been an option to do RFCOMM without pairing.

mringwal, 2017-04-28:
weird I only saw the "I am running it at ARM Cortex M4F processor with bare metal system." without the rest in the notification mail.

anyway, you can enable packet logging with hci_dump_open( NULL, HCI_DUMP_STDOUT ) in main(). This will slow you down, but it's a good way to check for errors. If that works at max possible speed, it usually is also stable without packet log at real full speed. The packet log will allow to check e.g. why the log_error shows up - was the rfcomm connection already closed, etc.

ismailhkose, 2017-04-28:
Thanks Matthias

mringwal, 2017-05-08:
I'm closing this, feel free to post another issue if needed.

closed 2017-05-08 by mringwal

ismailhkose, 2017-05-15:
Thanks Matthias,
Should RFComm server save link key on micro controller and use it to auto pair at next device reboot?

When i reboot the device windows cant connect unless remove the bt device with the driver and install it again.

Since whenever Windows 7/10 connects, it wants to establish a secure connection. I dont know if there is a way to disable that.

ismailhkose, 2017-05-15:
Thanks for your support

mringwal, 2017-05-15:
yes, the link key should be stored somewhere (e.g. in flash). for his, you need to implement the src/classic/btstack_link_key_db.h interface.