bluekitchen/btstack #32: SPP data duplication and drop and how to debug it

ismailhkose, 2017-05-02:
Hello,

I am trying to send some data to Android App using SPP at high speed, My current data throughput is ~20kb.

But, I am having trouble with duplicated data. I have a counter for each string that is sent at once.
I am seeing some of the data is transferred 3-5 times, some of them are dropped.

rfcomm_send sends data and always returns 0 even though packet is dropped. Is there a way to know if packet successfully transferred or not?

Why a packet is sent multiple times to SPP client? How can I debug this?

I derived my test application from SPP Counter and implemented top of that, Heart beat perioid is 8ms and have the following configuration.

#ifndef __BTSTACK_CONFIG
#define __BTSTACK_CONFIG
#include <stdint.h>
// Port related features
#define HAVE_INIT_SCRIPT
#define HAVE_EMBEDDED_TIME_MS
//#define HAVE_POSIX_TIME
//#define HAVE_MALLOC
// BTstack features that can be enabled

#define ENABLE_BLE
#define ENABLE_CLASSIC

#define ENABLE_LE_PERIPHERAL
#define ENABLE_LE_CENTRAL
//#define ENABLE_LOG_INTO_HCI_DUMP
//#define ENABLE_LOG_DEBUG
#define ENABLE_LOG_ERROR
//#define ENABLE_LOG_INFO
//#define ENABLE_SDP_DES_DUMP
//#define ENABLE_SDP_EXTRA_QUERIES
// #define ENABLE_LE_SECURE_CONNECTIONS

// BTstack configuration. buffers, sizes, ...

// BTstack configuration. buffers, sizes, ...
//#define HCI_INCOMING_PRE_BUFFER_SIZE 6
#define HCI_ACL_PAYLOAD_SIZE 1021
#define MAX_SPP_CONNECTIONS 1
#define MAX_NR_HCI_CONNECTIONS MAX_SPP_CONNECTIONS
#define MAX_NR_GATT_CLIENTS 0
#define MAX_NR_GATT_SUBCLIENTS 0
#define MAX_NR_HFP_CONNECTIONS 0
#define MAX_NR_L2CAP_SERVICES  2
#define MAX_NR_L2CAP_CHANNELS  (1+MAX_SPP_CONNECTIONS)
#define MAX_NR_RFCOMM_MULTIPLEXERS MAX_SPP_CONNECTIONS
#define MAX_NR_RFCOMM_SERVICES 1
#define MAX_NR_RFCOMM_CHANNELS MAX_SPP_CONNECTIONS
#define MAX_NR_BNEP_SERVICES 1
#define MAX_NR_BNEP_CHANNELS MAX_SPP_CONNECTIONS
#define MAX_NR_BTSTACK_LINK_KEY_DB_MEMORY_ENTRIES  2
#define MAX_NR_WHITELIST_ENTRIES 1
#define MAX_NR_SM_LOOKUP_ENTRIES 3
#define MAX_NR_SERVICE_RECORD_ITEMS 1

#define MAX_NR_LE_DEVICE_DB_ENTRIES 1

ismailhkose, 2017-05-02:
When I enable debug logsö I am seeing the following messages. Looks like there is no available buffer.

How to know reason of duplication data?
I need to find an API that checks ACL buffer availability. I assumed _rfcomm_send_ API checks if buffer available and expect it to be return with fail message. But, it does not.

[00:04:54.595] LOG -- hci.c.350: ACL classic buffers: 4 used of 4
[00:04:54.601] LOG -- hci.c.350: ACL classic buffers: 4 used of 4
[00:04:54.607] LOG -- hci.c.350: ACL classic buffers: 4 used of 4

ismailhkose, 2017-05-03:
Never mind, the problem is solved.

closed 2017-05-03 by ismailhkose

mringwal, 2017-05-03:
Glad you solved it somehow. You can look at example/spp_streamer.c 

For basic questions, you may post on the mailing list in the hope that it could be interesting to more people.

There's a bit of info on RFCOMM sending here: http://bluekitchen-gmbh.com/btstack/develop/protocols/#rfcomm-radio-frequency-communication-protocol

In general, as long as you have data to send, you request the can send now event, then send when you get it and request another can send now event. -- all buffering a single packet - there's only one outgoing buffer to assemble HCI packets.