bluekitchen/btstack #33: ST Nucleo BLE Module support?

mgiaco, 2017-05-08:
Hello,

I am wondering if there is any plan to support the http://www.st.com/en/wireless-connectivity/bluenrg.html chipset. 

Because there are some interesting modules available like this one. http://www.st.com/content/st_com/en/products/ecosystems/stm32-open-development-environment/stm32-nucleo-expansion-boards/stm32-ode-connect-hw/x-nucleo-idb05a1.html

cheers
mathias

mringwal, 2017-05-08:
Hi Mathias. The BlueNG module, like most BLE only modules, does not support an HCI interface, so at best they have some proprietary interface to control it. 

We're working with Nordic to provide an HCI compatible interface for their nRF5x chipsets: https://github.com/bluekitchen/btstack/tree/master/port/nrf5-zephyr

We also just finished a port of BTstack for the STM32 F4 Discovery board with an adapter from TI to use the TI CC256x: https://github.com/bluekitchen/btstack/tree/develop/port/stm32-f4discovery-cc256x

Hope that helps.
  Best Matthias

mringwal, 2017-05-08:
Following up: I did click a bit through the documentation and it looks like the BlueNG module does offer  (most) HCI commands via SPI. So commands should work. Now, we also would need to get ALL events and data from HCI. I didn't see how to control that. Might be worth a try though..

mgiaco, 2017-05-09:
Okay I understand. So there is a whole SW lib available I think I will try that first.

Yulong-espressif, 2017-09-28:
@mringwal Does the blueStack support bluetooth5.0 statck now?

mringwal, 2017-09-28:
@Yulong-espressif The Bluetooth Spec is backwards compatible in most cases, so BTstack and also Bluedroid could be qualified against the Bluetooth V5.0 specification. BTstack support LE Secure Connections and the Date Length Extension from the V4.2 spec (and the ESP32 has support for DLE, too). Bluetooth5 has new extended advertisements and new BLE radio modes (2mbit and long-range). BTstack does not support them yet, but I don't think the ESP32 BT lib supports it yet either. (Or did you ask for something else?)

Yulong-espressif, 2017-09-28:
@mringwal Thank you for your reply very much, I have download the btstack code, it look like very small, it is very cool. So can we push the code for improve the btStack later?

mringwal, 2017-09-28:
@Yulong-espressif Where do you want to push the code? What do you want to improve?
Feel free to contact me directly at contact@bluekitchen-gmbh.com

Yulong-espressif, 2017-09-28:
@mringwal OK, I will send e-mail to you later.