bluekitchen/btstack #34: Undefined btstack_run_loop_freertos_trigger_from_isr with certain FreeRTOS settings.

DvdSpijker, 2017-05-08:
When INCLUDE_xEventGroupSetBitFromISR is not defined in the FreeRTOS config the function 
"btstack_run_loop_freertros_trigger_from_isr"  (btstack_run_loop_freertos.c)  is not defined due to a pre-compiler condition. However, this function is being used without such a condition in btstack_uart_block_freertos.c, resulting in a compiler error.

mringwal, 2017-05-08:
Good catch. I can add the same condition around the "btstack_run_loop_freertros_trigger_from_isr" in btstack_uart_block_freertos.c but I would need to add an #error "btstack_uart_block_freertos requires that INCLUDE_xEventGroupSetBitFromISR is defined in the FreeRTOS config". 

I'm new to FreeRTOS. In this case, we get a callback from an ISR and need to trigger the run loop. Is there another way to do this without xEventGroupSetBits?

DvdSpijker, 2017-05-08:
Hi Mathias,
As far as I know Eventgroups are the way to go for this purpose, but I am not a profound FreeRTOS user myself either.

mringwal, 2017-05-09:
hi. EventGroups looked like the most appropriate way to trigger the run loop without using a finite resource (mutex, queue, ..).

I've added a compile error to explain it: https://github.com/bluekitchen/btstack/commit/92f3d82f1ec49599fe24e3a0a18ec2575d4a839a

closed 2017-05-09 by mringwal