bluekitchen/btstack #37: ESP32 panu_demo Windows Build Failure

GeorgeIoak, 2017-05-24:
Just did a new setup on my Win7x64 machine and tried to compile the panu_demo examples and got this error message:

```
...
CC panu_demo.o
D:/GitHub/btstack/port/esp32/panu_demo/main/panu_demo.c:58:21: fatal error: ifaddrs.h: No such file or director
compilation terminated.
make[1]: *** [/home/esp-idf/make/component_wrapper.mk:177: panu_demo.o] Error 1
make: *** [D:/msys32/home/esp-idf/make/project.mk:386: main-build] Error 2

George@George-Office2 MINGW32 /d/GitHub/btstack/port/esp32/panu_demo
```

mringwal, 2017-05-24:
hi. the panu_demo currently support OS X & Linux. It can receive/send Ethernet packets via BNEP, but lacks the ability to forward it on the ESP32. To support the ESP32, this mechanism would need to get implemented. The code of Benjamin Aginers 6LowPAN implementation should be quite similar to this (also needs to forward Ethernet packets)

GeorgeIoak, 2017-05-24:
Thanks for clarifying that. I've got a new setup and I wanted to confirm that everything was installed properly before I moved forward. Maybe I'm just having a bad day but I just tried the gatt_battery_query example and I get this error:
`D:/GitHub/btstack/port/esp32/gatt_battery_query/main/gatt_battery_query.c:52:32: fatal error: gatt_battery_quer      y.h: No such file or directory
compilation terminated.
make[1]: *** [/home/esp-idf/make/component_wrapper.mk:177: gatt_battery_query.o] Error 1
make: *** [D:/msys32/home/esp-idf/make/project.mk:386: main-build] Error 2
`

mringwal, 2017-05-24:
in all examples that have a GATT DB, there's an update_gatt_db.sh shell script. I'm afraid that doesn't work on Windows (although it should work in msys/mingw). It just calls a Python script, you may run that manually. It will generate gatt_battery_query.h.

GeorgeIoak, 2017-05-24:
That did create gatt_battery_query.h but then I get new errors:

`In file included from d:\github\btstack\src\btstack_util.h:58:0,
                 from d:\github\btstack\src\btstack_control.h:49,
                 from d:\github\btstack\src\btstack.h:55,
                 from D:/GitHub/btstack/port/esp32/gatt_battery_query/main/gatt_battery_query.c:51:
D:/GitHub/btstack/port/esp32/gatt_battery_query/main/gatt_battery_query.c: In function 'btstack_main':
d:\github\btstack\src\btstack_defines.h:51:32: warning: 'sizeof' on array function parameter 'argv' will return size of 'const char **' [-Wsizeof-array-argument]
 #define UNUSED(x) (void)(sizeof(x))
                                ^
D:/GitHub/btstack/port/esp32/gatt_battery_query/main/gatt_battery_query.c:337:5: note: in expansion of macro 'UNUSED'
     UNUSED(argv);
     ^
D:/GitHub/btstack/port/esp32/gatt_battery_query/main/gatt_battery_query.c:318:41: note: declared here
 int btstack_main(int argc, const char * argv[]){
                                         ^
AR libmain.a
LD gatt_battery_query.elf
esptool.py v2.0-beta3
To flash all build output, run 'make flash' or:
python /home/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/tty.usbserial-DN02B3PF --baud 1000000 --before default_reset --after hard_reset write_flash -u --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /d/GitHub/btstack/port/esp32/gatt_battery_query/build/bootloader/bootloader.bin 0x10000 /d/GitHub/btstack/port/esp32/gatt_battery_query/build/gatt_battery_query.bin 0x8000 /d/GitHub/btstack/port/esp32/gatt_battery_query/build/partitions_singleapp.bin
`
I'm beginning to wonder if I should start over and create a Ubuntu VM and reinstall everything there??

mringwal, 2017-05-24:
looks good. I did fix that warning just yesterday on the developer branch. you may try it now with "make flash" and "make monitor". Another example good for a feirst test is "le_counter". Just connect with any BLE Explorer tool on Android or iOS>

GeorgeIoak, 2017-05-24:
OK great, just thought I'd let you know about that warning. I need to integrate Benjamin Aginers' project with this now and get that to compile and then I'll be running some initial tests on hardware.

Any helpful pointers on doing that?

mringwal, 2017-05-25:
Benjamin's project does send Ethernet packets via BLE Connection-oriented channels. PAN sends them via Classic channels. The way packets are injected/received from the ESP32's IP stack should be the same. I don't know anything about TCP/IP on ESP32. 

Pointers: BTstack Manual, esp. architecture and how to debug problems via packet logger.

Feel free to post general "how to btstack" questions on the BTstack Mailing List (google dev group)

GeorgeIoak, 2017-05-25:
@mringwal May I contact you offline, perhaps through Blue Kitchen?

mringwal, 2017-05-25:
@GeorgeIoak like, visit us in Zurich? :) 
Sure, you may write to contact@bluekitchen-gmbh.com and we can take it from there.

GeorgeIoak, 2017-05-25:
OK, I'll send a short email soon, thanks. Unfortunately I'm in California so I can't just come knock on your door!

closed 2023-02-21 by mringwal