bluekitchen/btstack #38: Add IPSP PSM

benjaminaigner, 2017-05-29:
Hi @mringwal ,

here is a patch to add the IPSP PSM, afterwards it should work out-of-box with RFC7668.
[0001-Added-IPSP-PSM-for-L2CAP.patch.txt](https://github.com/bluekitchen/btstack/files/1036306/0001-Added-IPSP-PSM-for-L2CAP.patch.txt)

BTW: remove .txt, than you have the patch (no idea why it is not possible to upload one single patch here...)

mringwal, 2017-05-30:
thanks for the one-line patch. I started to get rid of manually added numbers in bluetooth.h in favor of writing more Bluetooth SIG website scrapers. I'll try to find out where this PSM is officially coming from - and if not just add that line.

GitHubs idea of "submitting a patch" is:
- fork the whole project
- create a bug fix branch
- do the fix
- commit the fix
- push the branch
- create a pull request

On the positive side: if I fully agree with your pull request, it's a single button click to accept. No command line manoeuvring and such. If not, you update your branch and we continue.

mringwal, 2017-05-30:
ah.. it's https://www.bluetooth.com/specifications/assigned-numbers/logical-link-control

mringwal, 2017-05-30:
don't feel like web site scraping now, I've manually added the PSM as suggested to develop (will merge to master eventually, too).

closed 2017-05-30 by mringwal