bluekitchen/btstack #39: Preprocessor - Feature activation -> also use in header files

benjaminaigner, 2017-05-29:
If BTStack is configured to omit one feature (e.g.: ENABLE_LE_DATA_CHANNELS), there are still all prototypes available in the headers --> other source files compile nice, BUT: undefined reference during linking.

Suggestion:
Disable function headers in header files too, might be easier to find disabled stack parts.

mringwal, 2017-05-30:
good point. I was wondering about what to do with the prototypes. Adding #ifdefs makes them uglier, but compile errors are easier to understand than link errors in my opinion. I guess I'll follow your suggestion here.

benjaminaigner, 2017-05-30:
Yes, #ifdefs are not beautiful in header files, but it's much more easy with them :-)