bluekitchen/btstack #40: hci free acl slot

Tsai1005, 2017-06-25:
The HCI layer query the free slot is valid in every sent action, but in dual-mode, when no free slots classic then make the le can't sent also (if le_acl_packets_total_num available), is it weird?

mringwal, 2017-06-25:
Hi! Could you explain this a bit more please? I don't understand.

Tsai1005, 2017-06-26:
Check the code first, here are the code i am wondering:
`static int hci_number_free_acl_slots_for_connection_type(bd_addr_type_t address_type){
    
    unsigned int num_packets_sent_classic = 0;
    unsigned int num_packets_sent_le = 0;

    btstack_linked_item_t *it;
    for (it = (btstack_linked_item_t *) hci_stack->connections; it ; it = it->next){
        hci_connection_t * connection = (hci_connection_t *) it;
        if (connection->address_type == BD_ADDR_TYPE_CLASSIC){
            num_packets_sent_classic += connection->num_acl_packets_sent;
        } else {
            num_packets_sent_le += connection->num_acl_packets_sent;
        }
    }`
In every "hci_send_acl_packet_fragments" (the basic function for upper layer to send ACL PDUs & SDUs, ex L2cap),first will check is any free slots for used, so, free slots refer to free classic slots & free le slots, my questions is about the judgements , why no more free classic slots but free le slots will cause the fail of "hci_can_send_acl_packet_now".

mringwal, 2017-06-26:
Do you refer to this:

`if (free_slots_classic < 0){
    log_error("hci_number_free_acl_slots: outgoing classic packets (%u) > total classic packets (%u)", num_packets_sent_classic, hci_stack->acl_packets_total_num);
    return 0;
}`

This is an error check: it checks for free_slots_classic < 0, which is an error. If there are no free slots, then free_slots_classic should be == 0. 

We never should send more ACL packets then there are slots in the Bluetooth Controller.

Does this code get executed for you? What Bluetooth Controller do you use?

Tsai1005, 2017-06-27:
Yes， i got your point and I run the example like “le streamer” with the CSR dongle， at the beginning ，HCI use the command “HCI_READ_BUFFER_SIZE” to get classic acl_packet_length（310 Bytes） & acl_packet_num（10） ，but in respond to “HCI_READ_LE_BUFFER_SIZE” the Controller return all 0， which means that the Host in Dual-mode share the classic buffer， but if the Controller have le buffer，  what will happen？if run out of the classic buffer then the le buffer should be ignore？Just like assumes some priority between this two buffer in usage。

mringwal, 2017-06-27:
If there are only Classic buffer or only LE buffers, everything is clear, right?

For the mixed case, I think I didn't see a Bluetooth Controller that does that, and did not add additional logic to deal with it. It's also not clear how to figure out if an ACL for an LE connection is stored in the LE buffers or in the Classic buffers.

If you have an example for such a controller, please let us know.

closed 2017-08-09 by Tsai1005