bluekitchen/btstack #42: NRF-5 stack unclear with existing zephyr-hci over uart modules

falkorichter, 2017-06-26:
Hi, we have hardware with nrf51822 with zephyr already deployed. The `port/nrf5-zephyr` code expects me to flash the nrf chip as part of the process, but the chips are already provisioned.

How Do I compile/run only the host version of the btstack application to connect to my HCI over UART bluetooth chips?

mringwal, 2017-06-26:
When you install the hci_uart example from Zephyr, the nRF51822 basically becomes a regular Bluetooth Controller with an H4 UART interface and you can either use port/posix-h4 for Linux & macOS or port/windows-h4 for Windows to use them. port/nrf5-zephyr is about running BTstack on top of the Link Layer provided by Zephyr on the nRF5 chipset directly.

Now, you need to find out the correct path to the serial port where the nRF51 chipset is connected - either COMx on Windows, or /dev/tty* on macOS or Linux. The default baud rate is 1 mbps. if you've followed our advise and used our patch from https://github.com/bluekitchen/btstack/tree/master/chipset it will run at the default 115200.

Please update the baud rate (if running at 1 mbps) in the transport config (beginning of file, "initial baud rate") and the path to the serial port/UART in the main(..) function of the main.c of the port you want to use and compile the examples. Then, just start one, e.g. LE Counter is quick to test.

Please let me know if you got them up and running. If not, please provide hci_dump (same dir on Windows, or /tmp/hci_dump.pklg for unix-ish systems).

mringwal, 2017-09-18:
We did add a new [posix-h4-zepyr](https://github.com/bluekitchen/btstack/tree/master/port/posix-h4-zephyr) port, that works with the unpatched HCI UART Firmware from Zephyr project (at 1 mbps, and provided a README there as well.

closed 2017-09-18 by mringwal