bluekitchen/btstack #43: Does btstack support RTL8761?

echosphor, 2017-06-27:
Hi I can not find RTL8761ATV in "Supported Chipsets" List.
Is there any plan support it?
thanks.

mringwal, 2017-06-27:
Hi. We'd be happy to support it (or any other not-yet-supported Bluetooth Controller) in general.

If you can provide us with a dev kit of the RTL8761 (or anything else that I can connect to my Mac) or let us know where we can buy one, and the necessary information on non-standard HCI Commands like configuring HCI over SCO, setting BD_ADDR (if not unique), and setting baud rate, we're glad to add support for it.

echosphor, 2017-08-02:
thank you for your prompt reply.

Sorry,but we don't have another RTL8761 dev board.May be it's hard to buy in Europe.I'll let you know if there some other way you can test it.

mringwal, 2023-02-21:
Since mid 2022, there's support for Realtek controller. Feel free to re-open if that doesn't work.

closed 2023-02-21 by mringwal