bluekitchen/btstack #44: [ESP32] Need to repair device after resetting the board

anteph, 2017-07-06:
Hi! 

I'm using the ESP32 port of the btstack and I'm running into an issue.

Basically, when I use, for example, the SPP counter program, I can pair it with the computer and exchange data with a Python application running PyBluez. 

Nevertheless, if I reset the device, it keeps being identified as paired by Windows but I'm no longer able to connect to it on the application.  I can still find the advertised service on my Python program but when I try to connect using a PyBluez Bluetooth socket, it refuses the connection..

The only way I found to solve this is by forgetting the device and repairing it again.

The thing is that this just happens if I reset the ESP32 and the bluetooth program runs from the start. If I leave the ESP32 running the program and I try to connect multiple times from my PC application, it works fine.

I'm just getting started with the BTStack and I don't have a deep understanding of the bluetooth stack and profiles, so maybe I'm doing something wrong.

I'm testing this on Windows 8.1. 

Best regards,
Nuno Santos

mringwal, 2017-07-07:
Hi Nuno.

You're not doing anything wrong. What happens during pairing, is that both sides agree on a link key for encryption.

The ESP32 port does not have support to to store link keys in Flash memory. By default, they are kept in RAM and will be gone after a Reset.

I've recently introduced a new Link key storage system that relies on a minimal Hardware Abstraction Layer to store link keys in Flash. It's [btstack_link_key_db_tlv.c] (https://github.com/bluekitchen/btstack/blob/master/src/classic/btstack_link_key_db_tlv.c) - where TLV stands for Tag-Length-Value, a common way to store binary data in memory.

To use it, you'd need to implement the [hal_flash_sector.h](https://github.com/bluekitchen/btstack/blob/master/platform/embedded/hal_flash_sector.h) interface.

I did not look into if/how Flash is used in the esp-idf. Maybe a few lines are enough.

There's also the [Link Key db for WICED](https://github.com/bluekitchen/btstack/blob/master/platform/wiced/btstack_link_key_db_wiced_dct.c) that might be suitable as well. It only uses wiced_dct_read_lock, wiced_dct_read_unlock, wiced_dct_write. I recently looked into this and found that it's quite heavy in the (hidden) implementation below, so the btstack_link_key_db_tlv.c is the better way.

anteph, 2017-07-08:
Thank you very much for your answer! 

I will take a look if I can solve it. Nevertheless, if I don't manage to do it, are you planning to include this feature on any future release of the ESP32 port?

Best regards,
Nuno Santos

mringwal, 2017-07-11:
Hi. Yes, we will certainly implement this soon-ish as there's clear interest in using the ESP32 commercially with BTstack.

anteph, 2017-07-13:
I will be waiting for it :) Meanwhile I will keep exploring other functionalities of the library. Thank you very much for porting it to the ESP32 and making available a very powerful Bluetooth library for the community.

Best regards,
Nuno Santos

mringwal, 2017-07-21:
Hi Nuno. Please try the current develop branch. There's a quick implementation of btstack_tlv.h on top of ESP-IDS's NVS Storage.

closed 2017-07-21 by mringwal

anteph, 2017-07-22:
Hi Matthias,

Thank you very much for this! It will be very useful for all the ESP32 community :) 

I will test it and let you know about the results.

Best regards,
Nuno Santos

mringwal, 2017-07-22:
No problem. I want to write a blog post about storing key value pairs (as tag-length-value) in Flash, as a basis for storing link keys and other persistent data. The ESP32 Flash interface is more advanced, so implementing mine using their API was simple. So, what I've created isn't really new, but at the same time, it's an indicator that I've picked a useful abstraction. (more on it in 1-2 weeks hopefully)

Thanks for writing about BTstack on ESP32 in your blog!