bluekitchen/btstack #48: Btstack port for ESP32 and WiFi problem

my-abousamra, 2017-07-19:
Hi,
I used btstack port for ESP32 example SPP_Counter and It worked well then I added my WiFi application (3 tasks) to it which connects to a server and uploads data, When I ran it WiFi not worked correctly and station got ip event wasn't received and hence can't connect to server and when I connect bluetooth WiFi disconnects and can't connect again.

Now, Can't btstack work with WiFi and causes this strange behaviour?

Regards

mringwal, 2017-07-19:
hi. Could you provide an example project that we can try? I've never tried Wifi, so I don't know if Bluetooth and Wifi can work together. They should :)

my-abousamra, 2017-07-19:
Hi,
I know they should so I made that merge.

The merged project consists of 3 components bt classic, wifi, uart. uart task send data to wifi (sequence data) and bt classic (latter 'a').
The project receives data from uart but for testing purpose I generated data internally from the module itself

Notes:
1.  Change btstack path in component.mk file of Bt_Stack component to yours.
2.  Change ssid and password in wifi_vJoin_To_AP function in wifi.c to yours (line 236)
3.  Change host address (gu8Server_Addr in wifi.c)

Feel free to ask about anything that is not clear in the project

example project

[spp_counter_2.zip](https://github.com/bluekitchen/btstack/files/1159145/spp_counter_2.zip)

server (open cmd and run "java -jar seq-server-printall.jar")

[seq-server-printall.jar.zip](https://github.com/bluekitchen/btstack/files/1159144/seq-server-printall.jar.zip)

Regards,

my-abousamra, 2017-07-19:
Hi,
I know they should so I made that merge.

The merged project consists of 3 components bt classic, wifi, uart. uart task send data to wifi (sequence data) and bt classic (latter 'a').
The project receives data from uart but for testing purpose I generated data internally from the module itself
Notes:

1. Change btstack path in component.mk file of Bt_Stack component to yours.
2. Change ssid and password in wifi_vJoin_To_AP function in wifi.c to yours (line 21)
3. Change host address (gu8Server_Addr in wifi.c)

Feel free to ask about anything that is not clear in the project

example project
[spp_counter_2.zip](https://github.com/bluekitchen/btstack/files/1159402/spp_counter_2.zip)

server (remove .zip extension and open cmd and run "java -jar seq-server-printall.jar")

[seq-server-printall.jar.zip](https://github.com/bluekitchen/btstack/files/1159294/seq-server-printall.jar.zip)

Regards,

mringwal, 2017-07-19:
thanks will try, but probably not before start of next week. I'll ask if anything is unclear.

xuxuchenwei, 2017-07-24:
hi,my-abousamra
Why can not I compile spp_counter_2 correctly in esp-idf?

CC Bt_Classic.o
/home/xucw/code/esp-idf-master_20170721/examples/bluetooth/spp_counter_2/components/Bt_Classic/./Bt_Classic.c:19:21: fatal error: btstack.h: No such file or directory
 #include "btstack.h"
                     ^
compilation terminated.
make[1]: *** [Bt_Classic.o] Error 1
make: *** [Bt_Classic-build] Error 2
Did you use btstack.h?

mringwal, 2017-07-24:
@xuxuchenwei Please don't post on unrelated issues and create a new issue for your problem next time.

It looks like you've copied spp_counter to your esp-idf folder. BTstack expects the example to be at btstack/port/esp32/ though and won't find it's sources. You may fix the Makefile and set BTSTACK_ROOT to point to your BTstack root folder.

mringwal, 2017-08-10:
Hi

I've compiled you example and entered my wifi config + computer ip. I did start the java server, too.
When I start the esp32 app, I only get "BTstack up and running".  (I pressed different keys..)

What should happen next?
Is the other problem with thread synchronization related? Is this issue alive?

Console output...

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:4100
load:0x40078000,len:0
load:0x40078000,len:11968
entry 0x40078cd8
I (1188) heap_init: Initializing. RAM available for dynamic allocation:
I (1189) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1208) heap_init: At 3FFCE798 len 00011868 (70 KiB): DRAM
I (1266) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1325) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1385) heap_init: At 40095E0C len 0000A1F4 (40 KiB): IRAM
I (1777) UART_vInit:
I (1779) Bt_Classic_vInit: SDP service record size: 95

I (2015) WiFi_vInit:
I (2016) wifi: wifi firmware version: 0e33bec
I (2017) wifi: config NVS flash: enabled
I (2017) wifi: config nano formating: disabled
I (2043) wifi: Init dynamic tx buffer num: 32
I (2043) wifi: Init data frame dynamic rx buffer num: 32
I (2043) wifi: Init management frame dynamic rx buffer num: 32
I (2047) wifi: wifi driver task: 3ffdca68, prio:23, stack:4096
I (2052) wifi: Init static rx buffer num: 10
I (2056) wifi: Init dynamic rx buffer num: 32
I (2060) wifi: Init rx ampdu len mblock:7
I (2064) wifi: Init lldesc rx ampdu entry mblock:4
I (2069) wifi: wifi power manager task: 0x3ffe2254 prio: 21 stack: 2560
I (2075) uart_send_task: uart_send_task start
I (2075) uart_recv_task: uart_recv_task start
I (2085) bt_classic_monitor_task: bt_classic_monitor_task start
I (2085) bt_classic_send_task: bt_classic_send_task start
BTstack up and running.

petemadsen, 2017-10-17:
Don't you if this is still an issue. Bluetooth and WiFi working for me. Code: https://github.com/petemadsen/alarma2/tree/master/esp32/src

mringwal, 2017-10-18:
Hi @petemadsen How well Bluetooth & Wifi works is unclear to me. Does Wifi work well for you when there's a Bluetooth connection at the same time?

petemadsen, 2017-10-18:
Don't know, I didn't do any stress test so far. I was sporadically access it thru browser and my app. Connecting, disconnecting. Could not see any problems.

I can do some more tests on Friday, so hang on.

petemadsen, 2017-10-20:
Some impressions: well, it looks like WiFi gets into trouble when there is a BT connection. In my case, BT was send a short text message once a second.

my-abousamra, 2017-12-19:
Sorry for the late reply.

The problem still exist, after more investigation BTStack and WiFi can work together only if no one interrupts the other work but this is suitable for low data rates which worked fine for me but for higher rates which take longer time to send interruption may occur.

When sending WiFi data (using send function) and BTStack sends data at that time, send function blocks for a very long time and may unblock after closing BTStack.

I don't know why send function blocks when BTStack works, normally it blocks when sending buffers are full and unblocks when they become free. Actually I don't know Bluetooth and WiFi protocols internals  and how they can coexist at the same frequency range and at the same module and how the module manage the shared resources to avoid such cases. So, Does BTStack use any resource that may be used for WiFi and causes it to block for it for a long time.

Example code is available to regenerate the case if you interested.

my-abousamra, 2018-01-18:
according to this post and others
https://esp32.com/viewtopic.php?f=2&t=3364
this problem seems to be related to ESP32 module not Btstack and ESP32 team is working on it

Thanks for your help

closed 2018-01-18 by my-abousamra

petemadsen, 2018-01-18:
idf now support bluetooth classic rfcomm: https://github.com/espressif/esp-idf/issues/626