bluekitchen/btstack #51: ESP32 SDK sdp_init conflict

lefedor, 2017-08-01:
Hi, have experimenting with esp32 and bumped to issue:

`
src/classic/sdp_server.c:78: multiple definition of `sdp_init'
/opt/esp/btstack-master/port/esp32/a2dp_source_demo/build/bt/libbt.a(sdp_main.o):/opt/esp/esp-idf/components/bt/bluedroid/stack/sdp/sdp_main.c:82: first defined here

`

Seems like recently sdk was updated and now there are intersection in namespace.

mringwal, 2017-08-01:
Hi. Did you follow the readme and tried t compile one of the exampes?

The conflicting file is from the default Bluedroid Bluetooth stack. It should be disabled in the sdkconfig file as CONFIG_BLUEDROID_ENABLED is not defined. Please verify this and also if you have a current esp-idf version.

asuivelentine, 2017-08-01:
Hi.
I got the same behavior today when compiling the example (spp-counter and spp-flowcontrol). It was a fresh clone of both, btstack and esp-idf.

`make menuconfig` setup: enabled Classic Bluetooth and disabled everything else

however it was really strange: I compiled the spp_counter example yesterday and it works. I also thought they changed something in the framework. However resetting the last commit(s) didn't fix that.
 
personal quickfix: rename sdp_init() function in `src/classic/sdp_server`

mringwal, 2017-08-01:
Thanks for reporting. I'm on vacation this week and can check next week, to see what's going on. It surely worked like e.g. two weeks ago or so.

lefedor, 2017-08-01:
Hello, have actual idf && btstack from GIT.
Thanks for disable bluedroid advice - now all compiles fine. 

Yes I've read port's page and have attempted to compile examples, there were no info about disbling bluedroid. Is there somewhere recomended settings?

But i've still not able achieve working result.

1. I see error in monitor:
E (6572) phy_init: esp_phy_load_cal_data_from_nvs: NVS has not been initialized. Call nvs_flash_init before starting WiFi/BT. Is it critical?

2. I've tried compile "gap_inquiry" - compiles, flashes, but:
 (!) device not visible and console monitor shows nothing.

3. Before I've post issue gap_inquiry works for me, but device wasnt visible

Update:
4. After some experiments I have now device visible, scanning working, but also enabled bluedroid support:
CONFIG_BLUEDROID_ENABLED=y

Update 2:
5. on issue "esp_phy_load_cal_data_from_nvs"
I've added:
+#include "nvs_flash.h"
..
int btstack_main(int argc, const char * argv[]) {
..
+nvs_flash_init();

no Error now..

WallGeorg, 2017-08-04:
I found the error as follow:

CC a2dp_source_demo.o
/home/str/esp/btstack/port/esp32/a2dp_source_demo/main/./a2dp_source_demo.c:46:20: fatal error: hxcmod.h: No such file or directory
compilation terminated.
/home/str/workspace/esp-idf/make/component_wrapper.mk:210: recipe for target 'a2dp_source_demo.o' failed
make[1]: *** [a2dp_source_demo.o] Error 1
/home/str/workspace/esp-idf/make/project.mk:390: recipe for target 'main-build' failed
make: *** [main-build] Error 2

Could you help me?
Best regards.

mringwal, 2017-08-06:
@lefedor thanks for reporting on the nvs_flash warning/error. I recently added support for storing link keys in the nvs. nvs_flash_init is called by btstack_tlv_esp32_get_instance() in btstack_tlv_esp32.c. So this was fixed by "accident" on the develop branch. I just merged it onto master.

it compiles & runs for me as expected with CONFIG_BLUEDROID_ENABLED commented out like this:
# CONFIG_BLUEDROID_ENABLED is not set

Please let me know if you have other problems.

closed 2017-08-06 by mringwal

WallGeorg, 2017-08-06:
Hi.
i download btstack-master.
But I can't find btstack_tlv_esp32.c.
Best regards.

mringwal, 2017-08-06:
https://github.com/bluekitchen/btstack/blob/master/port/esp32/template/components/btstack/btstack_tlv_esp32.c

WallGeorg, 2017-08-06:
Hi.
I found error.
[image]

mringwal, 2017-08-06:
@WallGeorg please pull esp-idf. esp_flash_erase() is defined here https://github.com/espressif/esp-idf/blob/master/components/nvs_flash/include/nvs_flash.h#L44