bluekitchen/btstack #52: I can't use your project.

WallGeorg, 2017-08-03:
Hi.
I can't do hsp_ag_demo and hfp_ag_demo.
How to define HAVE_BTSTACK_STDIN?
/*
 * hfp_ag_demo.c
 */

// *****************************************************************************
/* EXAMPLE_START(hfp_ag_demo): HFP Audio Gateway (AG) Demo
 *
 * @text This HFP Audio Gateway example demonstrates how to receive 
 * an output from a remote HFP Hands-Free (HF) unit, and, 
 * if HAVE_BTSTACK_STDIN is defined, how to control the HFP HF. 
 */
// *****************************************************************************

#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#include "btstack.h"
#include "sco_demo_util.h"
#ifdef HAVE_BTSTACK_STDIN
#include "btstack_stdin.h"
#endif
Could you help me.
Best regards.

WallGeorg, 2017-08-03:
I want to discuss about ESP32

mringwal, 2017-08-06:
All HAVE_ defines are port specific. The ESP32 port doesn't have support for STDIN currently. You can run the hfp_ag_demo on the ESP32, but you cannot do much without the STDIN support. The demo would play a sine tone as there's no support for audio either currently.

closed 2017-08-06 by mringwal