bluekitchen/btstack #54: I don't understand.

WallGeorg, 2017-08-04:
Hi. 
I don't understand in readme.
"In port/esp32, configure the serial port for firmware upload as described in the ESP-IDF setup guides."
Could you explain install method?
Best regards.

asuivelentine, 2017-08-04:
Hi,
from the ESP-IDF docs:
```
An “app” is an executable which is built by esp-idf. A single project will   
usually build two apps - a “project app” (the main executable, ie your custom firmware)  
and a “bootloader app” (the initial bootloader program which launches the project app).
```
So the meaning of firmware is defined. 

To configure the serial port follow this guide (It's not the port it's within your app):
[ESP-IDF doc - serial config](http://esp-idf.readthedocs.io/en/latest/get-started/index.html#configure)

But remember to set the path to ESP-IDF-working dir as described here (if you are on a linux machine):
[ESP-IDF doc - framework setup](http://esp-idf.readthedocs.io/en/latest/get-started/index.html#setup-path-to-esp-idf)

WallGeorg, 2017-08-04:
Hi. asuivelentine
Thanks for your answer
I have done default config.
That is, I click save and exit.
ok?

closed 2017-08-06 by mringwal