bluekitchen/btstack #55: a2dp_source_demo compile error!!!

WallGeorg, 2017-08-04:
Hi.
I can't compile a2dp_source_demo.
error is as follow :
--------------------------------------------------------------
CC a2dp_source_demo.o
/home/str/esp/btstack/port/esp32/a2dp_source_demo/main/./a2dp_source_demo.c:46:20: fatal error: hxcmod.h: No such file or directory
compilation terminated.
/home/str/workspace/esp-idf/make/component_wrapper.mk:210: recipe for target 'a2dp_source_demo.o' failed
make[1]: *** [a2dp_source_demo.o] Error 1
/home/str/workspace/esp-idf/make/project.mk:390: recipe for target 'main-build' failed
make: *** [main-build] Error 2
--------------------------------------------------------------------------
pls, Help me.
Best regards.

asuivelentine, 2017-08-05:
Hi WallGeorg,
the build configuration seems to be broken.
Finally I found a way to build this example. However I'm a new user and not sure if this is the correct / niciest solution.

What I've done:
- Navigate to `port/esp32/a2dp_source_demo/components/btstack`
- open `component.mk`
- change `COMPONENT_ADD_INCLUDEDIRS` and `COMPONENT_SRCDIRS` to:

```
COMPONENT_ADD_INCLUDEDIRS := \
	${BTSTACK_ROOT}/3rd-party/bluedroid/decoder/include \
	${BTSTACK_ROOT}/3rd-party/bluedroid/encoder/include \
	${BTSTACK_ROOT}/3rd-party/hxcmod-player \
	${BTSTACK_ROOT}/3rd-party/hxcmod-player/mods \
	$(BTSTACK_ROOT)/src/ble/gatt-service \
	$(BTSTACK_ROOT)/src/ble \
	$(BTSTACK_ROOT)/src/classic \
	$(BTSTACK_ROOT)/src \
	$(BTSTACK_ROOT)/platform/embedded \
	$(BTSTACK_ROOT)/platform/freertos \
	include \

COMPONENT_SRCDIRS := \
	${BTSTACK_ROOT}/3rd-party/bluedroid/decoder/srce \
	${BTSTACK_ROOT}/3rd-party/bluedroid/encoder/srce \
	${BTSTACK_ROOT}/3rd-party/hxcmod-player \
	${BTSTACK_ROOT}/3rd-party/hxcmod-player/mods \
	$(BTSTACK_ROOT)/src \
	$(BTSTACK_ROOT)/src/ble/gatt-service \
	$(BTSTACK_ROOT)/src/ble \
	$(BTSTACK_ROOT)/src/classic \
	$(BTSTACK_ROOT)/platform/freertos \
	. \
```

best regards, asui

WallGeorg, 2017-08-06:
Hello. Asuivelentine .
Thanks for your answer.
I have done compile a2dp_source_demo as your offer.
[image]
[image]
I want to connect esp32 to Phone(iPhone or Android).
But I can't it,

Could you help me?
Best regards

mringwal, 2017-08-06:
@asuivelentine Thanks for updating the source and include paths. It's 100% correct. I've updated the component.mk for the master and develop branches just now.

mringwal, 2017-08-06:
@WallGeorg Now that you have the a2dp_source_demo running, you'd like to connect to it. Unfortunately, none of the common systems (iOS, Android, Windows, macOS) support the Audio Sink role to test it. You would need to use a headset/loudspeaker to test it. However, these usually don't have a UI to trigger a connect. We'll look into at least update the demo to auto-connect to a fixed address.

closed 2017-08-06 by mringwal

ahsankaukab3, 2021-02-15:
i am running the make command and the following error occurred

C:\2\SDM-32-F411\btstack\port\stm32-f4discovery-cc256x>make

arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DUSE_HAL_DRIVER -DSTM32F407xx -IInc -IDrivers/STM32F4xx_HAL_Driver/Inc -IDrivers/STM32F4xx_HAL_Driver/Inc/Legacy -IDrivers/CMSIS/Device/ST/STM32F4xx/Include -IDrivers/CMSIS/Include  -Ibuild -I../../src/ble -I../../src/ble/gatt-service -I../../src/classic -I../../src -I../../3rd-party/micro-ecc -I../../3rd-party/bluedroid/decoder/include -I../../3rd-party/bluedroid/encoder/include -I../../3rd-party/hxcmod-player -I../../3rd-party/hxcmod-player/mods -I../../3rd-party/lwip/core/src/include -I../../3rd-party/lwip/dhcp-server -I../../3rd-party/md5 -I../../3rd-party/yxml -I../../3rd-party/segger-rtt -I../../platform/embedded -I../../platform/lwip -I../../platform/lwip/port -I../../chipset/cc256x -I Drivers/STM32F4xx_HAL_Driver/Inc -I bsp -I pdm -I port -I src -Os -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/hxcmod.d" -Wa,-a,-ad,-alms=build/hxcmod.lst ../../3rd-party/hxcmod-player/hxcmod.c -o build/hxcmod.o
make: *** [build/hxcmod.o] Error -1073741502

mringwal, 2021-02-15:
hi. looks like a problem with your setup. You seem to run Windows. In Windows, please try to compile in a mingw shell, see port/windows-h4/README.md for instructions or try in a Linux VM.

ahsankaukab3, 2021-02-15:
Thanks, @mringwal for the help, I have compiled the make file in mingw shell but it gives me another error as attached
 Error: 127

Ahsan@DESKTOP-LDLV1LG /c/2/SDM-32-F411/btstack/port/stm32-f4discovery-cc256x

arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DUSE_HAL_DRIVER -DSTM32F407xx -IInc -IDrivers/STM32F4xx_HAL_Driver/Inc -IDrivers/STM32F4xx_HAL_Driver/Inc/Legacy -IDrivers/CMSIS/Device/ST/STM32F4xx/Include -IDrivers/CMSIS/Include  -Ibuild -I../../src/ble -I../../src/ble/gatt-service -I../../src/classic -I../../src -I../../3rd-party/micro-ecc -I../../3rd-party/bluedroid/decoder/include -I../../3rd-party/bluedroid/encoder/include -I../../3rd-party/hxcmod-player -I../../3rd-party/hxcmod-player/mods -I../../3rd-party/lwip/core/src/include -I../../3rd-party/lwip/dhcp-server -I../../3rd-party/md5 -I../../3rd-party/yxml -I../../3rd-party/segger-rtt -I../../platform/embedded -I../../platform/lwip -I../../platform/lwip/port -I../../chipset/cc256x -I Drivers/STM32F4xx_HAL_Driver/Inc -I bsp -I pdm -I port -I src -Os -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/hxcmod.d" -Wa,-a,-ad,-alms=build/hxcmod.lst ../../3rd-party/hxcmod-player/hxcmod.c -o build/hxcmod.o
arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DUSE_HAL_DRIVER -DSTM32F407xx -IInc -IDrivers/STM32F4xx_HAL_Driver/Inc -IDrivers/STM32F4xx_HAL_Driver/Inc/Legacy -IDrivers/CMSIS/Device/ST/STM32F4xx/Include -IDrivers/CMSIS/Include  -Ibuild -I../../src/ble -I../../src/ble/gatt-service -I../../src/classic -I../../src -I../../3rd-party/micro-ecc -I../../3rd-party/bluedroid/decoder/include -I../../3rd-party/bluedroid/encoder/include -I../../3rd-party/hxcmod-player -I../../3rd-party/hxcmod-player/mods -I../../3rd-party/lwip/core/src/include -I../../3rd-party/lwip/dhcp-server -I../../3rd-party/md5 -I../../3rd-party/yxml -I../../3rd-party/segger-rtt -I../../platform/embedded -I../../platform/lwip -I../../platform/lwip/port -I../../chipset/cc256x -I Drivers/STM32F4xx_HAL_Driver/Inc -I bsp -I pdm -I port -I src -Os -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/nao-deceased_by_disease.d" -Wa,-a,-ad,-alms=build/nao-deceased_by_disease.lst ../../3rd-party/hxcmod-player/mods/nao-deceased_by_disease.c -o build/nao-deceased_by_disease.o
arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DUSE_HAL_DRIVER -DSTM32F407xx -IInc -IDrivers/STM32F4xx_HAL_Driver/Inc -IDrivers/STM32F4xx_HAL_Driver/Inc/Legacy -IDrivers/CMSIS/Device/ST/STM32F4xx/Include -IDrivers/CMSIS/Include  -Ibuild -I../../src/ble -I../../src/ble/gatt-service -I../../src/classic -I../../src -I../../3rd-party/micro-ecc -I../../3rd-party/bluedroid/decoder/include -I../../3rd-party/bluedroid/encoder/include -I../../3rd-party/hxcmod-player -I../../3rd-party/hxcmod-player/mods -I../../3rd-party/lwip/core/src/include -I../../3rd-party/lwip/dhcp-server -I../../3rd-party/md5 -I../../3rd-party/yxml -I../../3rd-party/segger-rtt -I../../platform/embedded -I../../platform/lwip -I../../platform/lwip/port -I../../chipset/cc256x -I Drivers/STM32F4xx_HAL_Driver/Inc -I bsp -I pdm -I port -I src -Os -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/uECC.d" -Wa,-a,-ad,-alms=build/uECC.lst ../../3rd-party/micro-ecc/uECC.c -o build/uECC.o
arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DUSE_HAL_DRIVER -DSTM32F407xx -IInc -IDrivers/STM32F4xx_HAL_Driver/Inc -IDrivers/STM32F4xx_HAL_Driver/Inc/Legacy -IDrivers/CMSIS/Device/ST/STM32F4xx/Include -IDrivers/CMSIS/Include  -Ibuild -I../../src/ble -I../../src/ble/gatt-service -I../../src/classic -I../../src -I../../3rd-party/micro-ecc -I../../3rd-party/bluedroid/decoder/include -I../../3rd-party/bluedroid/encoder/include -I../../3rd-party/hxcmod-player -I../../3rd-party/hxcmod-player/mods -I../../3rd-party/lwip/core/src/include -I../../3rd-party/lwip/dhcp-server -I../../3rd-party/md5 -I../../3rd-party/yxml -I../../3rd-party/segger-rtt -I../../platform/embedded -I../../platform/lwip -I../../platform/lwip/port -I../../chipset/cc256x -I Drivers/STM32F4xx_HAL_Driver/Inc -I bsp -I pdm -I port -I src -Os -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/SEGGER_RTT.d" -Wa,-a,-ad,-alms=build/SEGGER_RTT.lst ../../3rd-party/segger-rtt/SEGGER_RTT.c -o build/SEGGER_RTT.o
arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DUSE_HAL_DRIVER -DSTM32F407xx -IInc -IDrivers/STM32F4xx_HAL_Driver/Inc -IDrivers/STM32F4xx_HAL_Driver/Inc/Legacy -IDrivers/CMSIS/Device/ST/STM32F4xx/Include -IDrivers/CMSIS/Include  -Ibuild -I../../src/ble -I../../src/ble/gatt-service -I../../src/classic -I../../src -I../../3rd-party/micro-ecc -I../../3rd-party/bluedroid/decoder/include -I../../3rd-party/bluedroid/encoder/include -I../../3rd-party/hxcmod-player -I../../3rd-party/hxcmod-player/mods -I../../3rd-party/lwip/core/src/include -I../../3rd-party/lwip/dhcp-server -I../../3rd-party/md5 -I../../3rd-party/yxml -I../../3rd-party/segger-rtt -I../../platform/embedded -I../../platform/lwip -I../../platform/lwip/port -I../../chipset/cc256x -I Drivers/STM32F4xx_HAL_Driver/Inc -I bsp -I pdm -I port -I src -Os -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/SEGGER_RTT_Syscalls_GCC.d" -Wa,-a,-ad,-alms=build/SEGGER_RTT_Syscalls_GCC.lst ../../3rd-party/segger-rtt/SEGGER_RTT_Syscalls_GCC.c -o build/SEGGER_RTT_Syscalls_GCC.o
[... cut ...] rs/STM32F4xx_HAL_Driver/Inc -I bsp -I pdm -I port -I src -Os -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/bnep_lwip.d" -Wa,-a,-ad,-alms=build/bnep_lwip.lst ../../platform/lwip/bnep_lwip.c -o build/bnep_lwip.o
echo "Downloading cc256xb_bt_sp_v1.8.zip"
Downloading cc256xb_bt_sp_v1.8.zip
Extracting cc256xb_bt_sp_v1.8/initscripts-TIInit_6.7.16_bt_spec_4.1.bts
make: unzip: Command not found
make: *** [cc256xb_bt_sp_v1.8/initscripts-TIInit_6.7.16_bt_spec_4.1.bts] Error 127

mringwal, 2021-02-15:
please read the last two lines of the output

ahsankaukab3, 2021-02-15:
I have unzip the file manually but still 
make: *** [cc256xb_bt_sp_v1.8/initscripts-TIInit_6.7.16_bt_spec_4.1.bts] Error 127 
remained on its point.

Then I observed that the path the make command is trying to read is [cc256xb_bt_sp_v1.8/initscripts-TIInit_6.7.16_bt_spec_4.1.bts] but actual path was [cc256xb_bt_sp_v1.8/CC256XB_BT_SP/v1.8//initscripts-TIInit_6.7.16_bt_spec_4.1.bts] so it changed it to required one but still the issue remains the same.

ahsankaukab3, 2021-02-16:
I have tried to run make file in ubuntu (Windows Partition) but the same issue encountered:

Extracting cc256xb_bt_sp_v1.8/initscripts-TIInit_6.7.16_bt_spec_4.1.bts
Extracting cc256xb_bt_sp_v1.8/initscripts-TIInit_6.7.16_ble_add-on.bts
make: unzip: Command not found
make: unzip: Command not found
make: *** [../../chipset/cc256x/Makefile.inc:109: cc256xb_bt_sp_v1.8/initscripts-TIInit_6.7.16_bt_spec_4.1.bts] Error 127
make: *** [../../chipset/cc256x/Makefile.inc:114: cc256xb_bt_sp_v1.8/initscripts-TIInit_6.7.16_ble_add-on.bts] Error 127

@mringwal please help me solve this issue.

mringwal, 2021-02-16:
You don’t have unzip installed according to your log.

ahsankaukab3, 2021-02-16:
i have installed the unzip. Now I got this error:

Extracting cc256xb_bt_sp_v1.8/initscripts-TIInit_6.7.16_bt_spec_4.1.bts
Extracting cc256xb_bt_sp_v1.8/initscripts-TIInit_6.7.16_ble_add-on.bts
Creating bluetooth_init_cc2564B_1.8_BT_Spec_4.1.c
/usr/bin/env: ‘python3\r’: No such file or directory
make: *** [../../chipset/cc256x/Makefile.inc:274: bluetooth_init_cc2564B_1.8_BT_Spec_4.1.c] Error 127

mringwal, 2021-02-16:
If you don't have python3 installed, please install it. If python3 is installed, then it looks like the line endings in the Makefile is wrong. 

Please clone BTstack using git from the ubuntu and try again

ahsankaukab3, 2021-03-09:
I am running gatt_streamer_server example with bluetooth stack (STM32-F411) and i am getting the following error:
------------------------------------------------------------------------------------------------------------------------------------------
To start the streaming, please run the le_streamer_client example on other device, or use some GATT Explorer, e.g. LightBlue, BLExplr.
[00:00:00.000] LOG -- btstack_crypto.c.952: more ecc random
[00:00:00.000] LOG -- btstack_crypto.c.952: more ecc random
[00:00:00.000] LOG -- btstack_crypto.c.952: more ecc random
[00:00:00.000] LOG -- btstack_crypto.c.952: more ecc random
[00:00:00.000] LOG -- btstack_crypto.c.952: more ecc random
[00:00:00.000] LOG -- btstack_crypto.c.952: more ecc random
[00:00:00.000] LOG -- btstack_crypto.c.952: more ecc random
[00:00:00.000] LOG -- btstack_crypto.c.521: set uECC RNG for initial key generation with 64 random bytes
[00:00:00.000] LOG -- btstack_crypto.c.497: sm_generate_f_rng: size 32 - offset 0
[00:00:00.000] LOG -- btstack_crypto.c.487: Elliptic curve: X
[00:00:00.000] LOG -- btstack_util.c.250: 0x81, 0x3D, 0xA8, 0x03, 0xE8, 0x80, 0xDF, 0xA0, 0x76, 0xA0, 0xBA, 0xAA, 0x8D, 0x71, 0x5B, 0xBB,
[00:00:00.000] LOG -- btstack_util.c.250: 0x3E, 0xAA, 0x04, 0xE0, 0x4E, 0xF3, 0xAE, 0xB0, 0x84, 0x11, 0x4C, 0x35, 0x11, 0xA6, 0x5E, 0xDC,
[00:00:00.000] LOG -- btstack_crypto.c.489: Elliptic curve: Y
[00:00:00.000] LOG -- btstack_util.c.250: 0x7C, 0xDE, 0x28, 0xF4, 0x6D, 0x2C, 0x74, 0xB2, 0x87, 0xDC, 0x97, 0x62, 0x60, 0xDF, 0x7F, 0xD1,
[00:00:00.000] LOG -- btstack_util.c.250: 0x2C, 0x35, 0xD8, 0x01, 0xEB, 0x90, 0xFA, 0xF8, 0x3C, 0x3F, 0x31, 0x05, 0x34, 0x46, 0x48, 0x74,
[00:00:00.000] LOG -- btstack_util.c.290: irk    89F6C0518602133791928F43FC0B7B80
[00:00:00.000] LOG -- sm.c.2045: DKG_CALC_DHK started
[00:00:00.000] LOG -- btstack_util.c.290: dhk    3C8668FC9117D65F049A1449F8329A57
[00:00:00.000] LOG -- hci.c.2192: LE Connection_complete (status=0) type 1, 52:C5:88:80:EE:96
[00:00:00.000] LOG -- hci.c.194: create_connection_for_addr 52:C5:88:80:EE:96, type 1
[00:00:00.000] LOG -- sm.c.3468: sm: connected
- LE Connection 0401: connected - connection interval 48.75 ms, latency 0
- LE Connection 0401: request 15 ms connection interval
[00:00:00.000] LOG -- hci.c.856: acl_handler called with non-registered handle 1025!
- LE Connection 0401: connection update - connection interval 7.50 ms, latency 0
[00:00:00.000] LOG -- hci.c.856: acl_handler called with non-registered handle 1025!
[00:00:00.000] LOG -- hci.c.856: acl_handler called with non-registered handle 1025!
- LE Connection 0401: disconnect, reason 13
-------------------------------------------------------------------------------------------------------------------------------------------
My port_main is as follow:
void port_main(void){

	 btstack_memory_init();
	    btstack_run_loop_init(btstack_run_loop_embedded_get_instance());

		// hci_dump_open( NULL, HCI_DUMP_STDOUT );

		// init HCI
	    hci_init(hci_transport_h4_instance(btstack_uart_block_embedded_instance()), (void*) &config);
	    hci_set_chipset(btstack_chipset_cc256x_instance());

	    // setup TLV Flash Sector implementation
	    const hal_flash_bank_t * hal_flash_bank_impl = hal_flash_bank_stm32_init_instance(
	    		&hal_flash_bank_context,
	    		HAL_FLASH_BANK_SIZE,
				HAL_FLASH_BANK_0_SECTOR,
				HAL_FLASH_BANK_1_SECTOR,
				HAL_FLASH_BANK_0_ADDR,
				HAL_FLASH_BANK_1_ADDR);
	    const btstack_tlv_t * btstack_tlv_impl = btstack_tlv_flash_bank_init_instance(
	    		&btstack_tlv_flash_bank_context,
				hal_flash_bank_impl,
				&hal_flash_bank_context);
//
//	    // setup global tlv
	    btstack_tlv_set_instance(btstack_tlv_impl, &btstack_tlv_flash_bank_context);

	    // setup Link Key DB using TLV
	    const btstack_link_key_db_t * btstack_link_key_db = btstack_link_key_db_tlv_get_instance(btstack_tlv_impl, &btstack_tlv_flash_bank_context);
	    hci_set_link_key_db(btstack_link_key_db);

	    // setup LE Device DB using TLV
	    le_device_db_tlv_configure(btstack_tlv_impl, &btstack_tlv_flash_bank_context);

#ifdef HAVE_HAL_AUDIO
    // setup audio
   	btstack_audio_sink_set_instance(btstack_audio_embedded_sink_get_instance());
    btstack_audio_source_set_instance(btstack_audio_embedded_source_get_instance());
#endif

    hci_set_chipset(btstack_chipset_cc256x_instance());
		// hand over to btstack embedded code
	    btstack_main(0, NULL);

	    // go
	    btstack_run_loop_execute();

}

mringwal, 2021-03-09:
Please post on the mailing list (see project page).