bluekitchen/btstack #56: I want to connect esp32 to Phone.

WallGeorg, 2017-08-06:
Hi.
I am working a2dp_source_demo with esp-32s DevkitC.
I can't connect esp32 to Phone.
[image]
I don't know the connecting method.
Please Help me!
Best regards.

mringwal, 2017-08-06:
See issue #55 - the demo should automatically connect to the hard-coded device_addr_string if HAVE_BTSTACK_STDIN is not implemented

WallGeorg, 2017-08-07:
Hi. sorry.
I can't connect esp32 to bluetooth earphone.
I give device_addr_string my earphone's address.
But I can't connect.
Could you explain about it?
Best regards.

mringwal, 2017-08-07:
please enable packet log by uncommenting 
  hci_dump_open(NULL, HCI_DUMP_STDOUT);
in the main() of main.c and post the log file somewhere. and also provide the address of your headphone.

WallGeorg, 2017-08-07:
Hi.
I upload my log file.
[log.zip](https://github.com/bluekitchen/btstack/files/1205112/log.zip)
My earphone address:       1C:52:16:14:99:97
Please Help me.

mringwal, 2017-08-07:
Well. I was referring to the output on the debug console that you get when you run "make monitor"

WallGeorg, 2017-08-07:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
[0;32mI (398) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (398) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (404) heap_init: At 3FFCCCF0 len 00013310 (76 KiB): DRAM[0m
[0;32mI (423) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (443) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (462) heap_init: At 40094460 len 0000BBA0 (46 KiB): IRAM[0m
BTstack: setup
[00:00:00.523] LOG -- btstack_run_loop_freertos.c.258: run loop init, queue item size 8
[00:00:00.523] LOG -- main.c.250: transport_register_packet_handler
[00:00:00.530] LOG -- l2cap.c.3048: L2CAP_REGISTER_SERVICE psm 0x19 mtu 65535
[00:00:00.535] LOG -- l2cap.c.3048: L2CAP_REGISTER_SERVICE psm 0x19 mtu 65535
[00:00:00.542] LOG -- l2cap.c.3053: l2cap_register_service: PSM 25 already registered
[00:00:00.549] LOG -- l2cap.c.3048: L2CAP_REGISTER_SERVICE psm 0x17 mtu 65535
[00:00:00.556] LOG -- l2cap.c.3048: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535
loaded mod 'nao-deceased by disease', size 7600
[00:00:00.567] LOG -- hci.c.2588: hci_power_control: 1, current mode 0
[00:00:00.573] LOG -- main.c.200: transport_init
[00:00:00.578] LOG -- main.c.216: transport_open
[00:00:00.755] LOG -- hci.c.3635: BTSTACK_EVENT_STATE 1
[00:00:00.755] EVT <= 60 01 01 
[00:00:00.755] CMD => 03 0C 00 
BTstack: execute run loop
[00:00:00.757] EVT <= 6E 00 
[00:00:00.760] EVT <= 0E 04 05 03 0C 00 
[00:00:00.763] CMD => 01 10 00 
[00:00:00.766] EVT <= 0E 0C 05 01 10 00 08 0E 03 08 60 00 0E 03 
[00:00:00.772] LOG -- hci.c.1758: Manufacturer: 0x0060
[00:00:00.777] EVT <= 6E 00 
[00:00:00.779] CMD => 14 0C 00 
[00:00:00.782] EVT <= 0E FC 05 14 0C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[00:00:00.850] LOG -- hci.c.1672: local name: 
[00:00:00.855] LOG -- hci.c.1464: Received local name, need baud change 0
[00:00:00.861] EVT <= 6E 00 
[00:00:00.864] CMD => 02 10 00 
[00:00:00.867] EVT <= 0E 44 05 02 10 00 BF FF FF 03 CC FF EF FF 3F FF FC 1F F2 0F E8 FE 3F F7 8F FF 1C 00 04 00 61 F7 FF FF 7F 38 C0 FF 7F FF FF 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[00:00:00.887] LOG -- hci.c.1768: Local supported commands summary 0x3f
[00:00:00.893] EVT <= 6E 00 
[00:00:00.896] CMD => 09 10 00 
[00:00:00.899] EVT <= 0E 0A 05 09 10 00 FE 3F 80 C4 0A 24 
[00:00:00.904] LOG -- hci.c.1720: Local Address, Status: 0x00: Addr: 24:0A:C4:80:3F:FE
[00:00:00.912] EVT <= 6E 00 
[00:00:00.914] CMD => 05 10 00 
[00:00:00.917] EVT <= 0E 0B 05 05 10 00 FD 03 FF 09 00 04 00 
[00:00:00.922] LOG -- hci.c.1689: hci_read_buffer_size: ACL size module 1021 -> used 1021, count 9 / SCO size 255, count 4
[00:00:00.933] EVT <= 6E 00 
[00:00:00.936] CMD => 03 10 00 
[00:00:00.939] EVT <= 0E 0C 05 03 10 00 BF EE CD FE DB FF 7B 87 
[00:00:00.944] LOG -- hci.c.1747: Packet types cc18, eSCO 1
[00:00:00.950] LOG -- hci.c.1750: BR/EDR support 1, LE support 1
[00:00:00.955] EVT <= 6E 00 
[00:00:00.958] CMD => 01 0C 08 FF FF FF FF FF FF FF 3F 
[00:00:00.963] EVT <= 0E 04 05 01 0C 00 
[00:00:00.967] EVT <= 6E 00 
[00:00:00.969] CMD => 56 0C 01 01 
[00:00:00.972] EVT <= 0E 04 05 56 0C 00 
[00:00:00.976] EVT <= 6E 00 
[... 10 lines cut ...]
[00:00:01.075] EVT <= 0E 04 05 45 0C 00 
[00:00:01.079] EVT <= 6E 00 
[00:00:01.081] CMD => 1A 0C 01 03 
[00:00:01.084] EVT <= 0E 04 05 1A 0C 00 
[00:00:01.088] LOG -- hci.c.3747: BTSTACK_EVENT_DISCOVERABLE_ENABLED 1
[00:00:01.094] EVT <= 66 01 01 
[00:00:01.097] EVT <= 6E 00 
[00:00:01.100] CMD => 02 20 00 
[00:00:01.103] EVT <= 0E 07 05 02 20 00 FB 00 0A 
[00:00:01.107] LOG -- hci.c.1700: hci_le_read_buffer_size: size 251, count 10
[00:00:01.114] EVT <= 6E 00 
[00:00:01.116] CMD => 6D 0C 02 01 00 
[00:00:01.120] EVT <= 0E 04 05 6D 0C 00 
[00:00:01.123] EVT <= 6E 00 
[00:00:01.126] CMD => 0F 20 00 
[00:00:01.129] EVT <= 0E 05 05 0F 20 00 0C 
[00:00:01.133] LOG -- hci.c.1713: hci_le_read_white_list_size: size 12
[00:00:01.139] EVT <= 6E 00 
[00:00:01.142] CMD => 0B 20 07 01 E0 01 30 00 00 00 
[00:00:01.147] EVT <= 0E 04 05 0B 20 00 
[00:00:01.150] LOG -- hci.c.1331: hci_init_done -> HCI_STATE_WORKING
[00:00:01.156] LOG -- hci.c.3635: BTSTACK_EVENT_STATE 2
[00:00:01.161] EVT <= 60 01 02 
BTstack: up and running.
[00:00:01.166] EVT <= 6E 00

mringwal, 2017-08-07:
At this point, please press the space key to get the menu. You can press 'b' to connect to your headset and when the connection is active, press 'z' to play the mod.

WallGeorg, 2017-08-07:
I am sorry.
My file is arduino output.
I can't make flash and make monitor in ubuntu.
I don't know method.
I write bin files with ESP 32 download tools 3.4.9
Please help me.
Best regards.

mringwal, 2017-08-07:
Did you compile the demo with the Arduino IDE? (if yes, how?)

If you're on ubuntu, make flash and make monitor should just work.

Anyway, you can also wait a bit more until we implement the 'auto-connect on startup'

WallGeorg, 2017-08-07:
I compile demo on ubuntu.
But I can't make flash.
[image]
Please help me.
Regards.

mringwal, 2017-08-07:
make flash tries to flash the ESP32 connected at path /dev/tty.usbserial.DN02B3PF. That's the device name for *my* ESP32. You need to find the correct path to it and update it in sdkconfig either with a text editor or by running 'make menuconfig' and finding the serial port configuration.

WallGeorg, 2017-08-07:
Hi.
I am very sorry.
I don't know.
[image]
[image]
[image]
Please help me.
Regards.

mringwal, 2017-08-07:
you may need to run 'make flash' with sudo
$ sudo make flash
and enter your ubuntu user password

WallGeorg, 2017-08-08:
Hi.
I can't do "$ sudo make flash".
[image]
Please Help me.
Best regards.

mringwal, 2017-08-08:
the sudo make flash fails because the root user has a different environment, where the path to esp-idf isn't set.

auto-connect will be pushed when someone implements it.

here's another idea: install "picocom" terminal and then run
picocom --baud 115200 /dev/ttyUSB0

if you get an error because of permissions, you may try it with sudo

sudo picocom --baud 115200 /dev/ttyUSB0

WallGeorg, 2017-08-08:
Hi.
I want auto-connect and start.
Could you tell me about auto-connect and start method?
Best regards.

milamikica, 2017-08-23:
Hi,
it is fixed now for in develop branch. 

The following demos hfp_ag_demo and a2dp_source_demo autoconnect now
if HAVE_BTSTACK_STDIN is not defined in btstack_config.h

closed 2017-08-23 by milamikica

WallGeorg, 2017-08-23:
Thank's for your contact.
I want to discuss about bluetooth.
My skype ID is electric.mechanical@yandex.com.
Best regards.

mringwal, 2017-08-23:
We're very busy. What would you like to discuss specifically?

WallGeorg, 2017-08-23:
I want to develop Bluetooth remote controller with BCM20733.
But I can't develop.
I called cypress sales, but they didn't response.
I hope your help.
Best regards.

mringwal, 2017-08-23:
I'm sorry but we cannot help you with the BCM20733. We don't have a direct contact to Cypress either.  So far, we have good experience with Texas Instruments, Nordic, and EM Micro. See https://github.com/bluekitchen/btstack/tree/master/chipset