bluekitchen/btstack #59: ESP32: debugging methods for btstack problems

my-abousamra, 2017-08-07:
Hi,

I've an application where ESP32 module receives data over uart port and sends it to btstack over FreeRTOS queue as shown in this simple example

`
// a task receives data from FreeRTOS queue and request to send 
static void task1 (void *pvParameters)
{
	while (1)
	{
		xEventGroupWaitBits ( bt_classic_event_group, BT_CLASSIC_READY_BIT, pdTRUE, pdFALSE, portMAX_DELAY );

		if ( xQueueReceive( xBt_Queue, &bt_q_rd_msg, portMAX_DELAY ) == pdPASS )
		{
			rfcomm_request_can_send_now_event ( rfcomm_channel_id );
		}
	}
}
`

`
static void packet_handler (uint8_t packet_type, uint16_t channel, uint8_t *packet, uint16_t size)
{
	...

	switch (packet_type)
	{
		case HCI_EVENT_PACKET:
			switch (hci_event_packet_get_type(packet))
			{

				case RFCOMM_EVENT_CAN_SEND_NOW:
					if ( rfcomm_send ( rfcomm_channel_id, (uint8_t*) bt_q_rd_msg.u8Bdy, bt_q_rd_msg.u8Hdr[0] ) == 0 )
					{
						ESP_LOGI("bt_classic_packet_handler", "send success ID%d L%d", rfcomm_channel_id, bt_q_rd_msg.u8Hdr[0]);
					}
					else
					{
						ESP_LOGE("bt_classic_packet_handler", "send err");
					}

					xEventGroupSetBits ( bt_classic_event_group, BT_CLASSIC_READY_BIT );

				break;

				...
			}
			...
		}	
		...
	}
}
`

It works fine but for a few seconds until it stop working where btstack stop sending events for rfcomm_request_can_send_now_event call, It seems that btstack stop working but it continues if the bluetooth link dropped and established again and problem occurs again.

My question is about debugging methods for btstack to check why it stops sending events as I can't do anything for such case

my-abousamra, 2017-08-07:
I enabled btstack log and here's the output, It stopped until the queue gets full

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3416
load:0x40078000,len:10384
load:0x40080000,len:252
entry 0x40080034
I (1314) UART_vInit:
[00:00:01.316] LOG -- C:/Users/pulse/workspace/spp_counter_final/components/Bt_Stack/../../../../../../esp/btstack/platform/freertos/btstack_run_loop_freertos.c.256: run loop init, queue item size 8
[00:00:01.324] LOG -- main.c.247: transport_register_packet_handler
[00:00:01.330] LOG -- rfcomm.c.2280: RFCOMM_REGISTER_SERVICE channel #1 mtu 65535 flow_control 0 credits 10
[00:00:01.339] LOG -- l2cap.c.2014: L2CAP_REGISTER_SERVICE psm 0x3 mtu 65535
[00:00:01.346] LOG -- l2cap.c.2014: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535
I (1353) Bt_Classic_vInit: SDP service record size: 95
[00:00:01.359] LOG -- hci.c.2588: hci_power_control: 1, current mode 0
[00:00:01.365] LOG -- main.c.197: transport_init
[00:00:01.369] LOG -- main.c.213: transport_open
[00:00:01.564] LOG -- hci.c.3635: BTSTACK_EVENT_STATE 1
I (1565) bt_classic_packet_handler: phase1: 1
[00:00:01.565] LOG -- hci.c.1059: hci_initializing_run: substate 0, can send 1
I (1571) uart_recv_task: uart_recv_task start
[00:00:01.572] I (1572) uart_send_task: uart_send_task start
I (1572) bt_classic_send_task: bt_classic_send_task start
I (1572) bt_classic_monitor_task: bt_classic_monitor_task start
LOG -- C:/Users/pulse/workspace/spp_counter_final/components/Bt_Stack/../../../../../../esp/btstack/platform/freertos/btstack_run_loop_freertos.c.174: RL: execute
[00:00:01.609] LOG -- hci.c.1346: Command complete for expected opcode 0c03 at substate 1
[00:00:01.617] LOG -- hci.c.1059: hci_initializing_run: substate 2, can send 1
[00:00:01.624] LOG -- hci.c.1758: Manufacturer: 0x0060
[00:00:01.628] LOG -- hci.c.1346: Command complete for expected opcode 1001 at substate 3
[00:00:01.636] LOG -- C:/Users/pulse/workspace/spp_counter_final/components/Bt_Stack/../../../../../../esp/btstack/platform/freertos/btstack_run_loop_freertos.c.204: RL: portMAX_DELAY 4294967295
[00:00:01.653] LOG -- C:/Users/pulse/workspace/spp_counter_final/components/Bt_Stack/../../../../../../esp/btstack/platform/freertos/btstack_run_loop_freertos.c.220: RL: wait with timeout 4294967295
[00:00:01.671] LOG -- hci.c.1059: hci_initializing_run: substate 4, can send 1
[00:00:01.678] LOG -- hci.c.1672: local name:
[00:00:01.682] LOG -- hci.c.1346: Command complete for expected opcode 0c14 at substate 5
[00:00:01.690] LOG -- hci.c.1464: Received local name, need baud change 0
[00:00:01.696] LOG -- C:/Users/pulse/workspace/spp_counter_final/components/Bt_Stack/../../../../../../esp/btstack/platform/freertos/btstack_run_loop_freertos.c.204: RL: portMAX_DELAY 4294967295
[00:00:01.713] LOG -- C:/Users/pulse/workspace/spp_counter_final/components/Bt_Stack/../../../../../../esp/btstack/platform/freertos/btstack_run_loop_freertos.c.220: RL: wait with timeout 4294967295
[00:00:01.731] LOG -- hci.c.1059: hci_initializing_run: substate 8, can send 1
[00:00:01.738] LOG -- hci.c.1768: Local supported commands summary 0x3f
[00:00:01.744] LOG -- hci.c.1346: Command complete for expected opcode 1002 at substate 15
[00:00:01.752] LOG -- C:/Users/pulse/workspace/spp_counter_final/components/Bt_Stack/../../../../../../esp/btstack/platform/freertos/btstack_run_loop_freertos.c.204: RL: portMAX_DELAY 4294967295
[00:00:01.769] LOG -- C:/Users/pulse/workspace/spp_counter_final/components/Bt_Stack/../../../../../../esp/btstack/platform/freertos/btstack_run_loop_freertos.c.220: RL: wait with timeout 4294967295
[00:00:01.786] LOG -- hci.c.1059: hci_initializing_run: substate 22, can send 1
[00:00:01.793] LOG -- hci.c.1720: Local Address, Status: 0x00: Addr: 24:0A:C4:03:42:F2
[00:00:01.801] LOG -- hci.c.1346: Command complete for expected opcode 1009 at substate 23
[00:00:01.809] LOG -- C:/Users/pulse/workspace/spp_counter_final/components/Bt_Stack/../../../../../../esp/btstack/platform/freertos/btstack_run_loop_freertos.c.204: RL: portMAX_DELAY 4294967295
[00:00:01.826] LOG -- C:/Users/pulse/workspace/spp_counter_final/components/Bt_Stack/../../../../../../esp/btstack/platform/freertos/btstack_run_loop_freertos.c.220: RL: wait with timeout 4294967295
[00:00:01.843] LOG -- hci.c.1059: hci_initializing_run: substate 24, can send 1
[00:00:01.850] LOG -- hci.c.1689: hci_read_buffer_size: ACL size module 1021 -> used 1021, count 9 / SCO size 255, count 4
[00:00:01.861] LOG -- hci.c.1346: Command complete for expected opcode 1005 at substate 25
[00:00:01.869] LOG -- C:/Users/pulse/workspace/spp_counter_final/components/Bt_Stack/../../../../../../esp/btstack/platform/freertos/btstack_run_loop_freertos.c.204: RL: portMAX_DELAY 4294967295
[00:00:01.886] LOG -- C:/Users/pulse/workspace/spp_counter_final/components/Bt_Stack/../../../../../../esp/btstack/platform/freertos/btstack_run_loop_freertos.c.220: RL: wait with timeout 4294967295
[00:00:01.903] LOG -- hci.c.1059: hci_initializing_run: substate 26, can send 1
[00:00:01.911] LOG -- hci.c.1747: Packet types cc18, eSCO 1
[00:00:01.916] LOG -- hci.c.1750: BR/EDR support 1, LE support 1
[00:00:01.921] LOG -- hci.c.1346: Command complete for expected opcode 1003 at substate 27
[... cut ...]
[00:01:29.586] LOG -- C:/Users/pulse/workspace/spp_counter_final/components/Bt_Stack/../../../../../../esp/btstack/platform/freertos/btstack_run_loop_freertos.c.215: RL: first timer 0x400e4350
[00:01:29.603] LOG -- C:/Users/pulse/workspace/spp_counter_final/components/Bt_Stack/../../../../../../esp/btstack/platform/freertos/btstack_run_loop_freertos.c.220: RL: wait with timeout 4294967295
E (122713) uart_recv_task: Can't send data to bt queue [50]
E (122715) uart_recv_task: Can't send data to bt queue [50]
E (124711) uart_recv_task: Can't send data to bt queue [50]

mringwal, 2017-08-07:
hi. enabling the packet log is the best thing you can do. However, I would recommend not to ENABLE_LOG_DEBUG - that's way to verbose.

In your log, the last message is "can't send data to bt queue", so it looks like this queue is full.

I don't have all code, but you should try to send all packets that have been queued when receiving the signal. So after the "rfcomm_event_can_send_now", you should check if you have more data to send and then request another can send now.

closed 2017-08-07 by mringwal

my-abousamra, 2017-08-08:
Hi,
I tried what you suggested, when I got rfcomm can send now event I send all available packets in the queue which is 1 packet at the beginning and when I check again later for rfcomm can send now no event comes again! So I drop the bluetooth link from remote side then the event comes now! so I try to send all packets in the queue, the first one was sent successfully but the later packets didn't with error message 
"LOG -- rfcomm.c.2129: rfcomm_send_internal: l2cap cannot send now"
and refcomm_send returns BTSTACK_ACL_BUFFERS_FULL

Note that, with the former code before implementing your suggestion, the strange behavior occurs when I got data over uart port but when I generate data internally from the module it goes well and rfcomm can send now event comes always as expected and btstack never stops.

mringwal, 2017-08-08:
when you get the rfcomm_can_send_now_event, you sent the first packet in the queue. If you have more packets, you then call rfcomm_request_can_send_now again (without a trigger from the UART thread).

my-abousamra, 2017-08-08:
Yes, This is what I did first time, With every new packet came from queue I call rfcomm_request_can_send_now and wait for RFCOMM_EVENT_CAN_SEND_NOW event to send it and do that whenever new packet came in the queue. UART thread only reads packets from uart port and insert them to queue without triggering bluetooth.
With this implementation the problem occurs but if commented the part of UART port read in UART thread and only insert a dummy data to the queue the problem doesn't occur and bluetooth continues its work properly!

mringwal, 2017-08-08:
almost

after you get RFCOMM_EVENT_CAN_SEND_NOW and send the packet, you have to check if you have more packets to send. if yes, call rfcomm_request_can_send_now again

my-abousamra, 2017-08-08:
Yes, This is what I did already

mringwal, 2017-08-08:
weird. please post a new log, including the packet data (enable hci_dump_open .. in main()), and without ENABLE_LOG_DEBUG>

my-abousamra, 2017-08-08:
Here's another scenario, The application stops for about minute but I send a byte from the remote device (before queue gets full) that makes btstack works again and send data from the queue

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3416
load:0x40078000,len:10384
load:0x40080000,len:252
entry 0x40080034
I (1311) UART_vInit:
[00:00:01.313] LOG -- C:/Users/pulse/workspace/spp_counter_8/components/Bt_Stack/../../../../../../esp/btstack/platform/freertos/btstack_run_loop_freertos.c.256: run loop init, queue item size 8
[00:00:01.321] LOG -- main.c.247: transport_register_packet_handler
[00:00:01.327] LOG -- rfcomm.c.2280: RFCOMM_REGISTER_SERVICE channel #1 mtu 65535 flow_control 0 credits 10
[00:00:01.336] LOG -- l2cap.c.2014: L2CAP_REGISTER_SERVICE psm 0x3 mtu 65535
[00:00:01.343] LOG -- l2cap.c.2014: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535
I (1350) Bt_Classic_vInit: SDP service record size: 95

[00:00:01.356] LOG -- hci.c.2588: hci_power_control: 1, current mode 0
[00:00:01.362] LOG -- main.c.197: transport_init
[00:00:01.366] LOG -- main.c.213: transport_open
[00:00:01.561] LOG -- hci.c.3635: BTSTACK_EVENT_STATE 1
[00:00:01.561] EVT <= 60 01 01
[00:00:01.562] CMD => 03 0C 00
I (1563) uart_send_task: uart_send_task start
I (1563) uart_recv_task: uart_recv_task start
I (1571) bt_classic_monitor_task: bt_classic_monitor_task start
I (1571) bt_classic_send_task: bt_classic_send_task start
[00:00:01.571] EVT <= 6E 00
[00:00:01.587] EVT <= 0E 04 05 03 0C 00
[00:00:01.590] CMD => 01 10 00
[00:00:01.593] EVT <= 0E 0C 05 01 10 00 08 0E 03 08 60 00 0E 03
[00:00:01.599] LOG -- hci.c.1758: Manufacturer: 0x0060
[00:00:01.603] EVT <= 6E 00
[00:00:01.606] CMD => 14 0C 00
[00:00:01.609] EVT <= 0E FC 05 14 0C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:01.677] LOG -- hci.c.1672: local name:
[00:00:01.681] LOG -- hci.c.1464: Received local name, need baud change 0
[00:00:01.688] EVT <= 6E 00
[00:00:01.690] CMD => 02 10 00
[00:00:01.693] EVT <= 0E 44 05 02 10 00 BF FF FF 03 CC FF EF FF 3F FF FC 1F F2 0F E8 FE 3F F7 8F FF 1C 00 04 00 61 F7 FF FF 7F 38 C0 FF 7F FF FF 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:01.714] LOG -- hci.c.1768: Local supported commands summary 0x3f
[00:00:01.720] EVT <= 6E 00
[00:00:01.723] CMD => 09 10 00
[00:00:01.726] EVT <= 0E 0A 05 09 10 00 F2 42 03 C4 0A 24
[00:00:01.731] LOG -- hci.c.1720: Local Address, Status: 0x00: Addr: 24:0A:C4:03:42:F2
[00:00:01.738] EVT <= 6E 00
[00:00:01.741] CMD => 05 10 00
[00:00:01.744] EVT <= 0E 0B 05 05 10 00 FD 03 FF 09 00 04 00
[00:00:01.749] LOG -- hci.c.1689: hci_read_buffer_size: ACL size module 1021 -> used 1021, count 9 / SCO size 255, count 4
[00:00:01.760] EVT <= 6E 00
[00:00:01.763] CMD => 03 10 00
[00:00:01.766] EVT <= 0E 0C 05 03 10 00 BF EE CD FE DB FF 7B 87
[00:00:01.771] LOG -- hci.c.1747: Packet types cc18, eSCO 1
[00:00:01.776] LOG -- hci.c.1750: BR/EDR support 1, LE support 1
[00:00:01.782] EVT <= 6E 00
[00:00:01.785] CMD => 01 0C 08 FF FF FF FF FF FF FF 3F
[00:00:01.790] EVT <= 0E 04 05 01 0C 00
[00:00:01.793] EVT <= 6E 00
[00:00:01.796] CMD => 56 0C 01 01
[00:00:01.799] EVT <= 0E 04 05 56 0C 00
[00:00:01.803] EVT <= 6E 00
[00:00:01.805] CMD => 18 0C 02 00 60
[00:00:01.809] EVT <= 0E 04 05 18 0C 00
[00:00:01.812] EVT <= 6E 00
[00:00:01.815] CMD => 24 0C 03 12 3C 01
[... 798 lines cut ...]
[00:01:25.391] EVT <= 78 02 41 00
[00:01:25.406] EVT <= 13 05 01 80 00 01 00
[00:01:25.409] ACL <= 80 20 08 00 04 00 41 00 03 53 01 FD
[00:01:25.409] LOG -- rfcomm.c.1082: Received DISC #0, (ougoing = 0)
[00:01:25.412] EVT <= 78 02 41 00
[00:01:25.415] LOG -- rfcomm.c.1215: Sending UA #0
[00:01:25.419] LOG -- rfcomm.c.1216: Closing down multiplexer
[00:01:25.425] ACL => 80 00 08 00 04 00 41 00 03 73 01 D7
[00:01:25.430] LOG -- l2cap.c.1008: L2CAP_DISCONNECT local_cid 0x41 reason 0x13
[00:01:25.437] EVT <= 6E 00
[00:01:25.440] ACL => 80 00 0C 00 08 00 01 00 06 04 04 00 41 00 41 00
[00:01:25.446] EVT <= 6E 00
[00:01:25.449] EVT <= 78 02 41 00
[00:01:25.452] EVT <= 13 05 01 80 00 01 00
[00:01:25.455] ACL <= 80 20 0C 00 08 00 01 00 06 39 04 00 41 00 41 00
[00:01:25.462] LOG -- l2cap.c.1398: L2CAP signaling handler code 6, state 11
[00:01:25.469] ACL => 80 00 0C 00 08 00 01 00 07 39 04 00 41 00 41 00
[00:01:25.475] LOG -- l2cap.c.334: L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x41
[00:01:25.481] EVT <= 71 02 41 00
[00:01:25.485] LOG -- rfcomm.c.1032: L2CAP_EVENT_CHANNEL_CLOSED cid 0x41, mult 0x0
[00:01:25.492] LOG -- l2cap.c.423: l2cap_stop_rtx for local cid 0x41
[00:01:25.498] EVT <= 13 05 01 80 00 01 00
[00:01:25.502] ACL <= 80 20 0C 00 08 00 01 00 07 04 04 00 41 00 41 00
[00:01:25.508] EVT <= 13 05 01 80 00 01 00
[00:01:25.512] EVT <= 6E 00

mringwal, 2017-08-08:
BTstack doesn't give you an rfcomm can send now event. 
This happens, because the "HCI Transport" doesn't generate a "packet sent" event. Packet sent is the EVT <= 6E 00 in the logs. After the last valid RFCOMM send/ACL data packet, there is no Packet Sent Event.

Now, the question is: why is there no Packet Sent event?

It should be triggered by host_send_pkt_available_cb in main.c. 

Can you reproduce this with the spp_streamer / spp_streamer_client demos as well?

mringwal, 2017-08-08:
Thanks for the second log. 

At the moment you disconnect, a new HCI Event (HCI Disconnection Complete 13 05 ....) gets delivered, and that one also triggers the btstack run loop which then "realizes" that the packet has already been sent. 

I would assume that the packet is actually sent to the remote, but BTstack didn't go through all data sources again.

---

Uhm... Looking at your code again: you're calling rfcomm_request_can_send_now not from the BTstack thread - that's not supported and most likely explains the problem!

To integrate, you need to trigger your code - that checks the queue and sends data - by registering a btstack_data_source_t and then call btstack_run_loop_freertos_trigger() from the data generation thread. 

Please let me know if that works. You may also use a simple timer to check every x ms, but that's not as elegant/efficient.

my-abousamra, 2017-08-08:
Your conclusion seems right. But regarding to my code that you said it is not supported, It worked for me without this problem when UART thread generates dummy packets and inserts them into the queue in stead of reading packets from UART port. Until now I can't find explanation for why reading from UART port causes this problem. Can you?

After all, I will try your suggestion of triggering my code and I will let you know

mringwal, 2017-08-08:
Not supported = It might or might not work. If it works, it might work for a while, but in some context, it will fail... Right now, all BTstack functions must be called from the BTstack thread. Everything else might cause weird behaviour.

my-abousamra, 2017-08-09:
Hi,

> To integrate, you need to trigger your code - that checks the queue and sends data - by registering a btstack_data_source_t and then call btstack_run_loop_freertos_trigger() from the data generation thread.

I tried this solution but the problem still exist, It stops until I send a byte from the remote device. 
Now I call all btstack functions from btstack thread except btstack_run_loop_freertos_trigger() I call it from receiving queue thread

**code:**

static btstack_data_source_t queue_data;

static void bt_classic_send_task (void *pvParameters)
{
	ESP_LOGI("bt_classic_send_task", "bt_classic_send_task start");

	while (1)
	{
		xEventGroupWaitBits ( bt_classic_event_group, BT_CLASSIC_READY_BIT, pdTRUE, pdFALSE, portMAX_DELAY );

		if ( xQueueReceive( xBt_Queue, &bt_q_rd_msg, portMAX_DELAY ) == pdPASS )
		{
			ESP_LOGI("bt_classic_send_task", "new packet in the queue, call rfcomm_request_can_send_now_event");
			btstack_run_loop_freertos_trigger ();
		}
		else
		{
			ESP_LOGE( "bt_classic_send_task", "Can't receive data from bt classic queue [%d]", uxQueueMessagesWaiting (xBt_Queue));
		}
	}
}

static void queue_data_received_handler(btstack_data_source_t * ds, btstack_data_source_callback_type_t callback_type)
{
	UNUSED(ds);
	UNUSED(callback_type);

	if ( ( xEventGroupGetBits ( bt_classic_event_group ) & BT_CLASSIC_CONNECTED_BIT ) != 0 )
	{
		ESP_LOGW("queue_data_received_handler", "rfcomm_request_can_send_now_event");
		rfcomm_request_can_send_now_event ( rfcomm_channel_id );
	}
}

// In initialization
// turn on!
hci_power_control(HCI_POWER_ON);

btstack_run_loop_set_data_source_handler(&queue_data, &queue_data_received_handler);
btstack_run_loop_enable_data_source_callbacks(&queue_data, DATA_SOURCE_CALLBACK_POLL);
btstack_run_loop_add_data_source(&queue_data);

my-abousamra, 2017-08-09:
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3416
load:0x40078000,len:10384
load:0x40080000,len:252
entry 0x40080034
I (1309) UART_vInit:
[00:00:01.311] LOG -- C:/Users/pulse/workspace/spp_counter_8/components/Bt_Stack/../../../../../../esp/btstack/platform/freertos/btstack_run_loop_freertos.c.256: run loop init, queue item size 8
[00:00:01.319] LOG -- main.c.247: transport_register_packet_handler
[00:00:01.325] LOG -- rfcomm.c.2280: RFCOMM_REGISTER_SERVICE channel #1 mtu 65535 flow_control 0 credits 10
[00:00:01.334] LOG -- l2cap.c.2014: L2CAP_REGISTER_SERVICE psm 0x3 mtu 65535
[00:00:01.341] LOG -- l2cap.c.2014: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535
I (1348) Bt_Classic_vInit: SDP service record size: 95
[00:00:01.353] LOG -- hci.c.2588: hci_power_control: 1, current mode 0
[00:00:01.360] LOG -- main.c.197: transport_init
[00:00:01.364] LOG -- main.c.213: transport_open
[00:00:01.565] LOG -- hci.c.3635: BTSTACK_EVENT_STATE 1
[00:00:01.565] EVT <= 60 01 01
[00:00:01.566] CMD => 03 0C 00
I (1567) uart_send_task: uart_send_task start
I (1567) uart_recv_task: uart_recv_task start
I (1575) bt_classic_monitor_task: bt_classic_monitor_task start
I (1575) bt_classic_send_task: bt_classic_send_task start
[00:00:01.575] EVT <= 6E 00
[00:00:01.590] EVT <= 0E 04 05 03 0C 00
[00:00:01.594] CMD => 01 10 00
[00:00:01.597] EVT <= 0E 0C 05 01 10 00 08 0E 03 08 60 00 0E 03
[00:00:01.602] LOG -- hci.c.1758: Manufacturer: 0x0060
[00:00:01.607] EVT <= 6E 00
[00:00:01.610] CMD => 14 0C 00
[00:00:01.613] EVT <= 0E FC 05 14 0C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:01.681] LOG -- hci.c.1672: local name:
[00:00:01.685] LOG -- hci.c.1464: Received local name, need baud change 0
[00:00:01.692] EVT <= 6E 00
[00:00:01.694] CMD => 02 10 00
[00:00:01.697] EVT <= 0E 44 05 02 10 00 BF FF FF 03 CC FF EF FF 3F FF FC 1F F2 0F E8 FE 3F F7 8F FF 1C 00 04 00 61 F7 FF FF 7F 38 C0 FF 7F FF FF 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:01.717] LOG -- hci.c.1768: Local supported commands summary 0x3f
[00:00:01.724] EVT <= 6E 00
[00:00:01.726] CMD => 09 10 00
[00:00:01.729] EVT <= 0E 0A 05 09 10 00 F2 42 03 C4 0A 24
[00:00:01.734] LOG -- hci.c.1720: Local Address, Status: 0x00: Addr: 24:0A:C4:03:42:F2
[00:00:01.742] EVT <= 6E 00
[00:00:01.745] CMD => 05 10 00
[00:00:01.748] EVT <= 0E 0B 05 05 10 00 FD 03 FF 09 00 04 00
[00:00:01.753] LOG -- hci.c.1689: hci_read_buffer_size: ACL size module 1021 -> used 1021, count 9 / SCO size 255, count 4
[00:00:01.764] EVT <= 6E 00
[00:00:01.766] CMD => 03 10 00
[00:00:01.769] EVT <= 0E 0C 05 03 10 00 BF EE CD FE DB FF 7B 87
[00:00:01.775] LOG -- hci.c.1747: Packet types cc18, eSCO 1
[00:00:01.780] LOG -- hci.c.1750: BR/EDR support 1, LE support 1
[00:00:01.786] EVT <= 6E 00
[00:00:01.789] CMD => 01 0C 08 FF FF FF FF FF FF FF 3F
[00:00:01.794] EVT <= 0E 04 05 01 0C 00
[00:00:01.797] EVT <= 6E 00
[00:00:01.800] CMD => 56 0C 01 01
[00:00:01.803] EVT <= 0E 04 05 56 0C 00
[00:00:01.807] EVT <= 6E 00
[00:00:01.809] CMD => 18 0C 02 00 60
[00:00:01.813] EVT <= 0E 04 05 18 0C 00
[00:00:01.816] EVT <= 6E 00
[00:00:01.819] CMD => 24 0C 03 12 3C 01
[00:00:01.823] EVT <= 0E 04 05 24 0C 00
[... 513 lines cut ...]
[00:00:26.794] EVT <= 78 02 41 00
[00:00:26.797] EVT <= 89 02 01 00
I (26800) bt_classic_packet_handler: RFCOMM_EVENT_CAN_SEND_NOW
[00:00:26.807] ACL => 80 00 B1 00 AD 00 41 00 09 EF 50 01 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40 41 42 43 44 45 46 47 48 49 4A 4B 4C 4D 4E 4F 50 51 52 53 54 55 56 57 58 59 5A 5B 5C 5D 5E 5F 60 61 62 63 64 65 66 67 68 69 6A 6B 6C 6D 6E 6F 70 71 72 73 74 75 76 77 78 79 7A 7B 7C 7D 7E 7F 80 81 82 83 84 85 86 87 88 89 8A 8B 8C 8D 8E 8F 90 91 92 93 94 95 96 97 98 99 9A 9B 9C 9D 9E 9F A0 A1 A2 A3 A4 A5 A6 A7 40
I (26856) bt_classic_packet_handler: rfcomm_send success [24] sz[168]
I (26863) bt_classic_send_task: new packet in the queue, call rfcomm_request_can_send_now_event
[00:00:26.872] EVT <= 13 05 01 80 00 01 00
[00:00:26.876] EVT <= 13 05 01 80 00 01 00
[00:00:26.880] ACL <= 80 20 09 00 05 00 41 00 0B FF 01 04 86
[00:00:26.885] LOG -- rfcomm.c.1289: RFCOMM data UIH_PF, new credits: 4, now 25
W (26892) queue_data_received_handler: rfcomm_request_can_send_now_event
[00:00:26.900] EVT <= 6E 00
[00:00:26.902] EVT <= 78 02 41 00
[00:00:26.905] EVT <= 89 02 01 00
[... cut ...] 83 84 85 86 87 88 89 8A 8B 8C 8D 8E 8F 90 91 92 93 94 95 96 97 98 99 9A 9B 9C 9D 9E 9F A0 A1 A2 A3 A4 A5 A6 A7 40
I (26964) bt_classic_packet_handler: rfcomm_send success [23] sz[168]
I (26971) bt_classic_send_task: new packet in the queue, call rfcomm_request_can_send_now_event
[00:00:26.981] EVT <= 13 05 01 80 00 01 00
W (26985) queue_data_received_handler: rfcomm_request_can_send_now_event
[00:00:26.992] EVT <= 6E 00
[00:00:26.995] EVT <= 78 02 41 00
[00:00:26.998] EVT <= 89 02 01 00

.......

my-abousamra, 2017-08-09:
Note that this problem occurs after the first time of pairing but later connections (disconnect and connect again) it works fine every time

mringwal, 2017-08-09:
In the last log, BTstack doesn't provide a 'can send now' as it already used all 7 RFCOMM credits. During connection establishment, it gets 7 credits. Then a total of 7 RFCOMM user packets are sent. When you trigger the incoming RFCOMM data, it also contains credits and BTstack reports "new credits: 25, now: 25" which means BTstack had 0 credits before. This did not happen in the earlier logs.

hm... the first 7 RFCOMM packets are all empty! Could you try once more and send at least a single byte in each RFCOMM packet? It's quite likely that an empty RFCOMM packet doesn't count for the credit counting and BTstack decrements the credit counter nevertheless. If yes, I can fix in BTstack quickly...

uhm.. why do you send empty packets :) ?

my-abousamra, 2017-08-10:
No, I don't send empty packets, once btstack runs it calls data source handler which in turn calls rfcomm_request_can_send_now_event then rfcomm_send sends the empty buffer until send the actual packets but the credits were over.

Now, I created an event bit that set when new packet received and in data source handler checks for it to make sure that a new packet has arrived then call rfcomm_request_can_send_now_event and then rfcomm_send with the real packet but xEventGroupGetBits mostly indicates that all bits are empty although they are set. Why does that happen? does data source handler can't check actual event state of the event group?

my-abousamra, 2017-08-10:
Update:
calling xEventGroupClearBits before rfcomm_request_can_send_now_event solved event group problem.
Now, It works fine for a while. I will let it stream for a long time to make sure no problems occur

Thanks

mringwal, 2017-08-10:
You were sending empty packets, and BTstack did count them although it shouldn't. Thanks for helping me identify this issue. I just pushed an update to the develop branch that fixes this (doesn't count empty packets).

I don't understand your question. If you call rfcomm_request_can_send_now_event() (or any other BTstack function), it might call you back before it returns, so you should always update your state before calling BTstack functions. I guess that's how you've fixed it.

my-abousamra, 2017-08-10:
oh, I mean I don't send packets from application on purpose but btstack at start up (and before application starts to generate packets) calls data source handler which calls rfcomm_request_can_send_now_event and RFCOMM_EVENT_CAN_SEND_NOW event was sent and rfcomm_send sends the empty buffer (supposed to have the packet)

For the second problem, Yes, This solved it

mringwal, 2017-08-10:
On ESP32 (and FreeRTOS), data sources are currently polled. In the callback, you could call rfcomm_request_can_send_now_event only if you have data to send

Anyway, it looks like you're all set now. :)

my-abousamra, 2017-08-10:
Yes, This is what I did and it goes well :)

Thank you