bluekitchen/btstack #60: Stop scanning after few time

esantosjr, 2017-08-15:
Hi,

I am developing a project using ESP32 with the gap_le_advertisements example that must scan for beacons and print its parameters as MAC address. I was able to run and see the needed informations, but  I have to do this continuously and after some hours the scanning stopped. No error message is showed.

Do you guys think that is a problem with the hardware or the firmware?
What is the best way to implement this?

Thanks.

mringwal, 2017-08-15:
If it works for a few hours and then stops, it looks more like a hardware problem. All the software/stack does is passing up the LE Advertisements to the app. There should not be any bugs that are triggered after hours.

You can enable the packet logger in main.c (hci_dump_open) and the check the log when it stops. I can also start it later and check tomorrow if this happens to me, too.

mringwal, 2017-09-25:
Seems to have been fixed ESP32 in this commit https://github.com/espressif/esp32-bt-lib/commit/2d33374ca531f92859640f1d055fc4f28d60dfdc

Please update esp-idf and try again - works for me (it's running now about 6h)

closed 2017-09-25 by mringwal