bluekitchen/btstack #64: ESP32: Error with uploading sm_pairing_peripheral.c

srinadhpotnuru, 2017-08-17:
Error uploading sm_pairing_peripheral.c through make commands and the following error comes up
/btstack/port/esp32/sm_pairing_peripheral/main/sm_pairing_peripheral.c:55:35: fatal error: sm_pairing_peripheral.h: No such file or directory.

I could not find the file anywhere in the stack btstack also.

mringwal, 2017-08-17:
It is created by the update_gatt.sh script, which in turn is generated by port/esp32/create_examples.py. 

Are you on Windows?

srinadhpotnuru, 2017-08-17:
Yes

mringwal, 2017-08-17:
I see. You may run 
`../../../tool/compile_gatt.py ../../../example/sm_pairing_peripheral.gatt sm_pairing_peripheral.h`
 manually. 

I guess I need to create a python script instead of the shell script.

srinadhpotnuru, 2017-08-17:
Thanks, that solved the issue.

closed 2017-08-17 by srinadhpotnuru