bluekitchen/btstack #67: BCM20733 Problem---2!!!

WallGeorg, 2017-08-21:
Hi.
I have BT3GMD-B60P module.
[image]

I don't know SDK and download device and connect method...
I hope your help.
Best regards.

mringwal, 2017-08-22:
Hi. We don't know either :)

In the current WICED Studio 5.1 there is only support for 43xxx-Wifi and 20706-A2. Please check with Broadcom/Cypress how to work with this module and if allows to use a third-party Bluetooth stack.

WallGeorg, 2017-08-22:
Hi.
I heard that SDK2.2.3 is supported for BCM20733.
But I don't confirm it.
If you know this method, let's me know.
Could you addition this SDK to your project.
I don't know this method.
Please, please Help me.
Best regards.

mringwal, 2017-08-22:
The oldest WICED SDK on Cypress' site is for 3.x. There is a single question about the BCM20733 in the Cypress forum. Looks like this chip is not publicly supported by Cypress.

WallGeorg, 2017-08-22:
Hi.
https://community.cypress.com/community/wiced-smart/wiced-smart-documentation
Pls, See this URL.
I hope your help.
Best regards.

mringwal, 2017-08-22:
The WICED Smart SDK only covers BLE only chipsets. It does not cover the 20733, which seems to be an ARM Cortex M3 with a dual-mode Bluetooth radio.

Thanks for the nice overview pdf. Please contact Cypress Sales to get access to their HID Development Kit. It cannot be used with a different stack easily.

WallGeorg, 2017-08-22:
Hi.
I know BCM20733 hasn't BLE, but there is ARM Cortex-M3.
Please See attached PNG.
[image]
Could you explain?
I am a beginner.
I want to know develop method.
Best regards.

mringwal, 2017-08-22:
I'm sure it's technically possible. It looks like it's an SoC, which means the Bluetooth Controller is implemented in software and there might be an API to send/receive HCI packets. This would require reverse engineering existing BCM20733 firmware. 

As a beginner, I can only recommend to start with one of the already supported MCU + chipset combinations. And/or run BTstack on Linux with an USB Bluetooth dongle.

WallGeorg, 2017-08-22:
Hi. 
What is means?
[image]
I don't know develop method.
Also I don't understand "It looks like it's an SoC, which means the Bluetooth Controller is implemented in software and there might be an API to send/receive HCI packets. This would require reverse engineering existing BCM20733 firmware.

As a beginner, I can only recommend to start with one of the already supported MCU + chipset combinations. And/or run BTstack on Linux with an USB Bluetooth dongle."

Could you explain ?
I am very sorry.
Best regards.

mringwal, 2017-08-22:
To use the BCM20733 please contact Cypress sales team, not us.
I recommend you use a different port. The ESP32 is not that bad and you managed to get BTstack to run on it, just not the serial monitor.

WallGeorg, 2017-08-22:
Hi.
I found BCM20733 in SDK2.2.3.
What is mean?
[image]
Could you explain?
Please help me?

mringwal, 2023-02-21:
This ifdef is probably for code that runs on the 20733 itself. Again, help from Infineon/Cypress is needed here.

closed 2023-02-21 by mringwal