bluekitchen/btstack #71: ESP32 btstack can not scan all nearby beacons (BLE, LE)

hiik3n, 2017-08-31:
Hi all,
I tried to used btstack on esp32 to scan neayby beacons (BLE, LE).
I run gap_le_advertisements example to scan ble,  beside of it stops scanning after some hours (https://github.com/espressif/esp-idf/issues/942), I found that the scanning result is quite bad. I had 14 beacons arround but esp is able to scan few of them in one perioid of scanning (10s). While I use blueroid from esp-idf, it can scan all the beacons.
Anyone has any experience on this?

mringwal, 2017-08-31:
@hiik3n well. "all" is mainly just me. Feel free to post on the BTstack developer list: https://groups.google.com/forum/#!forum/btstack-dev

Scanning doesn't require much from the Bluetooth stack. It just sets the scanning parameters and tells the Bluetooth Controller to start scanning. The gap_le_advertisements also sets the scanning duty cycle to 100% and enables scan responses. If you disable them with 
    gap_set_scan_parameters(0,48,48); 
it might get a bit better.

What example did you use with Bluedroid?

mringwal, 2017-10-06:
hi. did you retest BLE Scanning after the fix for the problems scanning for longer times?

hiik3n, 2017-10-07:
i did not try with btstack lib, but i did try with the blueroid, and it can not fix the problem.
i also email to ask support team from espressif, they said they aware of this problem and still working on that

mringwal, 2017-10-09:
thanks for the update. so as you get the problem also with Bluedroid, it doesn't look like BTstack is "loosing" advertisement reports.

please let us know if there is any news on this, thanks!

closed 2018-03-28 by mringwal