bluekitchen/btstack #73: ESP32: btstack doesn't start in a single task

my-abousamra, 2017-09-13:
Hi,

My application starts btstack in app_main ESP32 thread and works fine, but when I start btstack in a single task (among others) it fails to start and no "BTstack up and running." appears.

**CODE:**

	void btstack_run()
	{
		btstack_run_loop_execute();
	}

	static void btstack_init(void){

		// enable packet logger
		 hci_dump_open(NULL, HCI_DUMP_STDOUT);

		/// GET STARTED with BTstack ///
		btstack_memory_init();
		btstack_run_loop_init(btstack_run_loop_freertos_get_instance());

		// init HCI
		hci_init(transport_get_instance(), NULL);
		hci_set_link_key_db(btstack_link_key_db_memory_instance()); // @TODO

		// inform about BTstack state
		hci_event_callback_registration.callback = &packet_handler;
		hci_add_event_handler(&hci_event_callback_registration);
	}

	static void bt_classic_init ()
	{
		btstack_init();

		// register for HCI events
		hci_event_callback_registration.callback = &packet_handler;
		hci_add_event_handler(&hci_event_callback_registration);

		l2cap_init();

		rfcomm_init();
		rfcomm_register_service(packet_handler, RFCOMM_SERVER_CHANNEL, 0xffff);  // reserved channel, mtu limited by l2cap

		// init SDP, create record for SPP and register with SDP
		sdp_init();
		memset(spp_service_buffer, 0, sizeof(spp_service_buffer));
		spp_create_sdp_record(spp_service_buffer, 0x10001, RFCOMM_SERVER_CHANNEL, "SPP Counter");
		sdp_register_service(spp_service_buffer);
		ESP_LOGI("bt_classic_init", "SDP service record size: %u", de_get_len(spp_service_buffer));

		gap_discoverable_control(1);
		gap_ssp_set_io_capability(SSP_IO_CAPABILITY_DISPLAY_YES_NO);
		gap_set_local_name("BTstack SPP ECG");
		gap_set_class_of_device(80914);	// check cod calculator => http://www.ampedrftech.com/cod.htm

		// turn on!
		hci_power_control(HCI_POWER_ON);

		btstack_run_loop_set_data_source_handler(&queue_data, &queue_data_received_handler);
		btstack_run_loop_enable_data_source_callbacks(&queue_data, DATA_SOURCE_CALLBACK_POLL);
		btstack_run_loop_add_data_source(&queue_data);
	}

	static void a_task (void *pvParameters)
	{
		bt_classic_init();

		while (1)
		{
			btstack_run();
		}
	}

**LOG:**

	[00:00:01.328] LOG -- btstack_run_loop_freertos.c.258: run loop init, queue item size 8
	I (1318) bt_classic_send_task: Start!
	[00:00:01.340] LOG -- main.c.247: transport_register_packet_handler
	[00:00:01.351] LOG -- rfcomm.c.2287: RFCOMM_REGISTER_SERVICE channel #1 mtu 65535 flow_control 0 credits 10
	[00:00:01.360] LOG -- l2cap.c.3101: L2CAP_REGISTER_SERVICE psm 0x3 mtu 65535
	[00:00:01.367] LOG -- l2cap.c.3101: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535
	I (1374) bt_classic_init: SDP service record size: 95
	[00:00:01.379] LOG -- hci.c.2639: hci_power_control: 1, current mode 0
	[00:00:01.386] LOG -- main.c.197: transport_init
	[00:00:01.390] LOG -- main.c.213: transport_open
	[00:00:01.421] LOG -- hci.c.3690: BTSTACK_EVENT_STATE 1
	[00:00:01.421] EVT <= 60 01 01
	[00:00:01.421] LOG -- hci.c.1078: hci_initializing_run: substate 0, can send 1
	[00:00:01.425] CMD => 03 0C 00
	[00:00:01.328] LOG -- btstack_run_loop_freertos.c.176: RL: execute
	[00:00:01.711] LOG -- btstack_run_loop_freertos.c.206: RL: portMAX_DELAY 4294967295
	[00:00:01.713] LOG -- btstack_run_loop_freertos.c.210: RL: now 1713, expires 1625
	[00:00:01.720] LOG -- btstack_run_loop_freertos.c.217: RL: first timer 0x400e7030
	0x400e7030: hci_initialization_timeout_handler at C:/Users/user/workspace/spp_counter_8/components/Bt_Stack/../../../../../../esp/btstack/src//hci.c:3844

	[00:00:01.727] LOG -- hci.c.1012: Resend HCI Reset
	[00:00:01.732] LOG -- btstack_run_loop_freertos.c.222: RL: wait with timeout 4294967295

But this way it works fine and "BTstack up and running." appears

	void app_main ()
	{
		...
		
		bt_classic_init();

		btstack_run();
	}

mringwal, 2017-09-14:
Your code looks fine so far but it doesn't get a response from VHCI.

Only idea: How large is the stack for the second task? (although you should get some other log messages when it crashes because the stack is to small).

my-abousamra, 2017-09-15:
a_task stack size is 3072 bytes and it doesn't crash just stopped

mringwal, 2017-09-15:
In the log above, there's no event that indicates that the HCI Reset packet was sent. Without that, the stack gets stuck. We need to find out why the packet sent callback is not called when running on the second thread.

my-abousamra, 2017-09-24:
Actually, I can't figure out the case!!
If you have time, Please try this example code, It doesn't do anything just starting btstack from a new task or from app_main stack according to BTSTACK_THREAD macro, The first one can't get the stack up, the second gets it up and running.

[spp_counter_btstack_thread.zip](https://github.com/bluekitchen/btstack/files/1327245/spp_counter_btstack_thread.zip)

mringwal, 2017-09-26:
hi. I've heard that Bluetooth cannot be used from a different thread and that there seems to be discussion in the EPS32 Forum. I didn't find that quickly, but I hope to get a link to this discussion.

in the meantime: what's the reason to run BTstack on another thread? (maybe the issue can be 'worked around')

my-abousamra, 2017-10-03:
bluetooth is part of a bigger application with multiple tasks so I want to put in a thread to be able to change its priority in order not to be starved with higher priority tasks as bluetooth started from app_main thread which has the lowest priority (1)

mringwal, 2017-10-13:
hi @my-abousamra I hope I have good news for you.
I've started with the default example code from the BTstack repo (develop branch) and changed the main.c code to look like this

```
// main
static void btstack_task(void *pvParameters){
    // all code from app_main goes here..
}

int app_main(void){
    if ( xTaskCreate( &btstack_task, "btstack_task", 3072, NULL, 7, NULL ) != pdPASS ){
        while(1){
            printf("error creating BTstack task\n");
        }
    }
    int counter = 0;
    while(1){
        printf("BTstack runs in other task - counter %u\n", counter++);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    return 0;
}

```

This starts up without problems. So, at least it is not a problem with BTstack running in another thread. Could you start over with the current BTstack code base and add your application back in piece by piece to see if it breaks again and when/where?

mringwal, 2017-11-09:
No feedback, works for me.

closed 2017-11-09 by mringwal