bluekitchen/btstack #75: CC2564 EHCILL

xfilesee, 2017-09-15:
Hi,

I am trying to use EHCILL in cc2564 chipset. I observed 2 issues:
1) When I use that protocol I observe up to 4 seconds delay in send and receive of data.
2) hal_uart_dma_set_csr_irq_handler is never called with NULL parameter.

Are these 2 issues normal?

mringwal, 2017-09-15:
The multi second delay isn't normal. Maybe you can explain your system setup and post a full packet log (see hci_dump_open..) that includes timestamps for analysis.

I don't understand the second one. hal_uart_dma_set_csr_irq_handler is called with NULL e.g. by btstack_uart_embedded_set_sleep in btstack_uart_block_embedded.c.

xfilesee, 2017-09-15:
I am trying to use cc2564 with spp_counter example.
I get a lot of these messages:
EVT <= 78 02 41 00
EVT <= 89 02 01 00
How can I interpret these messages?

The second one: What I mean is when I insert a debug message to check whether the function is called with null parameter that message is never printed.
Thanks.

mringwal, 2017-09-15:
The non-standard events are defined in src/btstack_defines.h

xfilesee, 2017-09-16:
Can you comment on the second issue? I am using hci_transport_h4_ehcill_embedded.c for hci transport. 
BTSTACK_UART_SLEEP_OFF is never used in that source file.

mringwal, 2017-09-16:
The hci_transport_h4_ehcill_embedded.c is old/deprectated.
Please use hci_transport_h4.c + btstack_uart_block_embedded.c instead.

xfilesee, 2017-09-16:
Thanks for your answer. This was what I thought so I had removed my question before you answered. For future visitors the question was related to:
I was using hci_transport_h4_ehcill_embedded.c for ehcill and that was causing the 2nd issue I mentioned in the previous post.

xfilesee, 2017-09-18:
Here is the log of going to deep state:

[00:00:05.432] LOG -- hci_transport_h4.c.532: eHCILL: Received GO_TO_SLEEP_IND RX
[00:00:05.440] LOG -- hci_transport_h4.c.505: eHCILL: schedule eHCILL command 31
[00:00:05.447] LOG -- hci_transport_h4.c.477: eHCILL: set timer for sending command 31
[00:00:05.505] LOG -- hci_transport_h4.c.469: eHCILL: timer triggered
[00:00:05.508] LOG -- hci_transport_h4.c.457: eHCILL: send command 31
[00:00:05.514] LOG -- hci_transport_h4.c.626: eHCILL: handle eHCILL sent, command was 31
[00:00:05.522] LOG -- hci_transport_h4.c.631: eHCILL: GO_TO_SLEEP_ACK sent, enter sleep mode

Although I enable hci commands log (hci_dump_open(NULL, HCI_DUMP_STDOUT);) I do not see any command sent that includes GO_TO_SLEEP_ACK (0x31) in the log. Are not sleep and wake hci commands logged?

mringwal, 2017-09-18:
The eHCILL commands are not in the log as they are not actual HCI commands and I don't think they can be stored in Apple's PacketLogger or the BTSnoop format.

In your log, the 'eHCILL: GO_TO_SLEEP_ACK sent, enter sleep mode' show that sleep was entered.

closed 2017-10-02 by xfilesee