bluekitchen/btstack #77: spp_counter does not work when i include an i2c read in the heartbeat handler

Yuggo1, 2017-09-16:
Hi, I'm trying to integrate an i2c IMU and adc code with the spp_counter example, the adc was very easy to integrate, i have made an interface of the esp32 with the BNO055 imu driver and it is working, every code that i put in the spp_service_setup of the BNO055 runs ok, but if i try to read the IMU data through i2c in the heartbeat_handler its returning the following error:
Guru Meditation Error of type LoadProhibited occurred on core 0. Exception was unhandled.
I'm using pin 22 and 21 for i2c communication. Does any one have any idea why is this happening?
Sorry for my bad english.

mringwal, 2017-09-16:
Hi. I did not try other peripherals on the ESP32. I know that someone used a software I2C implementation successfully.

If you do the same I2C read code in the setup, e.g. in btstack_main()... does it work?

Does it work, if you start the heartbeat timer in btstack_main() and don't start Bluetooth (hci_power_control(..ON))?

Does it work if you create a second thread which does the I2C handling? You could then store the data in an queue and use a btstack_data_source_t to trigger BTstack to process it.

Do you have a link to the I2C code you're using?

Feel free to post on the BTstack Mailing List: http://groups.google.com/group/btstack-dev
or try to ask in the ESP32 IRC channel. I'm probably the only one reading the issues here.

Yuggo1, 2017-09-18:
Hi @mringwal, i had two problems, the first was that i didnt test the i2c code properly after i integrated it with the spp_counter code, second was that it was not in a separated thread.   Fixing that i was able to run the code ok.
if you want to take a look at the code i'll put the links here:
The link for the BNO055 i2c interface for esp idf that i've made is here: https://github.com/Yuggo1/ESP32_BNO055
And here is the code where i integrated it:
https://github.com/Yuggo1/ESP_Gloves
Sorry for the bad code and thanks for the advices.

mringwal, 2017-09-18:
Glad you got it working. It should also working without a second task. If not, it will get tricky to debug.

closed 2017-09-18 by mringwal