bluekitchen/btstack #78: How to debug with .launch file in eclipse?

Yanjiong, 2017-09-21:
Hi,

For the steps "start it via the provided Debug Launch configuration "example-debug"", how to do that in current eclipse?

Thanks.

BR.

mringwal, 2017-09-22:
Do you refer to the STM32F4 Discovery port features in our blog?
What Eclipse do you have and what changed?

Yanjiong, 2017-09-27:
Yes, stm32-f4discovery-cc256x port examples.
Eclipse is Neon.3 Release.

Thanks.

mringwal, 2017-09-27:
I've used Neon.3 Release while writing the blog post.
If you click on the little triangle next to the "Debug" button, a list with debug launch configurations show up, like here:

[image]

Does this help?

Yanjiong, 2017-09-27:
no debug menu on my view...

[image]

Is it needed to add it in some way?

mringwal, 2017-09-27:
That's weird. The debug config is the file 'spp_counter-debug.launch' seen in your project explorer. I didn't use Eclipse before or after this port and I remember that the debug configs did show up twice before (not so bad, just not tidy).

As an alternative, you could flash the compiled .elf with openocd or the st-link tool. See the other STM32 examples.

Yanjiong, 2017-09-30:
Is there any instruction about how to develop btstack at STM32f4discovery+CC256X in command line mode under ubuntu?

mringwal, 2017-09-30:
I don't have anything written. The STM32F4Discovery has an ST-Link programming adapter. You can either use the st-link tool (https://github.com/texane/stlink) or OpenOCD (http://openocd.org) to download firmware and/or debug. Please google for tutorials on this.

In general, I usually just download the firmware and use log_info statements together with the packet logging, since pausing the program in the debugger often causes Bluetooth timeouts.

closed 2017-09-30 by mringwal