bluekitchen/btstack #79: HOG demos compilation

kat-kai, 2017-09-26:
Hi,
I used examples of BTstack port for ESP32 in the develop branch.
I could compile HID demos. However, I could not compile HOG demos
due to a lack of the corresponding header files: hog_keyboard_demo.h or hog_mouse_demo.h

Now, are the HOG demos in preparation?

Best regards,

mringwal, 2017-09-27:
If you run port/esp32/create_examples.py, the example folders will be created and also an update_gatt.py script. This script is run to generate hog_[keyboard|mouse]_demo.h from hog_[keyboard|mouse]_demo.gatt in the example folder.

I just run ./create_examples.py and the .h files are generated as expected. (Also hog_mouse_demo compiled, too)

closed 2017-09-27 by mringwal

kat-kai, 2017-09-28:
Hi,
Thank you for your kind explanation. I could do it.
Best regards,

mringwal, 2017-09-29:
You're welcome. I improved on the esp32 integration. Make sure to read the esp32/readme after the next update(s).  In short, run ./integrate_btstack.py instead of ./create_examples.py

kat-kai, 2017-10-01:
Thank you so much for your guidance. Sure, I'll read it.