bluekitchen/btstack #80: ESP32 ADC2 conflict

Yuggo1, 2017-10-03:
Hi, i want to use the adc1 and adc2 channels from esp32 integrated with the bluetooth, i'm using the spp_counter as a base code, this is the repo with the code:
https://github.com/Yuggo1/ESP_Gloves
The code to support the adc2 i've got from here https://github.com/ESP32DE/esp-idf/commit/f467f68ff50d679ac4004eea2dcaa812d50912a7
and it works when using it without the bluetooth, i read here https://github.com/espressif/arduino-esp32/issues/102 that ADC2 conflicts only with the wifi, but i'm getting the same problem using the bluetooth, does anyone know if btstack uses some wifi resources that could cause this error, or could help me to look where the ADC2 code conflicts with the btstack?

Best Regards

MichaelEL65, 2017-12-08:
I ran into this same issue with Bluetooth. Had a project that used 3 ADCs that ended up on ADC2 channels (the way the board was laid out, it made much more sense to route them there). When the bluetooth is turned off, there is no issue, but when it is turned on, all three immediately read 4095, even after rebooting from deep sleep mode.

mringwal, 2017-12-12:
Could/did you post on the esp-idf issue tracker? BTstack only uses the provided binary Bluetooth libraries and doesn't do any HW configuration by itself. If it happens with BTstack, it will also happen with the original Bluedroid stack.

mringwal, 2023-02-21:
Closed due to inactivity.

closed 2023-02-21 by mringwal