bluekitchen/btstack #81: connection issue

GustavoOlego, 2017-10-04:
Hi I have a ESP-ROOM-32 board , and could connect my celular phone and my raspberry pi3 by  BLUETOOTH means and saw working the spp_counter, but now I need to connect a SEPURA STP-9000 TETRA NETWORK with BT facility and I could pair the device but when i tried to connect the terminal keep waiting and nothing happens, I reconpiled with some featured recomended to have more log from ESP but I can no figure out what it happens the las line of the log is
 "btstack_link_key_db_tlv.c.95: tag 42544c0b, addr 00:1E:96:01:65:4F" I think the problem is with the key link,
Many Thanks for your help
Ing Gustavo Olego

mringwal, 2017-10-04:
HI Gustavo. Please post the packet log, too. You may also post on the BTstack Mailing list to get more people to see it.

GustavoOlego, 2017-10-04:
Sr Mwingwall many thanks for you advice, I could no find the Mailing list could you help me with that?

MONITOR
--- idf_monitor on COM14 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:4348
load:0x40078000,len:0
load:0x40078000,len:11312
entry 0x40078c98
I (566) heap_init: Initializing. RAM available for dynamic allocation:
I (566) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (585) heap_init: At 3FFC7B40 len 000184C0 (97 KiB): DRAM
I (641) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (700) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (759) heap_init: At 40093A14 len 0000C5EC (49 KiB): IRAM
BTstack: setup
[00:00:00.933] LOG -- btstack_run_loop_freertos.c.258: run loop init, queue item size 8
[00:00:00.934] LOG -- main.c.251: transport_register_packet_handler
[00:00:00.945] LOG -- rfcomm.c.2287: RFCOMM_REGISTER_SERVICE channel #1 mtu 65535 flow_control 1 credits 1
[00:00:00.947] LOG -- l2cap.c.3100: L2CAP_REGISTER_SERVICE psm 0x3 mtu 65535
[00:00:00.954] LOG -- l2cap.c.3100: L2CAP_REGISTER_SERVICE psm 0x1 mtu 65535
SDP service buffer size: 95
SPP FlowControl Demo: simulates processing on received data...

[00:00:00.969] LOG -- hci.c.2609: hci_power_control: 1, current mode 0
[00:00:00.975] LOG -- main.c.201: transport_init
[00:00:00.979] LOG -- main.c.217: transport_open
[00:00:01.159] LOG -- hci.c.3656: BTSTACK_EVENT_STATE 1
[00:00:01.160] EVT <= 60 01 01
[00:00:01.160] CMD => 03 0C 00
BTstack: execute run loop
[00:00:01.161] EVT <= 6E 00
[00:00:01.164] EVT <= 0E 04 05 03 0C 00
[00:00:01.168] CMD => 01 10 00
[00:00:01.171] EVT <= 0E 0C 05 01 10 00 08 0E 03 08 60 00 0E 03
[00:00:01.176] LOG -- hci.c.1779: Manufacturer: 0x0060
[00:00:01.181] EVT <= 6E 00
[00:00:01.184] CMD => 14 0C 00
[00:00:01.187] EVT <= 0E FC 05 14 0C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:01.255] LOG -- hci.c.1693: local name:
[00:00:01.259] LOG -- hci.c.1484: Received local name, need baud change 0
[00:00:01.266] EVT <= 6E 00
[00:00:01.268] CMD => 02 10 00
[00:00:01.271] EVT <= 0E 44 05 02 10 00 BF FF FF 03 CC FF EF FF 3F FF FC 1F F2 0F E8 FE 3F F7 8F FF 1C 00 04 00 61 F7 FF FF 7F 38 C0 FF 7F
 FF FF 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:00:01.291] LOG -- hci.c.1789: Local supported commands summary 0x3f
[00:00:01.298] EVT <= 6E 00
[00:00:01.300] CMD => 09 10 00
[00:00:01.303] EVT <= 0E 0A 05 09 10 00 82 F1 04 C4 0A 24
[00:00:01.308] LOG -- hci.c.1741: Local Address, Status: 0x00: Addr: 24:0A:C4:04:F1:82
BD-ADDR: 24:0A:C4:04:F1:82
[00:00:01.318] EVT <= 6E 00
[00:00:01.321] CMD => 05 10 00
[00:00:01.324] EVT <= 0E 0B 05 05 10 00 FD 03 FF 09 00 04 00
[00:00:01.329] LOG -- hci.c.1710: hci_read_buffer_size: ACL size module 1021 -> used 1021, count 9 / SCO size 255, count 4
[... 338 lines cut ...]
[00:03:25.356] LOG -- btstack_tlv_esp32.c.68: read tag 42544C0A
[00:03:25.361] LOG -- btstack_tlv_esp32.c.54: tag 42544c0b -> 42544C0B
[00:03:25.368] LOG -- btstack_tlv_esp32.c.68: read tag 42544C0B
[00:03:25.373] LOG -- btstack_link_key_db_tlv.c.95: tag 42544c0b, addr 00:1E:96:01:65:4F
[00:03:25.381] CMD => 0B 04 16 4F 65 01 96 1E 00 5B B3 5E 0B CD 8B 84 14 2C 9E F4 8C 2B 9C 84 0A
[00:03:25.390] EVT <= 13 05 01 80 00 01 00
[00:03:25.394] EVT <= 0E 0A 05 0B 04 00 4F 65 01 96 1E 00
[00:03:25.399] EVT <= 6E 00
[00:03:25.420] EVT <= 06 03 00 80 00
[00:03:25.420] CMD => 13 04 03 80 00 01
[00:03:25.421] EVT <= 0F 04 0C 05 53 0C
[00:03:25.421] EVT <= 0F 04 12 05 13 04
[00:03:25.423] EVT <= 6E 00
[00:04:25.592] EVT <= 05 04 00 80 00 13
[00:04:25.593] LOG -- l2cap.c.963: l2cap_stop_rtx for local cid 0x42
[00:04:25.593] LOG -- l2cap.c.857: L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x42
[00:04:25.598] EVT <= 71 02 42 00
[00:04:25.601] LOG -- rfcomm.c.1032: L2CAP_EVENT_CHANNEL_CLOSED cid 0x42, mult 0x0
[00:04:25.608] LOG -- l2cap.c.963: l2cap_stop_rtx for local cid 0x41
[00:04:25.614] LOG -- l2cap.c.857: L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x41
[00:04:25.621] EVT <= 71 02 41 00
[00:04:25.624] LOG -- rfcomm.c.1032: L2CAP_EVENT_CHANNEL_CLOSED cid 0x41, mult 0x0
[00:04:25.632] LOG -- hci.c.823: Connection closed: handle 0x80, 00:1E:96:01:65:4F
[00:04:25.639] LOG -- hci.c.3719: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 0
[00:04:25.645] EVT <= 61 01 00

mringwal, 2017-10-06:
hi. no problem about posting here. it actually looks like an issue. I wasn't able to reproduce it, but from your log, I conclude that it's a bug in the ESP32 Bluetooth controller. I've tried to post an issue at the esp-idf project, let's see.

GustavoOlego, 2017-10-06:
Sr Matthias Ringwald many thanks for your help I saw your post I can not
see so deep as you.
Best Regards

mringwal, 2019-04-28:
According to https://github.com/espressif/esp-idf/issues/1082, the bug may have been fixed.
Are you able to verify the fix? (I wasn't able to reproduce it before)

mringwal, 2019-06-03:
(closing due to no feedback and good chance that is was fixed)

closed 2019-06-03 by mringwal