bluekitchen/btstack #84: receive data after 'can send now'

asuivelentine, 2017-10-19:
Hi,
I have some trouble with the 'CAN_RECEIVE_NOW' event - timing. After a Connection is established the peer sends its first request. As soon as I receive that request I'll handle it and respond. 

However I cant respond because the CAN_SEND_NOW event gets called after I try to respond.

How can I change the default behavior to prioritise the send channel? Something like:
getting connected -> set up send channel -> receive first packet?

Thanks.

mringwal, 2017-10-20:
hi. you may not be able to respond in the packet handler callback. you need to keep application state and can call rfcomm_request_can_send_now and then send when you get the RFCOMM_EVENT_CAN_SEND_NOW.

Please read: http://bluekitchen-gmbh.com/btstack/protocols/#rfcomm-radio-frequency-communication-protocol

asuivelentine, 2017-10-21:
Thanks, I guess that problem can be solved using the credit system. 
However what I do not understand:
If btstack and main application runns in the same context (process/tast/thread) why does it matter if I block the callback handler? Btstack couldn't do anything while my task ist running.

mringwal, 2017-10-23:
Blocking in the callback handler is fine with BTstack, but nothing else will happen in parallel.
If the outgoing buffer is not available on packet callback entry, it won't change. As a consequence, you need to 'schedule' your send request one way or the other.

closed 2017-10-23 by mringwal