bluekitchen/btstack #85: Esp32 build errors in non-BT projects after installing btstack

pigrew, 2017-10-23:
Hi,

I installed bstack using your script (integrate_btstack.py) for the ESP32.

After doing that, my non-bluetooth projects now give a compile error:

`In file included from C:/Users/Nathan/AppData/Roaming/SPB_16.6/esp/esp-idf/components/btstack/main.c:58:0:
C:/Users/Nathan/AppData/Roaming/SPB_16.6/esp/esp-idf/components/btstack/main.c: In function 'transport_open':
C:/Users/Nathan/AppData/Roaming/SPB_16.6/esp/esp-idf/components/bt/include/bt.h:61:50: error: static assertion failed: "please enable bluetooth in menuconfig to use bt.h"
 #define BT_CONTROLLER_INIT_CONFIG_DEFAULT() {0}; _Static_assert(0, "please enable bluetooth in menuconfig to use bt.h");
                                                  ^
C:/Users/Nathan/AppData/Roaming/SPB_16.6/esp/esp-idf/components/btstack/main.c:222:41: note: in expansion of macro 'BT_CONTROLLER_INIT_CONFIG_DEFAULT'
     esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
                                         ^
`

Perhaps the btstack's main.c needs an ifdef that loads it only when bluetooth is enabled?

Or are you suggesting that we have a separate esp-idf director for btstack vs non-btstack?

Thanks!

mringwal, 2017-10-24:
hi @pigrew Thanks for the hint. I did try I2S recently and also wondered about the compile error due to missing Bluetooth support. I didn't realize at that time that this was caused by BTstack. Good idea with the #ifdef. I'll try to see if that's enough later and let you know.

pigrew, 2017-10-24:
Thinking about it again (without actually trying to solve the problem), perhaps BTStack needs to have its own configuration option in menuconfig, so it can be disabled.
If BTStack only does an #ifdef on the Bluetooth option, it may break the Espressif BT examples.

mringwal, 2017-10-25:
hi. I've added the #ifdef and also added code to print an error to the console if BTstack is actually started although CONFIG_BE_ENABLE is not defined. Please give it a try. (seems to work without an entry in the menuconfig. the linker will drop all BTstack code if it's never intialised, too).

commit ed732917 (2017-10-25): esp32: fix build of non-btstack examples when added as component to esp-idf. Fixes #85
closed 2017-10-25 by mringwal