bluekitchen/btstack #86: ESP32 compilation

chegewara, 2017-11-06:
Hi, 
there is small issue to compile any example:
```
esp-idf/components/btstack/platform/freertos/btstack_uart_block_freertos.c:54:2: error: #error "The BTstack HAL UART Run Loop integration (btstack_uart_block_freertos) needs to trigger Run Loop iterations from ISR context," "but neither 'INCLUDE_xEventGroupSetBitFromISR' is enabled in your FreeRTOS configuration nor HAVE_FREERTOS_TASK_NOTIFICATIONS is enabled in " "btstack_config.h. Please enable INCLUDE_xEventGroupSetBitFromISR or HAVE_FREERTOS_TASK_NOTIFICATIONS."
 #error "The BTstack HAL UART Run Loop integration (btstack_uart_block_freertos) needs to trigger Run Loop iterations from ISR context," \
```
But adding `#define HAVE_FREERTOS_TASK_NOTIFICATIONS` to btstack_config.h fix the problem and examples compile and seems to works fine. 
Thanks, nice project.

mringwal, 2017-11-06:
hi. thanks for reporting. It's a 'transitional' issue and fixed on develop branch, but not on master. We'll merge it back soon.

mringwal, 2017-11-09:
fixed in develop & master

closed 2017-11-09 by mringwal