bluekitchen/btstack #87: Unable to compile examples (esp32)

michalkalinowski-, 2017-11-06:
Hello,

I'm following steps from readme found in ports/esp32

* I have the latest toolchain (I'm at 0ee9d93e5837d184 from 06.11)
* I've run` ./integrate_btstack.py`

Whenever i try to build any of the examples with `make flash monitor` I get following error

```
CC build/btstack/3rd-party/bluedroid/decoder/srce/alloc.o
make[1]: *** No rule to make target `3rd-party/bluedroid/decoder/srce/bitalloc-sbc.o', needed by `libbtstack.a'.  Stop.
make: *** [component-btstack-build] Error 2
```

Any ideas how to solve that? I must admit I'm a newbie so please explain as I was 5 ;)

michalkalinowski-, 2017-11-06:
ok, I solved my first problem by randomly running `make clean` and then `make flash`. 

Now I have another issue:
```
 error: #error "The BTstack HAL UART Run Loop integration (btstack_uart_block_freertos) needs to trigger Run Loop iterations from ISR context," "but neither 'INCLUDE_xEventGroupSetBitFromISR' is enabled in your FreeRTOS configuration nor HAVE_FREERTOS_TASK_NOTIFICATIONS is enabled in " "btstack_config.h. Please enable INCLUDE_xEventGroupSetBitFromISR or HAVE_FREERTOS_TASK_NOTIFICATIONS."
 #error "The BTstack HAL UART Run Loop integration (btstack_uart_block_freertos) needs to trigger Run Loop iterations from ISR context," \
  ^
make[1]: *** [platform/freertos/btstack_uart_block_freertos.o] Error 1
make: *** [component-btstack-build] Error 2
```

mringwal, 2017-11-06:
See issue #86

closed 2017-11-06 by mringwal