bluekitchen/btstack #88: Keyboard + Mouse for BT Classic example

benjaminaigner, 2017-11-13:
Hi Matthias,

I've attached one example for a combination of mouse and keyboard in classic BT.
Joystick is partly added (based on Arduino/Paul Stoffregen's Joystick), but no changes to the GATT table are made (result: non-functional).

[keyboard_mouse.zip](https://github.com/bluekitchen/btstack/files/1467444/keyboard_mouse.zip)

Best wishes,
Benjamin

BTW: Sry I cannot use BTStack in this case, we need for the FABI/FLipMouse devices complete FOSS stacks :-(.

mringwal, 2017-11-14:
Thanks for the demo. In Classic HID, there's no GATT. Do you mean, there's no change to the HID Descriptor?

benjaminaigner, 2017-11-14:
Ok, sry, you're right:
I've changed the HID descriptor to add report a report ID, enabling mouse and keyboard.
I did not manage to have this descriptor to be recognized as Joystick (under Linux), but Keyboard and Mouse work.