bluekitchen/btstack #91: InstrFetchProhibited with ESP32

gtalusan, 2017-11-22:
Hi there,

I'm playing around with ESP-IDF and BTStack (git rev 9c2d55862e8aa00c90395a3fa4dd7bd70f80d0f9) trying to get the A2DP SRC demo up and running.  I flashed it but the demo causes a Guru meditation here:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4332
load:0x40078000,len:0
load:0x40078000,len:11992
entry 0x40078e34
I (123) cpu_start: Pro cpu up.
I (123) cpu_start: Single core mode
I (123) heap_init: Initializing. RAM available for dynamic allocation:
D (126) heap_init: New heap initialised at 0x3ffae6e0
I (132) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
D (138) heap_init: New heap initialised at 0x3ffb5da8
I (143) heap_init: At 3FFB5DA8 len 0002A258 (168 KiB): DRAM
I (149) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (155) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (162) heap_init: New heap initialised at 0x40087e44
I (167) heap_init: At 40087E44 len 000181BC (96 KiB): IRAM
I (173) cpu_start: Pro cpu start user code
D (185) clk: RTC_SLOW_CLK calibration value: 3373875
V (194) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (194) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (200) intr_alloc: Connected src 46 to int 2 (cpu 0)
V (205) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (211) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xC0E
D (219) intr_alloc: Connected src 57 to int 3 (cpu 0)
V (224) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (231) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E
D (239) intr_alloc: Connected src 24 to int 9 (cpu 0)
I (244) cpu_start: Starting scheduler on PRO CPU.
D (249) heap_init: New heap initialised at 0x3ffe0440
D (253) heap_init: New heap initialised at 0x3ffe4350
V (259) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (259) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (259) intr_alloc: Connected src 16 to int 12 (cpu 0)
BTstack: setup
[00:00:00.280] LOG -- btstack_run_loop_freertos.c.260: run loop task 0x3ffb7544
[00:00:00.287] LOG -- btstack_run_loop_freertos.c.263: run loop init, queue item size 8
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x00000000  PS      : 0x00060b30  A0      : 0x800e3ae9  A1      : 0x3ffb7470  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffb8ef8  A5      : 0x3ffb229c  
A6      : 0x3ffafff4  A7      : 0x00060023  A8      : 0x800e4d1b  A9      : 0x3ffb7460  
A10     : 0x400e6018  A11     : 0x00000000  A12     : 0x000007b4  A13     : 0x3ffb90ac  
0x400e6018: packet_handler at /Users/george/esp/esp-idf/components/btstack/src//hci.c:3843

A14     : 0x00000000  A15     : 0x0000007b  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb7470 0x400e3ae6:0x3ffb7490 0x400d1583:0x3ffb74b0
0x400e3ae6: app_main at /Users/george/esp/esp-idf/components/btstack/./main.c:365

0x400d1583: main_task at /Users/george/esp/esp-idf/components/esp32/./cpu_start.c:449

================= CORE DUMP START =================
nA4AAAUAAAB0AQAA
RHX7P7Bz+z88dfs/
wHH7P9B0+z/av4QrjBz7P4wc+z9Edfs/hBz7PxgAAADfcVCIEGTczUR1+z8AAAAA
AQAAAEBj+z9tYWluAK6OiadlCCR06eoAAAAAADx1+z8AAAAAIAcGAAEAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAHDo+j/Y6Po/QOn6PwAAAAAAAAAA
AQAAAAAAAACgI0A/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA7w5F
AHcCAAAAAAAwCwYA6ToOgHB0+z8AAAAAAAAAAPiO+z+cIvs/9P/6PyMABgAbTQ6A
YHT7PxhgDkAAAAAAtAcAAKyQ+z8AAAAAewAAAAQAAAAUAAAAAAAAAGzEAEB3xABA
AAAAALQHAACskPs/8BgIQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAACAJBgAAAAAAAAAAADCH+z+GFQ2AkHT7PwAAAAA4CPs/
AAAAAJwi+z/0//o/IwAGAAAAAACwdPs/AAAAAAAAAADo//o/gBv7P4Ab+z84CPs/
AAAAANB0+z8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAADcdPs/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAA
RHv7PxB6+z88e/s/
EHr7P9B6+z84KAASeBz7P3gc+z9Ee/s/cBz7PxkAAABSjasoTpr2+ER7+z8AAAAA
AAAAAEB3+z9JRExFAPhkbDvVzCNxpBYAAAAAADx7+z8AAAAAkATcNAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAHDo+j/Y6Po/QOn6PwAAAAAAAAAA
AQAAAAAAAACgI0A/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAmYjB
QBkIQFxBCEAwAAUAAAAAANB6+z8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAADcevs/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAA
uIX7P2CE+z+whfs/
YIT7P1CF+z8AAAAAUBz7P1Ac+z+4hfs/SBz7PxgAAADofPs/6Hz7P7iF+z/gfPs/
AQAAALR9+z9UbXIgU3ZjAOjt6+v+HPwAAAAAALCF+z8AAAAAIQAGAAEAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAHDo+j/Y6Po/QOn6PwAAAAAAAAAA
AQAAAAAAAACgI0A/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAALKau
QBkIQNFNCEAwAAYAAAAAACCF+z8AAAAAAAAAAAAAAABsHvs/aB77PwEAAADRTQiA
AIX7PwAAAABQCPs/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAADkbCEAAAAAAUE8IQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAUIX7PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAABchfs/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
cPz6P+D6+j9o/Po/
4Pr6PwD8+j84oqG0CBz7P9Bh+z9w/Po/ABz7PwMAAABA6vo/QOr6P3D8+j846vo/
FgAAAGzq+j9lc3BfdGltZXIAPcLfaxwAAAAAAGj8+j8AAAAAIQAGABYAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAHDo+j/Y6Po/QOn6PwAAAAAAAAAA
AQAAAAAAAACgI0A/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAk0Vj
QBkIQPA3CEAwAAYABRYNgKD7+j8U6vo/AAAAAGDq+j8AAAAAFOr6PwAAAADwNwiA
gPv6PwAAAAD/////ZB77PwAAAABQCPs//////wAAAAClpaWlpaWlpQAAAAAAAAAA
AAAAADkbCEAAAAAAUE8IQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA4Pv6Pygb+z8AAAAA
AAAAAAAAAAAAAAAAAAAAAP////8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAD8+j8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAM/Po/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAA=
yGH7P0Bg+z/AYfs/
QGD7P2Bh+z8CMCy2ePz6Pwgc+z/IYfs/ABz7PwEAAABk//o/ZP/6P8hh+z9c//o/
GAAAAMRd+z9pcGMwAHuQhYZiJ35QMJ8AAAAAAMBh+z8AAAAAIQAGABgAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAHDo+j/Y6Po/QOn6PwAAAAAAAAAA
AQAAAAAAAACgI0A/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAG59O
QBkIQPA3CEAwAAYAcxUIgABh+z84//o/AAAAAIT/+j8AAAAAOP/6PwAAAADwNwiA
4GD7PwAAAAD/////ZB77PwAAAABQCPs/AgAAAAAAAAClpaWlpaWlpQAAAAAAAAAA
AAAAADkbCEAAAAAAUE8IQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQGH7P3Ab+z9wG/s/
AAAAAAAAAAAAAAAAAAAAAP////8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAGBh+z8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACmDQiA0Dv+P+wb+z8DAAAA
AAAAAAAAAABsYfs/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
================= CORE DUMP END =================
CPU halted.
```

gtalusan, 2017-11-22:
Ugh, some slight progress.. I can connect to the speaker now, but I get another guru...

```
AVRCP: connected to 32:54:02:BB:01:EF, avrcp_cid 0x01
[00:00:09.874] EVT <= 13 05 01 80 00 01 00 
[00:00:09.878] EVT <= 6E 00 
[00:00:09.880] EVT <= 78 02 41 00 
[00:00:09.884] ACL => 80 00 07 00 03 00 41 00 10 07 04 
[00:00:09.889] EVT <= 6E 00 
[00:00:09.891] CMD => 35 0C 05 01 80 00 06 00 
[00:00:09.895] EVT <= 6E 00 
[00:00:09.898] EVT <= 13 05 01 80 00 01 00 
[00:00:09.957] ACL <= 80 20 06 00 02 00 41 00 12 07 
[00:00:09.958] LOG -- a2dp_source.c.273: Accepted 7
[00:00:09.958] EVT <= 78 02 43 00 
[00:00:09.959] LOG -- a2dp_source.c.516: A2DP source: a2dp cid 0x01 not known, expected 0x00
A2DP: Stream started.
[00:00:09.969] CMD => 35 0C 05 01 80 00 01 00 
[00:00:09.973] EVT <= 6E 00 
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x400f84f0  PS      : 0x00060630  A0      : 0x800ea5c2  A1      : 0x3ffd0160  
0x400f84f0: btstack_sbc_encoder_num_audio_frames at /Users/george/esp/esp-idf/components/btstack/src/classic/btstack_sbc_bludroid.c:572

A2      : 0x00000000  A3      : 0x00002705  A4      : 0x00060420  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00060023  A8      : 0x800e8b76  A9      : 0x3ffd0540  
A10     : 0x000026fa  A11     : 0x00002705  A12     : 0x3ffd1a8c  A13     : 0x00000002  
A14     : 0x0000000a  A15     : 0x00000080  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400f84f0:0x3ffd0160 0x400ea5bf:0x3ffd0180 0x400ea698:0x3ffd05a0 0x400e8c01:0x3ffd05c0 0x400e4b29:0x3ffd05f0 0x400e46c4:0x3ffd0610 0x400d1536:0x3ffd0630
0x400f84f0: btstack_sbc_encoder_num_audio_frames at /Users/george/esp/esp-idf/components/btstack/src/classic/btstack_sbc_bludroid.c:572

0x400ea5bf: a2dp_demo_fill_sbc_audio_buffer at /Users/george/build/btstack/port/esp32/example/a2dp_source_demo/main/./a2dp_source_demo.c:224

0x400ea698: a2dp_demo_audio_timeout_handler at /Users/george/build/btstack/port/esp32/example/a2dp_source_demo/main/./a2dp_source_demo.c:267

0x400e8c01: btstack_run_loop_freertos_execute at /Users/george/esp/esp-idf/components/btstack/platform/freertos/btstack_run_loop_freertos.c:172

0x400e4b29: btstack_run_loop_execute at /Users/george/esp/esp-idf/components/btstack/src//btstack_run_loop.c:181

0x400e46c4: app_main at /Users/george/esp/esp-idf/components/btstack/./main.c:384

0x400d1536: main_task at /Users/george/esp/esp-idf/components/esp32/./cpu_start.c:449
```

gtalusan, 2017-11-23:
I threw a ton of logs into `packet_handler`.

The code trips into `case AVDTP_SUBEVENT_STREAMING_CONNECTION_ESTABLISHED:` and sets `app_state = A2DP_STREAMING_OPENED;`.  At this point, `btstack_sbc_encoder_init` hasn't yet been called in `case A2DP_W2_OPEN_STREAM_WITH_SEID:`, so the code in `btstack_sbc_encoder_num_audio_frames` in the stack frame above will crap out.

mringwal, 2017-11-23:
hi. It's weird that it behaves differently for you. Could you paste the full log here (including your additional logs if they use log_info)?

mringwal, 2017-11-23:
asides from that btstack_sbc_encoder_init should be called from the demo code and not from a2pp_source. we'll clean this up tomorrow, but if you could post a full log, that would help to see if you did run into a different issue.

gtalusan, 2017-11-23:
Hi Mathhias, here's a log from master..

https://github.ops.kobo.com/gist/gtalusan/a190cfe63b6a8dd1a6d84bd3259a8b67

Sometimes I'll can get it to connect to the device without crashing.  The demo will start a stream, but I won't hear any sound from the speaker.  I also noticed that sometimes pressing 'p' to pause the stream won't print the expected console message - I believe it's because `if (!media_tracker.connected) break;` is getting tripped over.

Using one of these as the speaker:

https://www.amazon.ca/OKCSC-Wireless-Bluetooth-Portable-Speakers/dp/B075XM64G7/ref=sr_1_1?ie=UTF8&qid=1511456652&sr=8-1&keywords=bluetooth+speaker+dog

It works well with my various iDevices and iComputers.

mringwal, 2017-11-23:
Cute speaker :)
Could you check the log? I get a server error when clicking on it.

gtalusan, 2017-11-23:
Whoops, my bad.. I updated the link.

mringwal, 2017-11-23:
thanks, that works now.

just to confirm, in the case where you've looked at the crash, it's because sbc_encoder_state_singleton == NULL, which would be a consequence of btstack_sbc_encoder_init not being called before. 

I had a quick look at the log and BTstack assumes that your speaker only supports the minimal MTU of 48, which a) isn't correct - at least the speaker agrees to BTstack's MTU of 1691, and b) prevents the demo from sending any SBC frame as they are 118 bytes.

We'll have a look soon/tomorrowish and get back ASAP.

mringwal, 2017-11-24:
hi. next step: could you git pull, switch to the develop branch and try again?

We've moved the sbc decoder init into the demo app, made sure it gets the sbc config, and also learned that the default l2cap mtu is 672. Almost all other stacks send their MTU anyway, so the default value didn't matter. I'm glad that's fixed.

If everything goes to plan, you should get some stutter sound on the speaker. Now, please disable the packet logging, by commenting the call to hci_dump_open(..) in esp-idf/components/btstack/main.c - main() and try again. The console output slows down the audio transmission and may or may not cause a guru panic (overload...). You should hear actual music now.

gtalusan, 2017-11-24:
Hi Mathhias, thanks, I pulled develop, rebuilt and deployed but it still crashed.

Here's the gist: https://gist.github.com/gtalusan/a27d4527e5d70a580f5068fff69a69d9

It still looks like the SBC encoder isn't getting created.

mringwal, 2017-11-24:
Did you run ./integrate_btstack.py, too? The stack is copied/integrated into esp-idf and the example projecrs (re)created. Sorry, didn't mention that.

gtalusan, 2017-11-24:
Yup, I did.

mringwal, 2017-11-24:
Right, the remote mtu is listed as 672. We're getting closer :)

Will have a deeper look.

gtalusan, 2017-11-24:
I put a log_info into `AVDTP_SUBEVENT_SIGNALING_MEDIA_CODEC_SBC_CONFIGURATION` in `a2dp_source.c` and it's not getting called.

mringwal, 2017-11-24:
Thanks for the hint. I've ordered the same speaker (panda edition) from China (for fun), but free shipping takes 2-4 weeks. 

However, I've noticed that you're starting the connection from the speaker and tried reconnecting from my Sony headset. I don't get the same error, but it doesn't work either. At least we have something to test against/fix now.

If you like, you can try to connect from the esp32 again by putting the speaker into pairing mode and pressing 'b' as you did the very first time. It might work (after packet logging is disabled) :)

gtalusan, 2017-11-24:
I did manage to get it to play a snippet of the mod file once using the 'b' method, but I haven't been able to reproduce it.

Here's the gist: https://gist.github.com/gtalusan/2e520bb5163c38f63286e492f0540c11

mringwal, 2017-11-24:
Thanks. We'll look into it. Just a reminder, the packet logging must be turned off (log messages are fine) to stream music as console output happens at 115200.

gtalusan, 2017-11-24:
Got it!  I've ordered a few Bluetooth headphones from Amazon and Banggood so that I can play some more.  Looking forward to some fixes :D

gtalusan, 2017-11-25:
Hi Matthias, here are a few more logs from two different Amazon headphones:

https://gist.github.com/gtalusan/b26798e48de5f9b26e79a9be7f98a116
https://gist.github.com/gtalusan/486677ac14cebbc16e75a6bd40306fbd
https://gist.github.com/gtalusan/f534befc9fba530128dec92f265fd75f

mringwal, 2017-11-26:
Nice to have Amazon deliver next day (or so). 

Your animal speaker does not respond to the Get Configuration query correctly. However, since BTstack actively sets the stream configuration, there's no reason to ask again. We'll change this to forward the SBC config to the app when the remote side accepts the Set Configuration. (The missing/incorrect configuration causes the crash).

In your new 2. log, there's an assertion in the BTstack code that the esp32 Bluetooth task performs it's callbacks from a thread context. I didn't see this in a while and had assume that it was fixed by espressif. Do you have some current version of esp-idf? It not, please git pull master.

(Fix for the Get/Set Config is in the works...)

gtalusan, 2017-11-26:
Thanks for the quick response!

Yes, my ESP-IDF is fairly recent... Nov 22.  I'll pull latest and see what's up.

gtalusan, 2017-11-26:
Yup, same thing... the assertion happens every two or three runs.

mringwal, 2017-11-26:
The SBC config as Initiator (most of your logs) should be fixed on the develop branch now. If you like, give it a quick try. More fixes to come - will check if the assert for callbacks from IRQ can be avoided and just handled correctly next.

gtalusan, 2017-11-27:
Great, both the speakers and headphones now work.  Thanks a lot for the quick turn around!  You've been super helpful.

closed 2017-11-27 by gtalusan

mringwal, 2017-11-27:
Glad to hear. We still need to address more issues: A2DP allows most procedures to be started by either side, and we didn't implement the code to handle collision (both sides send a command at the same time) yet. We'll going to fix that soon, too. Also, as mentioned, I'll need to look at the esp-idf callbacks from IRQ context.

mringwal, 2017-11-27:
I've filed an issue about the callback from IRQ: https://github.com/espressif/esp-idf/issues/1322

In the packet received callback, the packet needs to be placed into a ring buffer. The current code assumes that it's not called from ISR context and only uses a regular mutex to protect the data structures.

Making this safe from ISR isn't easy, and I've verified that the Bluedroid code doesn't expect calls from ISR either. I'd rather have Espressif fix this bug instead of writing complex code. Let's see if they react.