bluekitchen/btstack #93: Can BTstack run within Mongoose OS ?

pavestru, 2017-12-12:
I have followed the [readme for ESP32 platform](https://github.com/bluekitchen/btstack/tree/master/port/esp32) and the examples worked like charm on my ESP32 chip.

Now, Mongoose OS has made it fast for me to start developing for ESP32, except that they do not support Classic BT.

So I tried wrapping the `$IDF_PATH/components/btstack` folder as a Mongoose OS library. The result is here:
https://github.com/pavestru/mongoose-os-lib-btstack

Then I have initialized a mongoose app using `mos init`, put there the mongoose-os-lib-btstack as a dependency and written almost no code in main.c - only log to console. The demo is here:
https://github.com/pavestru/mongoose-btstack-demo

I have also added `gap_inquiry.c` example and it runs just fine.

However, the *mongoose-os-lib-btstack* dependency is causing function `mgos_app_init`, which should be the entry point for the project, not to run at all - the log does not show in the console. If I remove mongoose-os-lib-btstack from dependencies, the log shows in the console.

Apart from this issue, Mongoose OS has trouble connecting chip to WiFi even though it seems to be running fine alongside BTstack.

**Note:** I had to rename app_main and run it in a separate task (using xTaskCreate) in order to run alongside Mongoose OS and not cause code dumps.

Do you have an idea what I should change in order to run the two (BTstack and Mongoose OS) seamlessly?

Thank you so much for your work!

mringwal, 2017-12-13:
Hi Pavel. Thanks for running BTstack on Mongoose OS. I don't know much about Mongoose OS yet, although I've used an older version of their HTTP server in some private projects.

Right now, I don't understand what's not working for you, or have any idea why.

Our ESP32  port uses the FreeRTOS Run Loop Implementation. Does Mongoose OS uses FreeRTOS as well? (In your repo, I don't see changes to that effort).

pavestru, 2017-12-13:
Thank you @mringwal for your answer.

Yes, Mongoose OS uses FreeRTOS as well, because it's built on top of esp-idf. I'll probably have to go deeper to understand FreeRTOS run loop and overall architecture in order to understand this problem.

mringwal, 2017-12-18:
If Mongoose OS uses FreeRTOS, the name is a bit weird :)
Anyway, if BTstack and the demo runs on their own FreeRTOS thread, there's no obvious reason why it would block in some circumstances. 
Please let me know if you figure out why.

fri-sch, 2018-01-03:
@pavestru, did you make some progress?
I'm also trying to get BTstack running with Mongoose OS. I tried your library, but I'm already failing while trying to build:
```
/opt/Espressif/esp-idf/Kconfig:111: recursive inclusion detected. Inclusion path:
  current file : '/opt/Espressif/esp-idf/Kconfig'
  included from: '/opt/Espressif/esp-idf/components/bt/Kconfig:2'
```

I also posted to the Mongoose OS board: https://forum.mongoose-os.com/discussion/1894/using-btstack-with-esp32-and-mongoose-os
Maybe there's someone who can help!?

Thanks!

geekspeed, 2018-01-04:
@pavestru it looks like in your mos_esp32.yml you leave the bluedroid subsystem enabled:
>CONFIG_BLUEDROID_ENABLED=
Try it with:
>CONFIG_BLUEDROID_ENABLED=n

fri-sch, 2018-01-04:
Ok, I just missed, that I still have the rpc-service-gattc library included. It sets `CONFIG_BLUEDROID_ENABLED=y` and causes the build failure. 

Apart from that either `CONFIG_BLUEDROID_ENABLED=` or `CONFIG_BLUEDROID_ENABLED=n` seems to work to disable the Bluedroid stack.

pavestru, 2018-01-04:
Hey, @fri-sch, I'm glad you made it build. I did not have issues building it, though. I'm getting by without mongoose os right now and have found using plain esp-idf not that cumbersome.

mringwal, 2018-03-28:
Seems solved so far. Please open a new issue if you identify other problems.

closed 2018-03-28 by mringwal